ZuluSCSI/ZuluSCSI-firmware #1: Consider removing remaining trivial BlueSCSI code

erichelgeson, 2022-04-15:
It seems there is only a few lines left from the "fork" of BlueSCSI around image name handling.

Consider removing this and rebasing the fork without BlueSCSI code in it so you can removing BlueSCSI from your site, marketing, and the like. You could even consider a non-blue-color based name in future revisions.

reinauer, 2022-04-15:
What's wrong with that code? Is it not Open Source? Security issues? This makes me a bit worried using my BlueSCSI now.

aperezbios, 2022-04-15:
There's no need to fix what isn't broken.

closed 2022-04-15 by aperezbios

erichelgeson, 2022-04-15:
The name of the project and the trivial amount of BlueSCSI code seem to just be here for marketing purposes. The codebase seems to have nothing to do with BlueSCSI - but the name and marketing make mention of it quite a bit. This is more a port of scsi2sd to a GD32 chip - which is a great and cool thing to do with the current chip shortage.

You of course are free to use the code under the license it was provided, but since the use is so trivial and the name BlueSCSI is used on the marketing side of things to garner the good will associated with the name.

I would be happy to re-licenses my trivial parts of code used in a way that you would not need to provide any attribution. If that's acceptable I can open a PR removing any association. Win win.

aperezbios, 2022-04-15:
The name is used because it's useful to explain to people who are interested, and have a basic understanding of how BlueSCSI works, from a user's point of view. The configuration mechanism is nothing like SCSI2SD, and it's very clear based on the commit history of this repository that we _started with_ a straight fork of BlueSCSI, and then iterated and improved upon that, over several months.

If "BlueSCSI" is or becomes a trademarked name, we will act accordingly, and add the trademark symbol where it's used. As of today, it does not appear to be trademarked, pending or otherwise, according to the USPTO's Trademark Electronic Search System, at https://tmsearch.uspto.gov/

I'm open to any modifications to wording that are reasonable, but what you're requesting is rather far from that.

erichelgeson, 2022-04-15:
> The name is used because it's useful to explain to people who are already familiar with how BlueSCSI works at a fundamental level.

Right, my point.

> we started with a straight fork of BlueSCSI

I saw that too. But it's confusing as part way through the scsi2sd code was just copied in and used in it's place. It may have been more straight forward to just do the port without starting with BlueSCSI.

> I'm open to any modifications to wording that are reasonable, but what you're requesting is rather far from that.

Again, you are free to do what you like under the license allowed. As I have indicated to you many times - I do not want myself associated with any of your commercial products. It would just be a kind gesture to cut ties and move along with your own projects since there is nothing of consequence from BlueSCSI used in this project.

aperezbios, 2022-04-15:
We replaced the original BlueSCSI command handling code because it was a known quantity, has broad compabitility, and required far less engineering effort to use in the opinion of our developer. 

If the ask is that we remove your name and copyright notices from the AzulSCSI source code repository, that's something we're willing to consider doing, but IANAL, and I'd need to talk to one, before I unilaterally decide to do something that could cause unintended legal consequences. The idea of removing copyright notices seems ill-advised, at least on the surface. All I can commit to doing at this time is investigating whether or not your request is something we can legally comply with. I'll let you know.

erichelgeson, 2022-04-15:
That is not the ask. Again:

Replace the ~30 trivial lines of code and there is no original BlueSCSI code left
or
Allow me to provide those ~30 lines of trivial code to you under a license which you will not need to reference me or BlueSCSI

There would then be no lineage of BlueSCSI as the rest is either written by your employee or from the scsi2sd repo. You can go on your way with azul scsi2sd and our paths never need to cross again.

aperezbios, 2022-04-16:
Since you're the one requesting this, please call out the relevant line numbers and filenames, and we'll see what can be done, if anything.