ZuluSCSI/ZuluSCSI-firmware #4: E-mu Emax sampler compatibility

jacdx, 2022-05-05:
Hello,

Just got my new AzulSCSI plugged in and am having difficulty getting it working. The Emax is a 1986-1989 sampler with a NCR 5380 controller, originally having a 20MB hard drive. It uses a proprietary drive format, but can be read/written by EMXP). It is known to work with SCSI2SD. I enabled verbose logging but don't get much of note, other than a note about image size not matching the drive geometry. It appears that the images are seen and that AzulSCSI starts up fine. One quick LED flash on power, then a slightly longer one. No errors, but the Emax never seems to read from the board (no disk access LED flashes). Attempting to select another SCSI ID doesn't produce any either. Thoughts on how to debug? Possible to enable read-level or command-level logging?

aperezbios, 2022-05-05:
@jacdx thanks for the report. Can you let us know which firmware version you're using, and preferably also include the entirety of your azullog.txt from your SD card, if possible. The firmware version is printed in the log.

Feel free to redact any personal details/filenames if there's a need to do so.

jacdx, 2022-05-06:
Here you go. I've been experimenting with settings a bit. Selection delay of 1, because that's what SCSI2SD requires for this board. I've tried Fat32 and ExFat, and have been experimenting with filenames. And quirks 1 vs 0.

`[0ms] DIPSW3 is OFF: SCSI termination disabled
[0ms] DIPSW2 is ON: enabling debug messages
[1ms] Optional GreenPAK detected, loading firmware
[53ms] GreenPAK firmware successfully loaded
[176ms] SD card detected, FAT64 volume size: 15186 MB
[177ms] SD MID: 0x03, OID: 0x53 0x44
[177ms] SD Name: SB16G
[177ms] SD Date: 8/2014
[178ms] SD Serial: 0xA74409A7
[179ms] Reading configuration from azulscsi.ini
[179ms] Active configuration:
[181ms] -- SelectionDelay: 1
[184ms] -- EnableUnitAttention is on
[841ms] Finding HDD images in directory /:
[842ms] -- Opening /HD10.HDA.EZ1 for id:1 lun:0
[843ms] ---- Image file is contiguous.
[843ms] ---- NOTE: Drive geometry is 63x255=16065 but image size of 38900 is not divisible.
[844ms] -- Opening /HD20_512.IMG for id:2 lun:0
[845ms] ---- Image file is contiguous.
[846ms] ---- NOTE: Drive geometry is 63x255=16065 but image size of 38900 is not divisible.
[847ms] -- Opening /HD30.HDA for id:3 lun:0
[848ms] ---- Image file is contiguous.
[848ms] ---- NOTE: Drive geometry is 63x255=16065 but image size of 38900 is not divisible.
[849ms] -- Opening /HD00_512.EZ1 for id:0 lun:0
[851ms] ---- Image file is contiguous.
[851ms] ---- NOTE: Drive geometry is 63x255=16065 but image size of 38900 is not divisible.
[900ms] SCSI ID:0 BlockSize:512 Type:0 Quirks:1 ImageSize:19450kB
[900ms] SCSI ID:1 BlockSize:512 Type:0 Quirks:1 ImageSize:19450kB
[901ms] SCSI ID:2 BlockSize:512 Type:0 Quirks:1 ImageSize:19450kB
[901ms] SCSI ID:3 BlockSize:512 Type:0 Quirks:1 ImageSize:19450kB
[1405ms] SCSI PHY operating mode: GREENPAK_DMA
[1406ms] Initialization complete!
[1406ms] Platform: AzulSCSI v1.1
[1406ms] FW Version: 1.0.1 May  2 2022 14:08:48`

aperezbios, 2022-05-06:
Thanks. Can you explain why SCSI termination is disabled? It must be enabled, unless it isn't the only device on the bus.

jacdx, 2022-05-06:
The 5380 supposedly has built-in termination. I have tried both ways though.

PetteriAimonen, 2022-05-09:
Can you try this to coax more debugging information out of the device:

1. Boot up with Azul/ZuluSCSI connected.
2. Remove SD card for 5 seconds, the LED should start blinking.
3. Insert SD card back, the LED should blink once.
4. Remove SD card and post the log file here.

Reinserting the SD card will force the log file to be saved, so all the latest debug messages should get to the log.
Otherwise the debug messages are only saved once per SCSI request, which is a bit problematic when no requests get through.

jacdx, 2022-05-09:
Ok, here's two logs. The 'reinsert' one follows your instructions. The 'emaxdiag' one, boots up, then runs a function in the Emax hard drive diagnostics menu to test the "SCSI to Echip". I believe this is a general health check of talking to the device on the SCSI bus. This does 200 passes. First one passed, the rest failed. Then I ran another function which does a read test, failed.

I did these again with minimal configuration only delay=1 enabled, same results.

I looked through the [5380 manual](https://cdn.hackaday.io/files/18974811783616/NCR_5380_SCSI_Interface_Chip_Design_Manual_May85.pdf) for clues, but it's above my head.

I was also looking through the SCSI2SD source and saw a couple of Emax things. Not sure how that translates to this project. [commit1](https://github.com/rabbitholecomputing/SCSI2SD/commit/076117156e1d8fe3c402436e752094c6ab66c756?diff=split) [commit2](https://github.com/rabbitholecomputing/SCSI2SD/commit/6c7ac0966fa1f9a4a1f6b53725cc542c32616760)

[azullog-emaxdiag.log](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8654366/azullog-emaxdiag.log)
[azullog-reinsert.log](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8654367/azullog-reinsert.log)

PetteriAimonen, 2022-05-10:
Thanks for the logs!
I think I have figured out the problem.
In SCSI-1, there exists a single-initiator mode where the host does not use BSY signal, but only SEL signal to start request.
This mode was later removed from SCSI-2 standard.

Please try the attached firmware build, it adds support for single-initiator mode and some extra debug messages.
[ZuluSCSIv1_1_single_initiator_test.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8658998/ZuluSCSIv1_1_single_initiator_test.zip)

jacdx, 2022-05-12:
@PetteriAimonen It worked!!!! I got it working last night and spent a couple of hours in 80s heaven. Nice job!

PetteriAimonen, 2022-05-12:
@jacdx Awesome! The fix will eventually be included in next firmware version :)

chickeneps, 2022-05-12:
Is this a Emax I or a Emax II?

chickeneps, 2022-05-13:
Here are two logs with this new firmware with a MPC60 and a S550. The MPC60 stalls, the LED on the board lights up for 30sec or so, then stops, but the MPC stays stalled and does not recover. The S550 just doesn't see the board at all.
[zululog_MPC60_2022_05_12_01.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8683352/zululog_MPC60_2022_05_12_01.txt)
[zululog_Roland_S550_2022_05_12_01.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8683353/zululog_Roland_S550_2022_05_12_01.txt)
.

chickeneps, 2022-05-13:
Also, why does the timestamp on the log file show Dec 31st 2021?

chickeneps, 2022-05-13:
> @PetteriAimonen It worked!!!! I got it working last night and spent a couple of hours in 80s heaven. Nice job!

jacdx, could you post a empty Emax 1 (if that's what you have) image, and also comment on how large they can be? If it's like the II, the max is just shy of 1000mb. Or maybe not?

chickeneps, 2022-05-13:
jacdx, I got (I think) a Msgr from you and now I can't find it (typical), please contact me via email via the Chicken Systems address.

aperezbios, 2022-05-23:
> Also, why does the timestamp on the log file show Dec 31st 2021?

There's no real-time clock on the ZuluSCSI microcontroller, so it has no way of knowing what the time and date is. The value has to be set to something. Closing as fixed.

closed 2022-05-23 by aperezbios