ZuluSCSI/ZuluSCSI-firmware #5: X68000 Pro SaSi

chris-jh, 2022-05-14:
It is not showing the SCSI Device ID i set the image file (x68000 v4) to (HD30_512.hda)

Does ZuluSCSI Support SaSI like the SCSI2SD ? which works great.

I have tried different settings in the ini file, to no avail.

The ZuluSCSI is attached to the internal 50pin connection, just like how i had my SCSI2SD.

I am unsure what else to try.

Attached is my ini file and the text file. i enabled the debug, but it does not show much.

(Had to rename the ini file for upload to git)
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8693550/zululog.txt)
[zuluscsi_ini.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8693552/zuluscsi_ini.txt)
.

PetteriAimonen, 2022-05-14:
The logs look similar to issue #1, where a test firmware fixed the issue: https://github.com/ZuluSCSI/ZuluSCSI-firmware/issues/4#issuecomment-1122001389

chris-jh, 2022-05-14:
Mmmm, i just downloaded that file, renamed it to ZuluSCSI.bin. but it does not seem to be updating the firmware

chris-jh, 2022-05-14:
I had to rename the file to the old name

chris-jh, 2022-05-14:
Awesome that works, Booted the image no problem 🙂

chris-jh, 2022-05-14:
Only difference i see now is the speed, where scsi2sd is faster in loading. But working and loading games up fine, just slower.

chris-jh, 2022-05-15:
Slow is the wrong word. Attached in the zip file are two videos. One is from SCSI2SD other the ZuluSCSI. Loading Bubble Bobble, the ZuluSCSI the music pauses for a second as it is loading the data.

Tried different SD Cards, this at the moment is the only difference i can see, otherwise if i didn't have the SCSI2SD i probably would not have noticed 😄

But so far what i have loaded, Super Street Fighter 2, New Zealand Story, Bubble Bobble it is working and only bubble bobble i noticed that pause on.
[Videos.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8694672/Videos.zip)

I will play more games, i mean more testing later today 😁

PetteriAimonen, 2022-05-15:
Verify that you have debug messages off for actual use, otherwise they will slow down operation.

chris-jh, 2022-05-15:
Debug was off.

Everything seemed fine, there were a few games that do the loading like bubble bobble that have the same issue.

aperezbios, 2022-05-26:
@chris-jh thanks for the feedback. It seems like you're experiencing some latency with reads. This could be due to the use of exFAT instead of FAT32. Since you're only using a ~4GB SD card, it might be a good idea to copy off the data, re-format the card at FAT32, and copy everything else back. If that makes any perceptible difference, please let us know.

On the subject of SASI compatibility, ZuluSCSI currentlyhas no official SASI support, simply because we don't have any SASI systems to test against. That said, I'm glad it seems to be working, and we appreciate you sharing this information with us.

chris-jh, 2022-05-29:
Hi @aperezbios i tried fat32 still the same.

Attached are 2 photo's of a disk bench utility i found and ran for both the zulu and scsi2sd. Zulu appears faster overall apart from the Sequential seek command and random seek command is a lot slower. Could this maybe what is causing the hiccup in loading ?

[image]
[image]

antorsae, 2022-05-31:
Hello @chris-jh I'm a total x68000 noob, in the process of getting x68000 pro to work and then zuluscsi. 

Question: I have a bunch of HDF files, but I don't think zuluscsi supports HDF format, how do you convert x68000 hard drive image files into something zuluscsi understands?

Thank you.

aperezbios, 2022-06-01:
@antorsae ZuluSCSI knows _nothing_ of image file formats. It literally just takes the contents of the file and exposes it as a block device to the host operating system. As long as an HDF file is the equivalent of a raw disk dump, it should just work. I've never used an X6800, nor do we have one to test against, so we're at the mercy of those who do to provide feedback. It's very likely that "HDF" files is just a so-called "hardfile", which is literally just a raw block-for-block/byte-for-byte dump of what would exist on a normal hard drive.

aperezbios, 2022-06-01:
@chris-jh please double-check that you have debug disabled. It can be enabled two different ways. With the ZuluSCSI V1.1, there's a physical debug DIP switch. Under normal operation, this should be in the off position. It can _also_ be set in your zuluscsi.ini file, with debug=1. If debug=1 is defined in the INI file, debug mode will be enabled regardless of the state of the debug DIP switch. Since not all ZuluSCSI platforms have physical DIP switches, there had to be another way to enable debug mode (with the ZuluSCSI Mini/DB25, for example)

The sequential and random seek values you're seeing are expected, _if_ debug mode is enabled. I would also encourage you to try a different SD card, and see if the latency changes in any measurable way.

chris-jh, 2022-06-01:
@antorsae just rename the extension to hda. as aperazbios says it is just a raw image.

chris-jh, 2022-06-01:
@aperezbios the debug is off. Off at the dip switch, and i don't have the debug line in my ini file. So i assume that is off by default.

I have tried different SD Cards before that benchmark test. (different sandisks, clouddisk). Currently using a Clouddisk 2GB. the hard drive file is small. 

Is there an SD card that you can recommend ?

I am also unsure on the disk bench how to do writing benchmarks.

It is not stopping anything from working, it is what i noticed in the difference between Zulu and SCSI2SD and only on certain games.

aperezbios, 2022-08-10:
No response in over two months, so closing.

closed 2022-08-10 by aperezbios