ZuluSCSI/ZuluSCSI-firmware #6: Amiga large drive compatibility issues

thepillock, 2022-05-15:
Tried to install a ZuluSCSI to my Amiga(s).
Small images work fine, 1GB and 4GB do not present any problems. They're using the default 512 block size.
Trying to use an 8GB image, it appears to work (I can partition the "drive" in HDToolbox and then format it) but if I write almost any amount of data to it, even just 100MB or so, when I reboot the drive has lost its information. It feels like the RDB is being corrupted once data is written so far into the disk, the drive name has been changed to something like DH1\o\o\o\o\o\" and it's unreadable.

I've tried changing block size, and using the appropriate filename such as HD20_4096.img

I can supply debug files if needed - let me know if it's more useful when the drive is empty (and therefore working) or once it's corrupted?

Tried on two different ZuluSCSI devices, both with latest firmware (although issue existed on original firmware they shipped with)

thepillock, 2022-05-15:
To provide a few more results of testing:

- I see block sizes of 4096 are not supported, so I've gone back to 512 for everything. Still not handling partitions over 4GB
- I took my SCSI2SD card from my A3000 (a 16GB card divided into partitions of 1GB, 3GB, 5GB and 6GB). Made an image of it, put it on an SD card. Using ZuluSCSI and that image, on exactly the same machine, the 1GB and 3GB partitions are available but 5GB and 6GB are corrupted - and that's using the same Kickstart ROMS which include the SCSI device drivers

aperezbios, 2022-05-15:
Thanks for the report. We'll look in to it, and take action if we can reproduce. I see you're using exFAT. As a reminder for others, any volumes larger than 4GB must reside on an exFAT-formatted filesystem, since the maximum supported file size for FAT32 is 4GB.

PetteriAimonen, 2022-05-16:
@thepillock The image taken from SCSI2SD being corrupted is useful datapoint. It could be useful to have a debug log from when ZuluSCSI tries to read that partition.

PetteriAimonen, 2022-05-16:
@thepillock Actually there was a simple integer overflow issue that caused writes and reads to wrap over to the first 4 GB.

There is a new build for testing here, it also adds support for 4096 and 8192 sector sizes:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/latest

thepillock, 2022-05-16:
I've flashed that firmware and it's looking good on the A3000 - I've been able to write 200MB of data to an 8GB partition and it's still there after a reboot. A sector size of 4096 also seems to work, there's a notable performance increase. Previously, even just 20-30MB of data would corrupt the partition.

I'll stress-test it later by sticking the whole 6GB games library on it, and also try the same firmware on the ZuluSCSI that's in my A4000. 

Really appreciate the quick response!

PetteriAimonen, 2022-05-17:
Thanks for testing!

closed 2022-05-17 by PetteriAimonen

aperezbios, 2022-05-25:
FYI, the fixes described above are included in firmware release 1.0.2, so there's no need to use the "latest". All newly-shipped ZuluSCSI V1.0 (mini) and V1.1 boards are shipping with firmware 1.0.2