ZuluSCSI/ZuluSCSI-firmware #8: Firmware Name

chris-jh, 2022-05-21:
What's the best way to update the bootloader so it can recognise new firmware based on the zulu name and not the old name ?

I have updated to the latest 1.0.2 firmware, but it still looks for the old name for the firmware, so i assume there is a boot loader firmware to update.

PetteriAimonen, 2022-05-21:
We are considering simpler update paths, but for now this will work:

1. Enable USB bootloader by turning DIP SW 4 on.
2. For Windows, download GD32 DFU tool from http://www.gd32mcu.com/en/download?kw=dfu&lan=en . On Linux/Mac, dfu-util can be used.
3. Upload the same firmware .bin file you would normally put on SD card. It contains both the SD bootloader and main application, but the SD bootloader currently doesn't update itself.

eacpereira, 2022-05-21:
I'm having a similar issue where I can't update an Azul Mini running the ZuluSCSI platform from ZuluSCSI firmware 1.0.1 to 1.0.2. Since it doesn't have dipswitches, what would I do to get USB bootloading?

chris-jh, 2022-05-21:
i nearly did the dfu-util, i am on mac and it lists alt0 and alt1 for the zulu and i wasn’t 100% which one

PetteriAimonen, 2022-05-21:
@eacpereira On Azul Mini, there is rectangle marked SW301 near the main CPU. Keeping the two pins of that connected when plugging in USB activates the USB bootloader.

@chris-jh For me `dfu-util --list` shows `alt=0` as `Internal Flash` and `alt=1` as `Option Bytes`. Of these `alt=0` is the correct choice.

chris-jh, 2022-05-21:
`dfu-util -a 0 -D ~/Downloads/ZuluSCSIv1_1_2022-05-17-v1.0.2.bin 
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 28e9:0189
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(10) = dfuERROR, status(10) = Device's firmware is corrupt. It cannot return to run-time (non-DFU) operations
Clearing status
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
dfu-util: Only DfuSe file version 1.1a is supported
dfu-util: (for raw binary download, use the --dfuse-address option)`

Mmm, i must be missing something else

PetteriAimonen, 2022-05-21:
@chris-jh Ah, sorry, indeed dfu-util requires also the address. For me this command works:

     dfu-util --download ZuluSCSIv1_1_2022-05-17-v1.0.2.bin --alt 0 --dfuse-address 0x08000000

chris-jh, 2022-05-22:
Cheers, the worked.

I almost used that address bases on the --list option, but thought i will wait just incase.

Can you update the README to add that line in ?

I shall leave this issue open until @eacpereira has sorted theirs.

aperezbios, 2022-05-22:
@eacpereira All Azul/ZuluSCSI Minis we've shipped to date have a momentary-contact button installed at the SW301 location @PetteriAimonen mentioned. The top of the button is black. It's difficult, but not impossible, to depress this button without removing the Mini from its enclosure, so I would strongly recommend removing the four screws to remove the clamshell plastic housing. Then, all you need to do is hold down the button while you plug in the Mini via USB. The button only needs to be pressed down when the board is initially powered on (via USB), and afterwards its state does not matter, once the microcontroller is in bootloader/DFU mode. This button closes the circuit in exactly the same way as the DIP switch does on ZuluSCSI V1.1.

closed 2022-05-26 by aperezbios