ZuluSCSI/ZuluSCSI-firmware #13: Kurzweil K2000R compatibility

mintaholic, 2022-05-28:
Read accesses cause the K2000R to reboot.
Kurzweil K2000R v3.87J
ZuluSCSI fw v1.02

Log attached, it fails on a Read10 command:
scsi_accel_dma_finishWrite() timeout

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8790384/zululog.txt)

Tried adjusting many of the settings (PhyMode, EnableSCSI2, MaxSyncSpeed) using zuluscsi.ini, with no success.
SCSI bus is terminated on both ends.
Borrowed my friend's SCSI2SD v5.0b this evening and it works perfectly with the K2000R.

aperezbios, 2022-05-31:
Thanks for the report, @mintaholic. Your firmware is downrev, and according to the log, your V1.1 board is at firmware version 1.0.1. It's highly likely that firmware version 1.0.2 fixes this problem, so before going any further, we need to have you upgrade the firmware. 

We released firmware version 1.0.2 fourteen days ago. All boards shipped since the release of the firmware have been shipped with version 1.0.2. Please grab https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v1.0.2/ZuluSCSIv1_1_2022-05-17-v1.0.2.bin and place it on your ZuluSCSI's SD card, without renaming the file. The next time you power on the ZuluSCSI, it should self-update its firmware, which takes under two seconds to complete. During that time, the activity LED on the ZuluSCSI will flash rapidly, then the board will re-set.

mintaholic, 2022-05-31:
@aperezbios I am running 1.0.2, as reported in my original comment and the zululog.txt that I attached.

[1443ms] Platform: ZuluSCSI v1.1
[1443ms] FW Version: 1.0.2 May 17 2022 07:11:21

aperezbios, 2022-05-31:
@mintaholic Apologies, I must've accidentally opened another zululog in error. Two nights of low-quality sleep... Please provide a debug log with PhyMode=1 set, in your zuluscsi.ini. We'd like to see if it fails the same way.

aperezbios, 2022-05-31:
@PetteriAimonen any thoughts on things we might want to try here? Disable DMA write acceleration?

mintaholic, 2022-05-31:
Here's a log with PhyMode=1 in zuluscsi.ini.

[zululog-pio.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8809730/zululog-pio.txt)

chickeneps, 2022-05-31:
Works here, all defaults. When debugging I rarely if ever change parameters, in my experience testing everything, both ZuluSCSI and SCSI2SD never need parameter adjustments. The exceptions are so incredibly rare I don't even remember them.

This with K2000R running 3.54J. I would expect 3.87 to work the same way. I used an image of the Take 6 CD,read the AAH off the SAMPLES/SUSTAINS folder, read it, wrote it to a image I formatted with the K2000 itself. Used a exFAT formatted 64GB Sansdisk Ultra SD card, class 10. Make sure you are powering the ZuluSCSI either from the back or the front USB jack, via wall charger. At what point in the Load does the Kurzweil show an error? Does it actually say "Read10"? Are you erroring reading, or writing?

I should say that in my experience with clients with K2000 and SCSI2SD v5.5, there were some clients that couldn't get it to work, but I sent them another and it worked.

aperezbios, 2022-06-01:
@mintaholic thanks. This is a helpful debug log. The relevant bit is: 
```
[44996ms] WATCHDOG TIMEOUT at PC 0x0801100A LR 0x00000000 attempting bus reset. It's very strange that it works fine for @chickeneps though. 
```

chickeneps, 2022-06-01:
@mintaholic Do you have experience with other SCSI peripherals with this particular K2000? What about another SCSI cable? Are you doing this externally (DB25 jack on the outside) or internally (50-pin IDC internally)? If internally, there was thought that the ribbon cable has to be flipped. Again, I'd like to know more about your actual experience, what the Kurzweil said, did it see your directory contents?

aperezbios, 2022-06-01:
@mintaholic Please reply to your original ZuluSCSI order e-mail, and let us know if you're open to trying a different ZuluSCSI unit. Since we can't reproduce the problem, we need to consider the possibility that the unit you have has some sort of unknown defect.

mintaholic, 2022-06-02:
This is a compatibility issue with K2000 v3.87.

I dug out my old v3.54 EPROMs and swapped them into the K2000R and ZuluSCSI started working.
I have two ZuluSCSI boards, both units work correctly in my other E-mu E4XT Ultra sampler and with K2000 v3.54, and both ZuluSCSI boards fail with K2000 v3.87.

From what I can find online, the release notes for K2000 v3.87 state:
_Disk Mode Enhancements; **faster SCSI read/write times**, improved compatibility with a greater variety of SCSI devices including CD-ROM drives, hard drives, Zip/Jaz drives, 2GB support for SCSI hard drives, the ability to read more Sample CDs (ISO formats), now reads K2500 programs, and supports looped wav files._

K2000 v3.87 works perfectly with a SCSI2SD v5.0b or an old internal Quantum HD swapped in place of the ZuluSCSI.

chickeneps, 2022-06-02:
>This is a compatibility issue with K2000 v3.87. K2000 v3.87 works 
>perfectly with a SCSI2SD v5.0b or an old internal Quantum HD swapped 
>in place of the ZuluSCSI.

Looks like you have it covered - guys, continue checking this.

Garth Hjelte
Sampler User

aperezbios, 2022-06-02:
@mintaholic thanks for taking the time to try the older ROMs. This is very helpful information. Clearly there's some unknown issue here. We're currently waiting on a better logic analyzer, which is necessary to further troubleshoot this in the lab.

mosspa1, 2022-06-05:
I have a couple K2000s, one of which is a 2vx (and a couple K2500Rs and a couple K2661s and a K2600RS) and I am trying to use ZuluSCSI.  I also have both SCSI2SD 5.1s in most of them and a v5.5 "mini" that I use between them.  For the past two days I've been trying to find information about how, exactly, one prepares a SD card for use in the ZuluSCSI devices.  While "write an image file with zero contents" may mean something to a lot of people, I'm not sure how to accomplish that, apparently, simple task.  If somebody could help me past that initial step of getting the media correctly prepared, I'd be happy to test the ZuluSCSI in all of my Kurzweil synths for comparison.

chickeneps, 2022-06-05:
So you already have a ZuluSCSI? I've attached a 2GB image, formatted with a K2000, but empty of contents, zipped up. Try that. Save a copy to use for posterity.

The idea behind the "write an empty file of 2GB" is that it'll come up as a image on ZuluSCSI, but your K2000 will say it's unformatted, because (obviously) it isn't formatted for the K2000. Just format it using the K2000, and you'll be gold. But... take this image and it's already formatted for you.

The company I tech for, SamplerZone, includes lots of "empty starter images" they distribute in their ZuluSCSI products and on their web sites (for free).

Also, if you are using 3.87, read the above issues, as it doesn't seem like they are resolved yet. If you can run 3.54, I'd do that. I can confirm ZuluSCSI works with the typical v3.02 of the K2500's (and I think the K2600's)
[HD40_512 Kurzweil K2000.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8840867/HD40_512.Kurzweil.K2000.zip)
.

mosspa1, 2022-06-05:
Thank you!  Actually, I was contemplating buying another Mini from SamplerZone if I can get the one I have to work.  Also, along with the two K2000s I have (one is a 2vx), I also have a couple K2500RSs, a couple K2661s and a K2600RS.  Having become aware of the apparent incompatibilities with 3.87J on the K2000, one of the things I wanted to do was to look at the other Kurzweils, and since the unit in question was a K2000R, I'd like to see if I can replicate the problem with a keyboard model (which may only make sense if you have been mucking around the insides of Kurzweils for a while).  Again, thank you for "starter image".  That will get me going but it would really be helpful if I could understand what I need to do to create one of my own.  If I wanted to say, have two drives expressed, would I use the same file and give it a name with a different SCSI ID #?... and for three do that twice, etc?  I'll get to trying it right now.

John

aperezbios, 2022-06-05:
@mosspa1 in response to " If I wanted to say, have two drives expressed, would I use the same file and give it a name with a different SCSI ID #?... and for three do that twice, etc? I'll get to trying it right now." yes, that's exactly how it works.

mosspa1, 2022-06-05:
@aprezbios  Thank you!

mosspa1, 2022-06-06:
OK, using the file that chickeneps sent me I tested the Zulu Mini with multiple Kurzweils I currently have access to in my studio.  I'm not going to assume that anybody knows more than just the basics about Kurzweil synths so I'm going to probably be a bit verbose in this description. 

After I downloaded the file, I created a file I called "Empty Image Disks" and copied the file into it.  In the folder I duplicated that original file two times.  I renamed one HD4.img and the other HD5.img, and copioed onto the blank FAT-formatted micro-SD.  I then removed the SD card from my reader and inserted into the Zulu Mini.  The first keyboard I tried it with was a Kurzweil K2vx, that had fully populated SRAM and PRAM.  The operating system was Kurzweil OS v3.87J.  Because it is a 2vx, the setup ROMs for that synth are a bit different than the generic (not vx) ROMs.  AFAIK the actual v3.87J OS ROMs are the same as for any other other Kurzweil K2000 running that version of the operating system.  Because the case of the Zulu Mini was too wide to fit inside the cutouts around the female DB-25 SCSI connector on the Kurzweil, I ended up attaching the Zulu to the Kurzweil using a five foot M-M SCSI DB-25 cable with a gender bender on one end, into which I plugged the Zulu Mini.  When I flipped the power switch on the synth, it booted correctly.  When I selected the "Disk" function and rotated the selector wheel, the display indicated the operations as expected.  When the SCSI IDs 4 and 5 were selected, the display showed that it was accessing the correct Zulu disk.  The synthesizer also has and SCSI2SD V5.1 card mounted internally in the position where an internal HDD would go, and is set to use SCSI IDs 0-3 for the four drives it controls.  After I connected the Zulu, I could see all six 2GB drives.  I then loaded the Kurzweil's 200 block of program locations with a few files from the SCSI2SD.  I then saved them to both of the new Zulu drives.  That worked perfectly and required about as much time as would be required if there were SCSI2SD units on either end.  I then wrote those files from each of the two Zulu disks into Kurzweil program location banks 300 and 400 respectively.  Accessing the programs from the "Program" menu of the Kurzweil revealed that they were fine ajnd that all disk operations had concluded successfully.  I then tried copying files among the various drives and all of those copys executed correctly, as well.

The only other K2000 at my immediate disposal was one that I had rebuilt, pretty much, from scratch.  Even though the shell was a mid-series K2000, and the motherboards came from another K2000 of about the same period, just about everything else that was removable was replaced by me (including the main capacitors).  Since I was building for compatibility, I installed the full compliment of PRAM (obtained from Radoslaw Kwiecien in Poland), the two sets of ROM sound block options (also from Radoslaw).  I installed OS v3.87J ROMS and the K2vs Set Up ROMs (obtained from Synths Plus, FL), and I installed a SCSI2SD v5.1 card (obtained from Inertial Computing) kn the floppy drive slot.  So, technically, this was another K2vx electronically very similar to the "real one".  The Zulu worked flawlessly with this synth as well.

Since the discussion here was about K2000s and especially OS v3.87J.  I thought that I'd go into detail on how I tested that.  I also have a K2500RS, two K2661s, and a K2600RS sitting around the studio.  I tested the Zulu Micro with each of them and there were no problems.

So, my testing was not "exhaustive" but it is clear that the incompatibility (if one really does exist) doesn't involve Kurzweil K2000 OS v3.87J, per se.  I haven't owned a K2000R for over 25 years and I really don't know if the OS ROMs are different between the rack and the keyboard.  They don't appear to list as different part numbers from anybody who sells them.  That the Zulu worked flawlessly with two K2000 synths that used that OS suggests very strongly that the incompatibility problem lies elsewhere.

chickeneps, 2022-06-06:
Perhaps mintaholic can send mosspa1 his image and see if he can read the image?

aperezbios, 2022-06-07:
@mosspa1 thank you for taking the time to test the ZuluSCSI with these machines. We greatly appreciate the effort, and it helps put some concerns to rest.

chickeneps, 2022-06-15:
I just got some 3.87J OS ROM's in my K2000 unit. I tested it with a ZipDrive first, all pass. Then tried it with a ZuluSCSI v1.1, it sees the ID's, but upon a LOAD (on empty formatted images or a verified Kurzweil image) the K2000 crashes and restarts. Same with a attempt to FORMAT. This happens on the Mini v1.0 also. Log attached
[zululog_k2000_387.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8911837/zululog_k2000_387.txt)
.

mosspa1, 2022-06-15:
I find this very odd.  Try this....
Go to ...  https://github.com/ZuluSCSI/EmptyFileCreator-win32  and get the EFIC zip that includes the EFIC.exe (i.e., create blank image file program).
---------------------------

(1) Format the SD as exFAT (I've found this step to be critical).  Thus far, I've tried 8GB, 16GB and 32GB SD and miniSD cards (all worked fine, BTW). 

(2) Run EFIC.  For "Location" choose the "Current Directory" from the Options drop down
(3) Choose the file size in GB.  For the K2000 I chose 2GB, for the K2661s and K2600RS I choose 8GB. (all of these worked fine, BTW)
(4) For the "File Name" dialog I typed in HD5 and WROTE IN ".img" as the extension
(5) To keep things uncomplicated I choose to write one file.
(6) After the write is complete you need to find the file (I think this is one directory parent to where you think it is going to be) the File name gets changed to something that has the name you typed an "underscore" character and, I think, something else in it.  Just change the file name to HDx.img, where x is whatever SCSI drive ID# you are creating.
(7) Write the file to the exFAT formatted SD card
(8) Put the SD card in the Zulu
(9) Attach the ZULU to the K2k SCSI port
(10) Select "Disk" from the K2k left-bank switches
(11) Turn Alpha-dial until you find the ZULU drive
(12)  Format the drive
(13) Write something to the drive

--------------------------------

As I said, for all of my Kurzs this procedure worked flawlessly.  Since I last emailed you, I have moved hundreds of .krz, .k25, and .k26 files between ZULU disks, SCSI2SD disks, and Jaz disks.  Everything worked as expected, including partitioning the 8GB drives to "four 2GB partitions" on the K26xx machines.  I realize that my "directions" may seem a bit overdone, but, they EXACTLY reflect the procedures I followed.  Since I never tried to do things any other way (because the first one I tried worked fine), I never tried to do it differently.  However, my conclusion after doing this (maybe seven times now) is that the ZULU is working exactly as Alex (or whomever the real spokesperson is for Rabbit Hole) says it should with an image file containing only zeros (as is the one created by EFIC).
Here is my zululog after formatting one of the partitioned 8GB drives in a K2661

[0ms] DIPSW3 is ON: Enabling SCSI termination
[59ms] SD card detected, FAT64 volume size: 30427 MB
[60ms] SD MID: 0x03, OID: 0x53 0x44
[60ms] SD Name: SC32G
[60ms] SD Date: 4/2014
[61ms] SD Serial: 0x10B644F8
[61ms] Config file zuluscsi.ini not found, using defaults
[62ms] Active configuration:
[62ms] -- SelectionDelay: 255
[62ms] -- EnableSCSI2 is on
[70ms] Finding HDD images in directory /:
[71ms] -- Opening /HD5.img for id:5 lun:0
[71ms] ---- Image file is contiguous.
[72ms] ---- NOTE: Drive geometry is 63x255=16065 but image size of 16777216 is not divisible.
[73ms] SCSI ID:5 BlockSize:512 Type:0 Quirks:0 ImageSize:8388608kB
[574ms] SCSI PHY operating mode: PIO
[574ms] Initialization complete!
[574ms] Platform: ZuluSCSI v1.0
[574ms] FW Version: 1.0.1 May  6 2022 21:12:58

Let me know how this goes if you try it.
John

chickeneps, 2022-06-15:
It is odd, I agree. I (guess I am) the principal tester for samplers with ZuluSCSI and SCSI2SD, so I do all the checks you suggested. I just don't mention all of it anymore. =)

I tried several blank image (zeroed out) of various sizes and all "crashed" the K2000 3.87. I was using a new out-of-the-sleeve SD card, and I tried another one as well. I tried a SCSI2SD v5.5 and v5.2 unit and both worked (with validated Kurzweil format SD cards). And the same ZuluSCSI units that crashed the K2000 3.87 worked just fine with our K2500, AND I tried the same units on 3.54 before I changed the chips and it worked fine as well.

Having a crash/bad experience with legitimate media etc. is a good thing, as it gives the programmer's goal posts that don't move. However, like you said, the fact that some K2000 3.87 work and some don't doesn't help, so they need to concentrate on our's and mintaholic's to get it figured out. I agree it's weird to have some units that don't work and some that do, beyond our abilities to find out where the apples-and-oranges are. And it has been my experience this happened with SCSI2SD too; some units just didn't work with certain clients.

At least we know that 3.54 is stable with ZuluSCSI, and going back to that isn't that bad of a thing. (Unlike the present issues with the Akai MPC60/3000 and Roland S550, where there is no workaround until it gets fixed.) But I'm confident we'll get all this figured out.

chickeneps, 2022-06-15:
Oh, forgot to mention - I did backgrade the firmware to what you reported, which was Mini 1.0 May 6th. I was using Mini 1.0 May 17. But that didn't help, both gave me the same behavior.

mosspa1, 2022-06-15:
Just to check.... the "K2000" you are using is a K2000 with a keyboard, not a K2000R as is the case with mintaholic's?  If that is the case, then there remains the possibility that there is something different about all the things that make a K2vx a K2vx.  As I said, even in my K2000 that is now a K2vx by nature of adding the K2vx setup ROMS, the fully expanded expansion ROMS as well as full PRAM, is still a K2vx, not a K2000.  I know the setup ROMs shouldn't make a difference, but there aren't a lot of things that have been left to vary.  If one of mine didn't work and the other one did, then there may be something to look at.  It is kind of hard for me to believe that I would end up with two machines, one so much a Frankenstein that it would be unrecognizable as a K2000, that both worked without incident with, now three (counting the mini I just got from Sampler Zone) ZULU implementations.
Just for my information.  What makes an SD card a "validated Kurzweil SD card"?  I've got 8GB SD cards that are old enough to make them relics and I haven't ever had a problem with the SCSI2SD units.  In fact, back when I was experimenting a lot with SCSI in the mid 90s, I was really amazed at how robust SCSI was on K2000s (compared to Macs, for example). So, it would really surprise me that something as small as the amount of variance between any two SD cards would account for a catastrophic failure in a K2000.

John

chickeneps, 2022-06-16:
It's a K2000R. I mean "validated" meaning that I know the Kurzweil disk format intimately and I verified that the image I'm using is a fully checked out image. And I'm guessing there are small mechanical differences in K2000's that are causing the inconsistencies, what they are we may or may not ever know. The thing to do now is to test with the non-functioning ones using logging or other strategies that the ZuluSCSI programmers come up with.

mosspa1, 2022-06-16:
As I said, or at least think I said (I've lost the ability to discern whether I',m talking to a forum or to directly to a person in an email that is going to both), I haven't owned a K2000R in over 25 years and haven't opened one up in at least that amount of time.  Judging by how dissimilar the K2500 boards are to the K2500R boards, I can't even think about how to comment something i am not looking at directly.  From what I do know about K2000 "Calvin" vs "Janus" boards in K2000s with keyboards (is there even a similar distinction for K2000Rs?), is that the jumpers, and what positions they are in a largely a factor in what, if anything, is plugged into the ROM expansion connector.  It's been a while now, but I vaguely remember having to reconfigure jumpers on the K2000 "Calvin" board when I had installed Radoslaw's ROM expansion board.  I also remember needing to "bend up" one of the pins on each of the setup ROMs and jumper them to some other place.  [Note: Pin 31 on both setup ROMs are bent up and jumpered to Pin 3 (leftmost pin) of JP1 that is immediately under the ROMs].  Also, another thing I also remember is that the jumpers in JP1 are arranged differently depending on what setup ROMs are installed.  However, I think that I remember that having something to do with the K2vx ROMS because of the way the programs in slots 600-799 are accessed.  Again, this is a hazy memory and is complicated by the fact that when I last did those mods I probably hadn't "given up" drinking yet :)  I can say one thing about the four ROM sockets (and what is plugged into them) and that is that they will take a LOT OF PHYSICAL ABUSE.  Outside of breaking the leads off the chips, I don't think I've ever done anything that permanently altered their function.  My vision isn't that of a teenager any longer and I know that I manage to put the chips in the sockets wrong from time to time.  On the K2000 keyboards, the spacing between the four ROM chips even makes it hard to determine whether or not all of the pins are actually in the socket holes without using a curved dental mirror.  However, never did I make a mistake in plugging in the ROM chips that I couldn't "fix" by getting them plugged in correctly.  So, what I guess I'm saying is that if you eventually get all four chips correctly placed in their respective sockets you should be OK!  

For what it's worth.  I just pulled out the K2000/K2000R Service manual (dated 1992, so I don't think it reflected the Calvin/Janus difference, yet).  Anyway, the schematic diagram for the logic board (with the ROMs) is apparently the same for the K2000 and K2000R (so, I'm guessing that the two main boards are layered in the K2000R as opposed to bridged horizontally in the K2000).  So, what that diagram shows is that JP1 is the same in either version.  I just don't remember what the default jumper condition for JP1 is.  However, this is what I would try first....

Locate the four ROM chips  (Lo and Hi for both the SETUP and ENGINE) and set them pins down on a piece of aluminum foil (or plug them into anti-static foam if you have a sheet available).  I am giving these instructions assuming that you are looking down on the logic board and that the notch in each chip is facing your front side as you look down and that the MIDI and SCSI connectors are facing away from you (i.e., out the back).  In this orientation, the leftmost chip is the SETUP LO ROM.  The one to the right of that is the ENGINE LO (e.g., 3.54J Lo) ROM, to the right of that is the SETUP HI ROM, and the rightmost chip is the ENGINE HI ROM.  Just remember, the left/right orientations are dependent on where the notch in the ROM chips are located (either facing you, or facing away from you).  Again, I'm not too sure how much difference this makes when you open a K2000R vs. a K2000, but I know that when I open a K2000, I generally start with the keyboard (underneath) facing me and swing the cover up from the front to the back.  I think that if you do that it reverses the way the chips are positioned when looking down from the way they appear in the schematics.  If you work in reference to the orientation of the notches on the chips, it really doesn't matter.

How you insert the chips in the sockets is, to a great extent, dependent on how you are most comfortable "insert-bending" long DIPs.  For me, I like to make sure that the right side pins of the package are all seated first before I coax the left side pins into their respective holes.  Then starting from the right most chip and working left makes the most sense.  Anyway, as I suggest above, after you have each chip plugged into its socket, use a dental mirror to verify that all of the pins are going into holes and aren't bent under the chip.  After you've done this for all four chips, that should resolve any issues related to placement or connections.

For the jumpers, there are only three possible locations, and I'm pretty sure that only two of them are jumpered.  So, taking the pins two at a time there are only three configurations possible.  1-->3; 1--2; and 2-->3.  So, if you don;t have any reason for choosing one possibility over another, just try each of them.   **However, since these things come from the factory with a 2-adjacent-pin jumper on either one side or the other, the only real potential configurations are  (the two leftmost pins, or the two rightmost pins)**  [Just remember that the pins of JP1 are numbered oddly.  Orientated with the ROMs in front of them (away from you), the leftmost pin of JP1 is Pin #3, the center pin is Pin #1 and the rightmost pin is Pin#2 should you be reading any instructions that refer to the pins of JP1 by name].

That should work to get you back up and running, at least.

dfvincent, 2022-08-15:
Exact same issue as mintaholic on my K2000 Keyboard:
Read accesses cause the K2000 to reboot.
Kurzweil K2000R v3.87J
ZuluSCSI fw v1.04
Please keep us updated on progress of getting this issue resolved.
Thank you!

dfvincent, 2022-08-15:
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9339688/zululog.txt)
Attached is my zululog.txt file showing similar issues as mintaholic
Zellow status light stays lit during reboot and eventually goes off to allow disk access.  I tried to load 3 times from SCSI0-2

dfvincent, 2022-08-15:
One other issue, I am unable to update my firmware by putting either:
ZuluSCSI.bin or
AzulSCSI.bin files on the SD card root.  The files are simply ignored and the firmware version remains the same.

aperezbios, 2022-08-27:
@dfvincent this problem has been fixed in firmware 1.0.7. Please upgrade and let us know if you're still having trouble. I'm closing the issue as resolved.

closed 2022-08-27 by aperezbios

francoisjacques, 2022-09-05:
I confirm with the new firmware, it's a lot better. I wouldn't call it stable but I've managed to load a preset file while before it was consistenly rebooting the K2.

aperezbios, 2023-01-30:
@francoisjacques what was the verdict here? We never heard back from you.

mosspa1, 2023-01-30:
I'm not sure that the "what was the verdict here? We never heard back from you" was directed exclusively to me or not, but I never had a K2000R, and every K2vx running 3.87J has never had an issue with the Zulu mini (from whatever firmware it was running when I purchased it.  I'm currently in the process of selling a K2661, and I just checked that out as well the other day and the Zulu mini (for which I never updated the firmware) still works as well as any SCSI2SD implementation I have tried.  Now that I have been redirected here, I'll update the firmware to whatever is current and see if that makes any difference (although, unless something gets worse, I don't know how I will tell)

francoisjacques, 2023-01-31:
@chickeneps I think this should be closed. I've recently updated the firmware and the it looked stable. I've loaded a full bank at full throttle without any issue.