ZuluSCSI/ZuluSCSI-firmware #14: Are ZIP files supposed to be detected?

chickeneps, 2022-06-01:
If I have these two files in the specified DIR folder

HD40_512 Ensoniq ASR-10 OS.img.zip
HD40_512 Ensoniq ASR-10 OS.img

ZuluSCSI (5-17 release) detects the Zip file instead. Should ZuluSCSI not be detecting that at all?

[zululog ZIP FILE ERROR.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/8818378/zululog.ZIP.FILE.ERROR.txt)

aperezbios, 2022-07-27:
@chickeneps seems like a legitimate bug to me. Thanks for reporting, we'll get this fixed.

aperezbios, 2022-08-10:
This was fixed in firmware release 1.0.3: https://github.com/ZuluSCSI/ZuluSCSI-firmware/commit/a5735890db266d92274ba3c4336ce8334be3862d#diff-da47346a7cd6565238684284cbfc01ce2aa75101c80574c7cce8f14955c1e60b

closed 2022-08-10 by aperezbios