ZuluSCSI/ZuluSCSI-firmware #16: .ini Drive Configuration Options and .ini Filename Issue

peterkomorowski, 2022-06-12:
I am having trouble getting the ZuluSCSI to be recognized with an Atari Mega STE.    There is a 1GB drive size limit and it appears that parity is not supported.   

QUESTION 1:

I'm trying to clone this stock drive that came with the system.  How can I specify custom drive parameters in the .ini file?   Some of these parameters are documented in the sample zuluscsi.ini file, but not all.

SEAGATE ST157N
Cylinders            615
Heads                 6
Sector/track       26
Bytes/Sector      512
Capacity form/unform    49/   57 MB 
Seek time / track       40.0/ 8.0 ms
Controller                   SCSI1 SINGLE-ENDED 

What I got so far for .ini:
EnableSCSI2 = 0
Vendor = "SEAGATE"
Product = "ST157N"
#Version = "1.0"
#Serial = "0123456789ABCDEF"
Type = 0     # 0: Fixed, 1: Removable, 2: Optical, 3: Floppy, 4: Mag-optical, 5: Tape
#TypeModifier = 0  # Affects only INQUIRY response
SectorsPerTrack = 26
HeadsPerCylinder = 6  ??? IS THIS CORRECT ???
#RightAlignStrings = 0 # Right-align SCSI vendor / product strings, defaults on if Quirks = 1

QUESTION 2:

Using updated firmware 1.0.2 and the zuluscsi.ini is not loaded.  I had to rename to azulscsi.ini.  Is this a known issue and will future firmware work with azulscsi.ini?

aperezbios, 2022-06-12:
If you've dumped a cloned image from a drive on the ZuluSCSI, there's probably no reason or need to replicate the exact geometry settings of the source drive. I'm not terribly familiar with TOS, but the default values that the ZuluSCSI exposes should be more than sufficient, in most scenarios. 

How have you concluded that parity isn't supported, exactly? Is it because there isn't a config setting for it? The SCSI command handling code in ZuluSCSi comes straight from SCSI2SD V6's codebase, and the SCSI2SD V6 code always emits parity. It doesn't care if the host controller does or does not. 

The .ini file-naming issue isn't something we've seen before, nor is it something I can even see as being possible, given that it works fine for hundreds of others, and they're all using the exact same 1.0.2 firmware, built from the same source. The word/string "azul" doesn't exist in the compiled version of the 1.0.2 firmware.

Have you tried without any ini file at all? It sounds like a simpler config would be best to start with, at least for troubleshooting. Try renaming the ini file to disable it altogether. Once you do, please supply a zululog.txt file so we can see what the firmware is doing when it initializes.

peterkomorowski, 2022-06-12:
Found the issue with zulucsi.ini not being read.  The firmware wasn't automatically updating by dropping the .bin file on the SD card.  I had to manually update to 1.0.2 using USB cable flash method.

The firmware update seems to have resolved some of the issues.  I'm still getting errors that host is trying to write data outside of the image size.  

Any suggestions on that?

See log file below.

[0ms] DIPSW3 is ON: Enabling SCSI termination
[1ms] Optional GreenPAK detected, loading firmware
[53ms] GreenPAK firmware successfully loaded
[166ms] SD card detected, FAT32 volume size: 29801 MB
[167ms] SD MID: 0x89, OID: 0x03 0x03
[167ms] SD Name: NCard
[167ms] SD Date: 1/2016
[167ms] SD Serial: 0x12293019
[169ms] Reading configuration from zuluscsi.ini
[169ms] Active configuration:
[175ms] -- SelectionDelay: 255
[1356ms] Finding HDD images in directory /:
[1359ms] -- Opening /HD0.img for id:0 lun:0
[1619ms] ---- Image file is contiguous.
[1619ms] ---- NOTE: Drive geometry is 63x255=16065 but image size of 1638400 is not divisible.
[1621ms] -- Opening /HD1.img for id:1 lun:0
[1862ms] ---- Image file is contiguous.
[1862ms] ---- NOTE: Drive geometry is 63x255=16065 but image size of 1638400 is not divisible.
[1913ms] SCSI ID:0 BlockSize:512 Type:0 Quirks:0 ImageSize:819200kB
[1914ms] SCSI ID:1 BlockSize:512 Type:0 Quirks:0 ImageSize:819200kB
[2417ms] SCSI PHY operating mode: GREENPAK_DMA
[2417ms] Initialization complete!
[2417ms] Platform: ZuluSCSI v1.1
[2418ms] FW Version: 1.0.2 May 17 2022 07:11:21
[1271737ms] WARNING: Host attempted write at sector 2097151+1, exceeding image size 1638400 sectors (512B/sector)
[1271769ms] WARNING: Host attempted write at sector 1835006+1, exceeding image size 1638400 sectors (512B/sector)
[1271772ms] WARNING: Host attempted write at sector 1703934+1, exceeding image size 1638400 sectors (512B/sector)
[1271779ms] WARNING: Host attempted write at sector 1671166+1, exceeding image size 1638400 sectors (512B/sector)
[1271781ms] WARNING: Host attempted write at sector 1654782+1, exceeding image size 1638400 sectors (512B/sector)
[1271784ms] WARNING: Host attempted write at sector 1646590+1, exceeding image size 1638400 sectors (512B/sector)
[1271787ms] WARNING: Host attempted write at sector 1642494+1, exceeding image size 1638400 sectors (512B/sector)
[1271789ms] WARNING: Host attempted write at sector 1640446+1, exceeding image size 1638400 sectors (512B/sector)
[1271792ms] WARNING: Host attempted write at sector 1639422+1, exceeding image size 1638400 sectors (512B/sector)
[1271794ms] WARNING: Host attempted write at sector 1638910+1, exceeding image size 1638400 sectors (512B/sector)
[1271797ms] WARNING: Host attempted write at sector 1638654+1, exceeding image size 1638400 sectors (512B/sector)
[1271799ms] WARNING: Host attempted write at sector 1638526+1, exceeding image size 1638400 sectors (512B/sector)
[1271802ms] WARNING: Host attempted write at sector 1638462+1, exceeding image size 1638400 sectors (512B/sector)
[1271804ms] WARNING: Host attempted write at sector 1638430+1, exceeding image size 1638400 sectors (512B/sector)
[1271807ms] WARNING: Host attempted write at sector 1638414+1, exceeding image size 1638400 sectors (512B/sector)
[1271809ms] WARNING: Host attempted write at sector 1638406+1, exceeding image size 1638400 sectors (512B/sector)
[1271812ms] WARNING: Host attempted write at sector 1638402+1, exceeding image size 1638400 sectors (512B/sector)
[1271814ms] WARNING: Host attempted write at sector 1638400+1, exceeding image size 1638400 sectors (512B/sector)

aperezbios, 2022-08-25:
@peterkomorowski this was a bug that was fixed a few revisions ago. Please update to 1.0.7 (or later), and then re-open this issue if it's still a problem for you. Closing as fixed for now.

closed 2022-08-25 by aperezbios