ZuluSCSI/ZuluSCSI-firmware #22: Firmware flash documentation update needed for original AzulSCSI cards

garthbeagle, 2022-07-30:
If you have (apparently) an original AzluSCSI board, I assume with some original boot-loader FW (?) you cannot flash the firmware onto the card unless you name the firmware "AzulSCSI.bin".

The documentation here:
https://github.com/ZuluSCSI/ZuluSCSI-firmware#programming--bootloader

does not point this out, it just notes it needs to be named "ZuluSCSI.bin"

If it matters, my board is the AzulSCSI 1.1 board

dfvincent, 2022-08-15:
I am unable to update my firmware by putting either:
ZuluSCSI.bin or
AzulSCSI.bin files on the SD card root.
The file is simply ignored and the firmware version remains the same.

aperezbios, 2022-08-15:
@dfvincent what's the current firmware version on your ZuluSCSI, and which operating system are you using on your normal desktop? You may need to attempt to upgrade via the USB port.

aperezbios, 2022-08-15:
@dfvincent it looks like you have a ZuluSCSI Mini. Is that correct? If so, you need to use the V1.0 firmware. You're trying with V1.1, which is only for hardware version 1.1. Assuming you do have a ZuluSCSI Mini, please try with https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v1.0.4/ZuluSCSIv1_1_2022-08-11-v1.0.4.bin, which is the current version.

dfvincent, 2022-08-16:
I have an External ZuluSCSI v1.1 SCSI SD Drive.
I manually updated to fw 1.0.4 using the correct version.  It appears to be a bootloader issue.
In upgraded on a windows 10 pc.

dfvincent, 2022-08-16:
I updated fw manually to 1.0.4 by using:
(1) DFU Driver: GD32DfuDrivers_V3.6.6.6167 for Windows
(2) Put ZuluSCSIv1.1 into boot DFU mode (switched BOOT switch on board to on)
(3) Attached ZuluSCSI to my Windows 10 pc via USB cable
(4) Used GD32AllInOneProgrammer_V2.0.3.13854 to flash 1.0.4 firmware to ZuluSCSI
(5) Removed ZuluSCSI from USB port and switched Boot switch back to OFF position)
(6) Plugged in ZuluSCSI, and now reporting version as 1.0.4

dfvincent, 2022-08-16:
I purchased my external ZuluSCSI v1.1 from SamplerZone.com
The renaming of ZuluSCSI.bin to AzulSCSI.bin and saving to SD card root, did NOT work on my device.
I'm including their recommendations directly below:
*******************************************************
From: SamplerZone.com
Date: Sun, Aug 14, 2022 at 10:33 PM
Subject: Re: Order #211xxx

It looks like we had a series of ZuluSCSI boards that had the latest
(at the time) firmware on it (ZuluSCSI 1.0.2), but the bootloader was
still AzulSCSI (the previous name of the product).

We were not informed that we needed to update the bootloaders.

This only applies to the set of 20-30 AzulSCSI we had in stock at the
time of the name change. Apparently you got one of them.

A workaround is to just rename the firmware file with 'Azul' instead
of 'Zulu'. And that workaround will be in force until you upgrade the
bootloader.

We will have instructions how to upgrade the bootloader soon.

In the meantime, ask the people on the GITHUB
https://github.com/ZuluSCSI/ZuluSCSI-firmware
About flashing a new BOOTLOADER, not firmware.

dfvincent, 2022-08-16:
I was able to get my ZuluSCSI v1.1 to read the firmware on SD root by using filename:
zuluSCSIv1_1_2022-08-11-v1.0.4.bin
(lower case z, NOT Upper)

Forum has the actual firmwares posted as:
[ZuluSCSIv1_1_2022-08-11-v1.0.4.bin](https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/v1.0.4/ZuluSCSIv1_1_2022-08-11-v1.0.4.bin)
The first capital Z will NOT work.
first "z" MUST be lower case.
@dfvincent

PetteriAimonen, 2022-08-17:
@dfvincent That is interesting. The code does not care about upper / lower case of the filename. So it seems renaming the file has changed something else, possibly there is some corruption of the FAT filesystem.

summitmike, 2022-08-18:
PetteriAimonen:
I could not update by dropping the .bin onto sd card.
So,  
I downloaded the Windows Dfu driver from the link you provided in readme
The current link you provide is just to the Windows drivers:
GD32 Dfu Drivers:
The drivers for the GD32 Dfu device
Link: https://www.gd32mcu.com/en/download?kw=dfu&lan=en

I found the Windows programmer from here:
GD32 All-In-One Programmer:
This programming software based on the bootloader in the chip, the communication interface supports UART and USB, and supports the basic operations of programming / erasing / reading MCU flash and ob.
Link: https://www.gd32mcu.com/en/download?kw=GD32+All-In-One+Programmer&lan=en

Was able to do the lastest firmware flash.
Thought the windows programmer link might be helpful if someone is more comfortable using a gui

wohali, 2022-08-19:
I'm also unable to upgrade past the original AzulSCSI v1.1 hardware -> ZuluSCSI firmware using any of the versions posted here (1.0.2 thru the 1.0.5 beta) by copying them onto a FAT32-formatted, single-primary-partition SD card, named either `AzulSCSI....bin` or `ZuluSCSI....bin`, or even `zuluSCSI....bin`.

The board was previously flashed with the `AzulSCSIv1_1_2022-05-06-v1.0.1-ZULU.bin` file (provided by email) to rename it:

```
[1690ms] Platform: ZuluSCSI v1.1
[1691ms] FW Version: 1.0.1 May  6 2022 21:09:11
```

Fortunately `dfu-util` does see the device, and I was able to recover this way. You may want to add the correct invocation (for Linux and presumably macOS) to your docs:

```
sudo dfu-util -p <your-usb-port-here> -a 0 --dfuse-address 0x08000000 -D ZuluSCSIv1_1_2022-08-11-v1.0.4.bin 
```

and watching for the `File downloaded successfully` notification. (Re-run if the download wasn't successful.) Then, after toggling SW4 off and checking `zululog.txt`, I see:

```
[1314ms] Platform: ZuluSCSI v1.1
[1315ms] FW Version: 1.0.4-release Aug 11 2022 20:34:24
```

I was then able to try out the 1.0.5 beta firmware by just copying the `.bin` file direct to the SD card, no renaming required.

A suggestion: it might also be good to build the `.dfu` files as part of the build process, in addition to the `.bin`s, and add them to the release assets.

aperezbios, 2022-10-20:
> A suggestion: it might also be good to build the .dfu files as part of the build process, in addition to the .bins, and add them to the release assets.

Thanks for the feedback. Agreed, we've planned on doing this.

closed 2022-10-20 by aperezbios