ZuluSCSI/ZuluSCSI-firmware #29: Bootloader issue

dfvincent, 2022-08-16:
I am unable to update by placing ZuluSCSI.bin (or AzulSCSI.bin) file on SD root.
My bootloader appears to be the issue.

How do I flash a new BOOTLOADER, not firmware,
to solve this issue?

aperezbios, 2022-08-16:
When you flash firmware via USB/DFU, which you have, the bootloader is replaced. This isn't a bug.

closed 2022-08-16 by aperezbios

dfvincent, 2022-08-16:
But, my issue is that I can't update my firmware using the drop ZuluSCSI.bin onto SD card method.
It is STILL not working even after manually updating to fw 1.0.4 so something is off with the bootloader.

dfvincent, 2022-08-16:
You are 100% correct.
The fw update filename need to be:
zulu*.bin

I was trying Zulu*.bin, which did not work.