ZuluSCSI/ZuluSCSI-firmware #33: "Illegal instruction" on Mac Plus with Apple's SCSI drivers

fmeschia, 2022-08-17:
When booting a Mac Plus with a drive image that uses Apple's SCSI drivers (such as https://www.savagetaylor.com/wp-content/uploads/68k_Macintosh/Bootdisks/608_2GB_volume.zip), the Mac boots but then any action (it can be reliably replicated by choosing "About the Finder..." results in a "bomb" alert with "Illegal Instruction".
The problem doesn't happen with SCSI2SD v5.5, from which the ZuluSCSI firmware is derived.
The problem can be avoided by installing LaCie SCSI drivers and disabling "blind transfers", but it would be nice to offer compatibility with Apple's drivers.
Replicated with ZuluSCSI mini with firmware 1.0.2, 1.0.3, 1.0.4, 1.0.5.

aperezbios, 2022-08-17:
@fmeschia we're close to releasing a firmware update that will fix (62178) this known bug. It should happen in the next 24 hours. Until then, if you could test the latest development build, which incorporates the fix, and provide feedback, that'd be much appreciated: https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/latest/ZuluSCSIv1_1_2022-08-17_a1c1100.bin

fmeschia, 2022-08-17:
I tried with https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/latest/ZuluSCSIv1_0_2022-08-17_a1c1100.bin since my hardware is v1.0, but the "Illegal Instruction" problem still exists with this version.

aperezbios, 2022-08-17:
@fmeschia thanks. What's the model of Macintosh you're experiencing this with?

fmeschia, 2022-08-17:
It's a Macintosh Plus.

aperezbios, 2022-08-17:
@fmeschia Thanks. Can you test with PrefetchBytes=0 set in the [SCSI] section of your zuluscsi.ini, and see if that makes any difference?

fmeschia, 2022-08-17:
I just tried. No difference, unfortunately.

aperezbios, 2022-08-17:
Do you experience the same problem if you initialize your own volume on the Mac Plus, using HD SC Setup?

fmeschia, 2022-08-17:
Yes, same identical problem

aperezbios, 2022-08-25:
@fmeschia Please update to firmware version 1.0.7, which I just released, and let me know whether or not this resolves this issue for you. Thanks!

aperezbios, 2022-08-27:
@fmeschia thanks for the report. Since it's been a couple of days, and we've not heard back, I'm going to close this as complete, but please feel free to re-open it if you encounter this specific issue again.

closed 2022-08-27 by aperezbios

fmeschia, 2022-08-27:
Yes it works perfectly. Thanks again, and sorry for the delayed reply.

MasonAkershoek, 2022-09-13:
@fmeschia Hey, I am also using the zulu scsi with my mac plus and i was wondering if you could share your ini file with me?

fmeschia, 2022-09-14:
Sure, I'll do it later tonight

fmeschia, 2022-09-16:
Here are the settings from my zuluscsi.ini file:

`[SCSI]
# Settings that apply to all devices
Debug = 0   # Same effect as DIPSW2, enables verbose log messages
SelectionDelay = 0   # Millisecond delay after selection, 255 = automatic, 0 = no delay
PhyMode = 0   # 0: Best available  1: PIO  2: DMA_TIMER  3: GREENPAK_PIO   4: GREENPAK_DMA
PrefetchBytes = 8192
#Dir = "/"   # Optionally look for image files in subdirectory
#Dir2 = "/images"  # Multiple directories can be specified Dir1...Dir9

# Settings that can be needed for compatibility with some hosts
Quirks = 0   # 0: Standard, 1: Apple, 2: OMTI, 4: Xebec, 8: VMS
EnableUnitAttention = 0 # Post UNIT_ATTENTION status on power-on or SD card hotplug
EnableSCSI2 = 0 # Enable faster speeds of SCSI2
EnableSelLatch = 0 # For Philips P2000C and other devices that release SEL signal before BSY
MapLunsToIDs = 0 # For Philips P2000C simulate multiple LUNs
MaxSyncSpeed = 0 # Set to 5 or 10 to enable synchronous SCSI mode, 0 to disable`

MasonAkershoek, 2022-09-19:
@fmeschia how did you set up your hard disk? i keep having trouble writing to my drive without every thing locking up.