ZuluSCSI/ZuluSCSI-firmware #39: Please disable whole card pass-through when raw disk image is specified

rokuyama, 2022-08-21:
For `v1.0.6RC2`, raw pass-through for the whole SD card is enabled when no disk image file is present, but raw disk image is explicitly specified in `zuluscsi.ini`, e.g.:

```
(snip)
[SCSI0]
IMG0 = RAW:2097152:35651583
(snip)
```

For this example, the whole card is recognized as `SCSI1` if no image file is present. As a result, sectors between `2097152` to `35651583` can be accessed both as `SCSI0` and `SCSI1`. This is dangerous behavior.

Therefore, I'd like to propose to disable the whole card pass-through when a raw disk image is explicitly specified in `zuluscsi.ini`, as done when a image file is present.

PetteriAimonen, 2022-08-21:
This is a bug indeed. I will take a look at it next week.

As a workaround until this is fixed, adding a small dummy image file will disable the automatic fallback.

rokuyama, 2022-08-21:
@PetteriAimonen Thank you for your attention!

> As a workaround until this is fixed, adding a small dummy image file will disable the automatic fallback.

Thanks, this actually works around the problem.

aperezbios, 2022-08-21:
@rokuyama thanks for the report, we will address this promptly and release new firmware, hopefully within a day or two. Since effectively all of our existing customer base is using image-based volumes, the severity of this bug is not as bad as it would otherwise be.

rokuyama, 2022-08-21:
@aperezbios Thank you always for your help! Please take your time. This should not be urgent. Also, I will be out during Tuesday to Sunday. I'm sorry in case I cannot test new firmware timely.

aperezbios, 2022-08-25:
@rokuyama This is fixed with the release of [firmware version 1.0.7](https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/v1.0.7). If you have any further issues once upgrading, let us know.

closed 2022-08-25 by aperezbios

rokuyama, 2022-08-28:
I've confirmed that it has been fixed for firmware `v1.0.7`. Thank you very much for your kind support!

aperezbios, 2022-08-29:
Great, thanks for the original report, @rokuyama  :)