ZuluSCSI/ZuluSCSI-firmware #49: Bus Error on Macintosh SE/30 when using System 7.5.3

aperezbios, 2022-09-02:
A customer reports the following issue with their SE/30:

> As was foretold, I am having some issues with the Zulu in my SE/30. Here's what I have been able to establish in my limited morning tinker time.
> 
> Mac OS 7.0.1 - Works well, boots normally and behaves as expected.
> Mac OS 7.5.3 - Does not boot, recognises valid OS but throws "Bus Error" before loading anything
> 
> The weird part is that both those systems are on the same partition. I had suspected a driver issue but I'm not sure how that correlates to the two systems behaving differently.

I've asked them to provide a debug log.

MrPABCo, 2022-09-04:
I'm the user with this issue, here are my logs and my config file. 

[zululog SE30-753.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9484697/zululog.SE30-753.txt)
[zuluscsi SE30-753.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9484699/zuluscsi.SE30-753.txt)

MrPABCo, 2022-09-04:
I just updated to 7.5.5 and the issue is no longer present. I am not sure what this signifies but figured it would be useful to share.

PetteriAimonen, 2022-09-05:
Hmm, nothing immediately apparent in the debug logs.
One possibility would be some corruption in the system files, if the image is not known good to boot on some other setup.

Considering you have it working with 7.5.5, are you still interested in troubleshooting the 7.5.3 problem further?

MrPABCo, 2022-09-12:
This issue was resolved by using a fourth different installer for 7.5.3. I somehow managed to get three different damaged installers.

aperezbios, 2022-09-12:
Great, thanks for the update, @MrPABCo. Can you link to the installer/ISO that did work?

closed 2022-09-12 by aperezbios