ZuluSCSI/ZuluSCSI-firmware #59: Probloms installing onto zuluscsi on macintosh plus

MasonAkershoek, 2022-09-11:
I have a macintosh plus paired with a zuluscsi, the apple HD SC software can initialise the disk just fine. but when I go to install version 5 of the system software it gets to a certain point and just halts any suggestions.

[zuluscsi.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9543055/zuluscsi.txt)
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9543056/zululog.txt)

MasonAkershoek, 2022-09-11:
my real question is how to implement Respond to short SCSI selection pulses in the zuluscsi.ini file

PetteriAimonen, 2022-09-12:
That setting is `EnableSelLatch = 1`

MasonAkershoek, 2022-09-12:
Thanks, any idea why I can't write to the zuluscsi without the instalation halting?

aperezbios, 2022-09-12:
@MasonAkershoek, can you try doing the install again, but with debug=1 set in the [SCSI] section of your zuluscsi.ini file?

@PetteriAimonen there's a watchdog timeout occurring:

[234379ms] WATCHDOG TIMEOUT at PC 0x08012578 LR 0x00000008 attempting bus reset

PetteriAimonen, 2022-09-12:
That watchdog timeout is in the write routine. It seems that the mac just stops sending data in the middle of a write operation.

MasonAkershoek, 2022-09-13:
Thank you. you have been very helpful so far. My only problem not is when i go to install any software with an installer. the installer will begin writing data to the drive and i can see the activity light flashing as it should to indicate drive activity and then all of a sudden the light stays on without flashing for about 5 seconds then turns off and there is no more noticeably activity with the drive or the mac. I updated the firmware and that seemed to sort of help but i relay feel like iv hit a brick wall. i dont know if there is some setting i should have changed or something else. what is strange is i can use the apple file translate software to copy things from disks without a problem.

aperezbios, 2022-09-13:
@MasonAkershoek we need another log file, as requested above, with debug enabled, where you encounter the described bug. Without that, it will be nearly impossible for us to dig deeper.

MasonAkershoek, 2022-09-13:
Ok, I tried installing something on the drive and here is the zululog file and my current ini file. it did just what it did before, it began to write then the light stayed on for a few seconds and then turned off.
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9561496/zululog.txt)
[zuluscsi.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9561497/zuluscsi.txt)

aperezbios, 2022-09-14:
@MasonAkershoek thanks. There's nothing obviously wrong in the log. Can you try setting PhyMode=2 in your zuluscsi.ini and see if that results in any change in behavior? If not, you can also try with PhyMode=1 as well.

aperezbios, 2022-09-14:
@MasonAkershoek  one other thing, the ini file you sent shows SCSI2 mode is enabled. On the Macintosh Plus, it should absolutely NOT be enabled, and it could be the reason why you're seeing this lockup. Before you try the above suggestion regarding setting PhyMode, please set EnableSCSI2 = 0

MasonAkershoek, 2022-09-15:
Alright here is the zululog after the changes and my ini file. i did notice I a small difference and I was able to install oregon trail on the disk however when I tried again the same problem began happening.
[zululog.txt](https://github
[zuluscsi.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9571711/zuluscsi.txt)
.com/ZuluSCSI/ZuluSCSI-firmware/files/9571709/zululog.txt)

MasonAkershoek, 2022-09-15:
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9571714/zululog.txt)
[zuluscsi.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9571715/zuluscsi.txt)

PetteriAimonen, 2022-09-15:
Hmm, there is some weird log corruption going on.
If you have another SD card available, it could be worth a try.

aperezbios, 2022-09-15:
@MasonAkershoek if you could also share with us exactly which SD card you're using (photos would be helpful) I'd appreciate it, as this might help us understand what's going on here.

MasonAkershoek, 2022-09-16:
[image]
Here is the image of the two sd cards iv been trying. they both act up in the same way

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9579864/zululog.txt)
[zuluscsi.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9579866/zuluscsi.txt)
Here is the log and ini files from the onn drive

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9579867/zululog.txt)
[zuluscsi.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9579870/zuluscsi.txt)
Here is the log and in files from the other drive

I tried installing oregon trail on both sd cards and both failed the same way.
Also I have working operating systems working on both drives so reading clearly isn't a problem but for some reason writing kills it almost instantly.

aperezbios, 2022-09-16:
@PetteriAimonen any insights here?

PetteriAimonen, 2022-09-16:
I have no idea.

There is no log corruption in the latest files, so that was probably just some fluke.

MasonAkershoek, 2022-09-16:
Ok so I was reading on some forums that the mac plus prefers disks that are initiated with a 3:1 interweave and I ended up trying that i I got stuff to start installing. And I ended up installing system 6 onto one of the drives on the Zuluscsi. Thank you guys for all your help if I have any more issues I will open a new discussion.

closed 2022-09-16 by aperezbios

MasonAkershoek, 2022-09-16:
No the whole time I’ve been using system 6.0.8 but I had downloaded a pre made drive image from some guy because I couldn’t get my system disks to install. However the drive image i had clearly wasn’t using the 3:1 interweave so that’s why nothing would install on the drive

aperezbios, 2023-04-01:
@MasonAkershoek 

Did you ever get this sorted? I'm able to boot my Mac Plus fine with the following ini file paramters:

```
[SCSI]
EnableSCSI2=0
EnableSelLatch=1
SelectionDelay=0
```

MasonAkershoek, 2023-04-10:
My Mac Plus boots fine last time I tried however my main issue was that I was unable to install anything from a disk onto the zulu. I had to use mini vMac to open an image of a bootable drive, install it with mini vmac then put it on the zulu and boot it.

aperezbios, 2023-04-10:
@MasonAkershoek thanks for the clarification. Can you describe exactly what you mean when you say "from a disk" on to the ZuluSCSI? Are you attempting to copy files from another SCSI device, or another SCSI disk image that's on the ZuluSCSI itself, or something else? Be as detailed as you can be in your explanation, so someone who is unfamiliar with the specifics of your issue can follow your directions and replicate the issue.

MasonAkershoek, 2023-04-10:
I hope that is more detailed if not let me know and I will try again. I haven't been home in a while and that's where my Mac is.