ZuluSCSI/ZuluSCSI-firmware #61: RP2040 showing similar issues when the GD32 v1.1 board was not working with samplers

morio, 2022-09-14:
The Kurzeweil K2000R is hit and miss with the ZuluSCSI RP2040 when formatting the drive. When it does not work properly it stalls out exactly and the same write10 offset (32) as when the ZuluSCSI v1.1 failed. There is also an 

> [20546ms] WARNING: Host attempted read at sector 2097151+256, exceeding image size 514080 sectors (512B/sector)

when it fails to format properly in the log file:
[zululog-newboard-failed-kurzweil-example.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9568688/zululog-newboard-failed-kurzweil-example.txt)

As for the MPC60 the log file does not show any SCSI transactions with debug mode on:
[zululog-new-board-failed-mpc60.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9568689/zululog-new-board-failed-mpc60.txt)

But unlike when ZuluSCSI v1.1 didn't work, the RP2040 doesn't hang and boots to the main screen.

I'll add logic analyzer data for the MPC60 once I get it up and running again.

morio, 2022-09-14:
Here is where the MPC60 selects SCSI ID 0 with no response from the RP2040:
[image]

About 350ms later the MPC60 sets the data bus to 0, then deactivates the SEL line, after that it reasserts the SEL line but looking for SCSI ID 1:
[image]

morio, 2022-09-14:
This is when RP2040 is trying to select SCSI ID 0. There is almost 1V of ring when the single goes low.
Blue is the SEL line and purple is data line 0. I tried to trigger on BSY and C/D but it never found anything.

[image]

PetteriAimonen, 2022-09-15:
Looks like the fix made for GD32 was never added to RP2040.
Should be fixed by this:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/pull/63

The ringing in scope trace is likely to be from probing setup, if using a ground clip cable for grounding.
A ferrite ring around the ground clip cable can help, or using a ground spring instead.

morio, 2022-09-15:
This probably won't help, but here is the log file for the MPC60. It detects the SCSI card but stall on loading samples from the ZuluSCSI RP2040
[zululog-mpc60-stalls-while-loading.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9577481/zululog-mpc60-stalls-while-loading.txt)

morio, 2022-09-15:
The Kurzweil K2000R is still sometimes failing to format. This is basically the same log file as before for the Kurzweil, when it fails it still fails at the 32 block offset.

[zululog-kurweil-failing-at-the-32offset.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9577566/zululog-kurweil-failing-at-the-32offset.txt)

PetteriAimonen, 2022-09-16:
Is it possible to get logic analyzer traces from when it stall? Preferrably also with the logic analyzer file so I can zoom in.

morio, 2022-09-19:
Here is the first READ(6) that has a warning in the log.
[SCSI_bus_zulu-rp2040-read6.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9602140/SCSI_bus_zulu-rp2040-read6.zip)
[zululog-read6.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9602141/zululog-read6.txt)

morio, 2022-09-19:
This is a successful format of the Kurzweil. It does a READ(10) instead of a READ(6). Here are the waveforms for that READ(10). 

[SCSI_bus_zulu-rp2040-read10.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9602379/SCSI_bus_zulu-rp2040-read10.zip)

[zululog-read10.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9602380/zululog-read10.txt)

PetteriAimonen, 2022-09-20:
Thanks, those last two captures from Kurzweil show a clear problem.
The response to `READ CAPACITY` is cut short, last bytes are not transferred correctly.

[image]

I think this is some bug in accelerated RP2040 SCSI routines that only happens with short transfers.
I'll try to replicate the bug and find out the cause.

--

In the meanwhile, there is this warning which is curious:

> Failed to offload SCSI DMA interrupts to second core, using first core

Are you using some debugging system (such as pico-debug) which might be reserving the second core?
In that case the warning is normal and doesn't matter.
Otherwise it's worth investigating what causes it to fail to initialize second core in some logs and not in others.

morio, 2022-09-20:
No, just the debug logging feature from the dip switch. Neither the JTAG or the SWD were in use.

PetteriAimonen, 2022-09-22:
I think this is fixed here:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/3104651983

It wasn't correctly waiting for the last byte, so if the host took more than a few microseconds to read it out it was discarded.

morio, 2022-09-22:
MPC60 looks great. Ran the Kurzweil formatting 5 times and it never failed.

closed 2022-09-22 by morio