ZuluSCSI/ZuluSCSI-firmware #66: RP issues 1.08 with Ensoniq ASR/16/EPS

chickeneps, 2022-09-15:
On boot of Ensoniq, LED stays on for about 30sec, then turns off. ASR-10 can't boot from ZuluSCSI. But when it boots from floppy, it does see the ID and I can traverse directories. However, if I attempt to read a large instrument I get FILE OPERATION ERROR. This is with DEBUG=1.
[zululog RP ASR-10.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9579595/zululog.RP.ASR-10.txt)

PetteriAimonen, 2022-09-16:
Could be related to #61, but I'm not sure yet what is going on.

PetteriAimonen, 2022-09-22:
#61 got resolved, so newest release could be worth trying:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/latest

chickeneps, 2022-09-22:
With RP2040 looks good! Not a problem, and tested thoroughly with ASR-10, 16-PLUS and EPS.

aperezbios, 2022-09-22:
Thanks. Closing as fixed. If you encounter any other RP2040-specific problems with any _other_ samplers, please open a new ticket.

closed 2022-09-22 by aperezbios