ZuluSCSI/ZuluSCSI-firmware #71: Akai S1000 / S3000 timing/packet issue

chickeneps, 2022-09-22:
Seemingly on longer transmissions (maybe not) on a Akai S1000-type and S3000-type (not XL though) complete the process but the packets seem to be out of order. These are music samplers and longer transmissions are the sound data. I look at the corrupted sound data graphically and I can see that the corruption is order-related, not wrong-data-related.

Setting DEBUG = 1 fixes this issue for some reason. SCSI2Enabled, whether 0  or 1, does not have any effect.

Tried with both Mini and Maxi ZuluSCSI models, and both 1.07 and 1.08 firmwares. This has been the case way back to the first firmwares.

On a semi-related note, RaSCSI does the same thing; the author has not investigated yet after months knowing about it. This does not happen with any SCSI2SD model or firmware.

chickeneps, 2022-09-22:
Here's some logs with 1.08.
[zululog S1000 1.08 NODEBUG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9629053/zululog.S1000.1.08.NODEBUG.txt)
[zululog S1000 1.08 DEBUG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9629054/zululog.S1000.1.08.DEBUG.txt)

PetteriAimonen, 2022-09-23:
Could you try adding a bit of selection delay and disabling prefetch and see if it changes anything:

     [SCSI]
     SelectionDelay = 10
     PrefetchBytes = 0

It sounds like the sampler may be confused by the low access latency of solid state media, causing some bug in their software to appear.

aperezbios, 2022-10-02:
@chickeneps see above

chickeneps, 2022-10-02:
Sorry, I had checked it back then. It didn't help, but I was trying either 1.0.7 or 1.0.8. I will try 1.0.8 and 1.0.9 on Monday.

aperezbios, 2022-10-03:
@chickeneps no need to try 1.0.9 on a V1.1. There are no relevant changes to any code path on V1.1. All of the changes in 1.0.9 are RP2040-specific

chickeneps, 2022-10-03:
I understand, but my checks 10 days ago may have dealt with 1.0.7 and not 1.0.8. I have to recheck.

chickeneps, 2022-10-03:
Checked with 1.0.8 and 1.0.9, condition still exists. Tried on full-size this time (no Mini around lately)

PetteriAimonen, 2022-10-04:
I have a debugging build here:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/3179595231

@chickeneps If possible, please post debug & non-debug logs of an identical situation (e.g. loading same sound file) with that firmware. Either one of mini or full-size is ok for this test.

That version logs the read address even with debug switch is off, but should avoid other delays.
It will tell whether the commands are received incorrectly for some reason.

chickeneps, 2022-10-05:
Both DEBUG and NODEBUG sounded good after loading. PhyMode = 2 on both.
[zululog S1000 1.09a DEBUG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9711986/zululog.S1000.1.09a.DEBUG.txt)
[zululog S1000 1.09a NODEBUG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9711987/zululog.S1000.1.09a.NODEBUG.txt)

chickeneps, 2022-10-05:
(I might have the logs marked opposite, sorry)

PetteriAimonen, 2022-10-05:
@chickeneps Thanks. That it worked with just that one log message was a surprising but useful result.

My current hypothesis is that the sampler needs some extra delay before SCSI phase changes. The SCSI standard only specifies delay needed after phase changes, but no delay needed before. From SCSI2SD code it is also known that e.g. EMU EMAX needs extra delay after changes, larger than specified by SCSI standard.

To verify this, I have another debug build here:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/3187823331

If you can, test it this way:
1. It should now work with debug log off. If it doesn't, report back and no need to continue to 2.
2. Try **zuluscsi.ini** with the setting below, and reduce the value until it no longer works. The default delay without config is 400 us in the debug build, slightly higher than what the logging caused. I suspect the real delay needed will be below 100 us.
```
[SCSI]
PhaseChangeDelay=400
```

chickeneps, 2022-10-05:
I'm supposed to be using the "ZuluSCSI Binaries" link on the bottom, right?

chickeneps, 2022-10-05:
Looks like 300 and higher sounds good, 200 and lower gets clicks.
Logs are
(b) Oct 4 build, PhyMode=2 (good)
(c) Oct 5 build, PhyMode=0, PhaseChangeDelay=400 (good)
(d) Oct 5 build, PhyMode=0, PhaseChangeDelay=100 (bad)
(e) Oct 5 build, PhyMode=0, PhaseChangeDelay=200 (bad)
(f) Oct 5 build, PhyMode=0, PhaseChangeDelay=300 (good)
[zululog S1000 1.09b DEBUG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9716586/zululog.S1000.1.09b.DEBUG.txt)
[zululog S1000 1.09b NODEBUG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9716587/zululog.S1000.1.09b.NODEBUG.txt)
[zululog S1000 1.09c DEBUG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9716588/zululog.S1000.1.09c.DEBUG.txt)
[zululog S1000 1.09c NODEBUG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9716589/zululog.S1000.1.09c.NODEBUG.txt)
[zululog S1000 1.09d NODEBUG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9716590/zululog.S1000.1.09d.NODEBUG.txt)
[zululog S1000 1.09e NODEBUG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9716591/zululog.S1000.1.09e.NODEBUG.txt)
[zululog S1000 1.09f NODEBUG.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9716592/zululog.S1000.1.09f.NODEBUG.txt)

aperezbios, 2022-10-05:
@PetteriAimonen what's the default phase change delay, if nothing is set in the INI?

chickeneps, 2022-10-05:
I think he said above that it was 400; when I said 400 above, I actually mean nothing was set.

aperezbios, 2022-10-05:
> I think he said above that it was 400; when I said 400 above, I actually mean nothing was set.

Thanks for clarifying.

PetteriAimonen, 2022-10-06:
The SCSI2SD code autodetects SCSI1 hosts by checking whether they send IDENTIFY message at start of command.
The `EnableSCSI2 = 0` can also be used to force SCSI1 mode.
Currently there is 100 us of extra delay added after phase changes if SCSI1 host is detected, with the comment that EMU EMAX needs it.

However 400 us (to leave some margin) at each phase change is a bit excessive.
I think most reasonable is to add the 400 us delay before phase change only on read commands on SCSI1 hosts.
The firmware can start reading the SD card during this delay, so it should not affect read latency or performance significantly.

I'm still not sure why SCSI2SD does not have this problem.
ZuluSCSI without these fixes had 7 us of delay from end of command to change of phase, while SCSI2SD V5.1 has 42 us.
Based on this I expected that less than 100 us of extra delay would have been enough.

aperezbios, 2022-10-06:
@PetteriAimonen how do you want to handle merging this change? Is the PhaseChangeDelay branch/addition suitable for merging as-is? If so, please submit a PR.

PetteriAimonen, 2022-10-07:
@aperezbios No, the current debug build is not suitable for merging. I'll do a new commit & PR when I have time, it'll probably have to wait until next week.

PetteriAimonen, 2022-10-12:
@chickeneps There is new build here, which should now be ready for merging. If you can verify that it also works correctly for you, that would be great!
https://github.com/ZuluSCSI/ZuluSCSI-firmware/suites/8733987944/artifacts/395179059

@aperezbios It might be a good idea to check with some of the older SCSI-1 Macs that there are no performance degradations. I tried with the LC III and QuickBench gave similar results as for 1.0.9.

PetteriAimonen, 2022-10-12:
Thanks!

The option no longer does anything. I hardcoded 400 µs delay to give it some margin, and put the delay only between command and data phases.

closed 2022-10-12 by aperezbios