ZuluSCSI/ZuluSCSI-firmware #80: V1.0.9 mismatches in reported capacity vs file sizes

cdhooper, 2022-10-08:
It seems I've run across one or two minor bugs in the ZuluSCSI RP2040 firmware related to how files are presented as devices. I created two files. The first is 512MB and the second is 512MB+1 sector:
-rw-r--r-- 1 cdh cdh  536870912 Oct  4 15:43 HD1.img
-rw-r--r-- 1 cdh cdh  536871424 Oct  7 17:21 HD3.img

The Geometry and accessible space provided to the host is significantly different between these two files:

<PRE>
9.OS321b2:4091> devtest -g a4091.device 1
                 SSize TotalSectors   Cyl  Head  Sect  DType Removable
TD_GETGEOMETRY     512      1048576  8192    16     8  0x00  No
Inquiry                                                0x00  No
READ_CAPACITY_10   512      1048576
READ_CAPACITY_16     -            -                    Fail 52 ERROR_SENSE_CODE
Read-to capacity   512      1048575
Mode Page 0x03     512                             63
Mode Page 0x04                         65   255
9.OS321b2:4091> devtest -g a4091.device 3
                 SSize TotalSectors   Cyl  Head  Sect  DType Removable
TD_GETGEOMETRY     512      1048640  8192    16     8  0x00  No
Inquiry                                                0x00  No
READ_CAPACITY_10   512      1048640
READ_CAPACITY_16     -            -                    Fail 52 ERROR_SENSE_CODE
Read-to capacity   512      1048640
Mode Page 0x03     512                             63
Mode Page 0x04                         65   255
</PRE>

The 1048576 value is what is reported to SCSI READ_CAPACITY_10. The 1048575 value is the number of sectors which can be read (which is off by 1). This issue does not manifest with the V1.1 and same 1.0.9 firmware version and same SD Card. Output below.

<PRE>
9.OS321b2:4091> devtest -g a4091.device 1
                 SSize TotalSectors   Cyl  Head  Sect  DType Removable
TD_GETGEOMETRY     512      1048576  8192    16     8  0x00  No
Inquiry                                                0x00  No
READ_CAPACITY_10   512      1048576
READ_CAPACITY_16     -            -                    Fail 52 ERROR_SENSE_CODE
Read-to capacity   512      1048576
Mode Page 0x03     512                             63
Mode Page 0x04                         65   255
</PRE>

For the second drive shown in the first block (target 3), 1048640 sectors is reported for the capacity, but the correct value should be 1048577 (512 byte) sectors because 536871424 / 512 = 1048577. In this case, it seems *more* drive space is being made accessible than the size of the file. This issue does also manifest on the V1.1.

These are not high priority issues for me. Just minor items I noticed while testing with the new device.

PetteriAimonen, 2022-10-08:
Could you provide a debug log (written to `azullog.txt` when `DBG` dip switch is on) of this?
It would let me see the checksums on the ReadCapacity requests so that I can verify what ZuluSCSI is actually sending.

cdhooper, 2022-10-08:
[zululog_rp2040_sector_end_bug.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9738455/zululog_rp2040_sector_end_bug.txt)

I won't be able to do another test run on that configuration until tomorrow. In the meantime, I do have a previously captured zululog.txt taken when the DBG dip switch was off. I hope this helps at least a bit.

PetteriAimonen, 2022-10-08:
Thanks, I think I know what is happening.

After commit 1c98755 the image size for contiguous files is being taken from the range SD card sectors allocated to it.
But FAT32 typically has 32kB allocation cluster size.
So the extra space is actually allocated to the file even though the file size is set to smaller.

This is still a bug and the sector count should be limited to file size.

aperezbios, 2022-10-17:
Should be fixed in firmware release 1.1.0. @cdhooper, can you test? If not, please feel free to re-open this issue.