ZuluSCSI/ZuluSCSI-firmware #81: Issue Flashing Firmware v1.0.9 to ZuluSCSI RP2040

WJG6260, 2022-10-10:
Platform: ZuluSCSI RP2040
FW Version: 1.0.9-release Sep 30 2022 18:19:23

Earlier tonight, I attempted to upgrade my ZuluSCSI RP2040 from firmware 1.0.8 to 1.0.9 using the methods described. I renamed the v1.0.9 RP2040-specific bin file to ZuluSCSI.bin, inserted the SD card with such file at the root directory, and booted the device.

The light on the device flashed numerous times (more than the five described as a failure), and I am now unable to boot from the device.

**The following zuluerr.txt was generated:**

```
[10ms] DIP switch settings: debug log 0, termination 1
[10ms] SCSI termination is enabled
[11ms] Bootloader version: Sep 30 2022 18:19:21 ZuluSCSI RP2040
[267ms] Found firmware file: ZuluSCSI.bin
[268ms] DBG Writing flash at offset 0x00020000 data 0x00 0xC0 0x03 0x20 

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : BC0C47B8
R1   : 000206FC
R2   : 00000002
R3   : BC0C47B8
R4   : 000206FC
R5   : 1FFEC3DC
R6   : 00021000
R7   : D0000000
R8   : 00000000
R9   : 00000000
R10  : 00000000
R11  : 00000000
R12  : 4001801C
SP   : 2000A0F8
LR   : 100046FB
PC   : D0000000
xPSR : 80000000
PSP  : 2000A0D8
MSP  : 2003BFC0
CPUID: 410CC601
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

[316ms] --------------
[316ms] CRASH!
[317ms] Platform: ZuluSCSI RP2040
[317ms] FW Version: 1.0.9-release Sep 30 2022 18:19:23
[317ms] error_status: 0x80FF013D
[318ms] error_address: 0xD0000000
[318ms] error_value: 0x2000D680
[318ms] STACK 0x2000A0F8:    0x00020000 0x2000C3DC 0x2000C3DC 0x00000004
[319ms] STACK 0x2000A108:    0x2000C3DC 0x00000000 0x2000A180 0x10000929
[320ms] STACK 0x2000A118:    0x2000A180 0x00000024 0x00000000 0x2000A738
[321ms] STACK 0x2000A128:    0x20011991 0x00000000 0x00000000 0x100009B5
[321ms] STACK 0x2000A138:    0x00000000 0x756C755A 0x49534353 0x6E69622E
[322ms] STACK 0x2000A148:    0x00000000 0x1000DD2B 0x10008DA4 0x61000000
[323ms] STACK 0x2000A158:    0x2000BBF4 0x1000DD2B 0x2000BB34 0x2000B924
[324ms] STACK 0x2000A168:    0x2000B148 0x1000D385 0x00000000 0x00000000

```

At this point, I cannot seem to access the device via Mac/Windows and cannot boot the device on an Adaptec AHA-2842W, as I had done before.

I attempted another flash, this time to the nightly build and received the same exact error.

Any thoughts would be appreciated greatly.

PetteriAimonen, 2022-10-10:
There have been some reliability problems with RP2040 bootloader. They should be resolved by the next firmware release.

To upgrade the bootloader or to recover from bad flashing on RP2040 platform:
1. Download the `.uf2` format of the firmware.
2. Enable the `BOOTLOADER` DIP switch on the ZuluSCSI RP2040 PCB.
3. Connect with USB cable to computer. A drive should show up.
4. Copy the `.uf2` file to the USB drive.

aperezbios, 2022-10-20:
@WJG6260 no response in over a week, so I'm going to close this as resolved, since we've provided the only workaround. Feel free to re-open this if need be.

closed 2022-10-20 by aperezbios