ZuluSCSI/ZuluSCSI-firmware #82: AKAI S3000XL ZULUSCSI 1.1  CRASH AFTER GOOD OPERATION

linochi, 2022-10-10:
Same results with firms 1.06  1.07  1.08 and 1.09
ZuluScsi 1.1 installed and connected on internal 50pin conector. 
Added hda0 to hda4 (510Mb) and one CD5 (ISO image).
All hda blank files (510Mb) created on W10 recognized and formated on AKAI (14 partitions of 32MB).
No problems and no bad sectors when fomat fisnished.
The ISO image on CD5  can be see it on AKAI with the correct structure and can read (load) on the sampler memory. (ID 5 default on AKAI sampler, ID6 is the SAMPLER default host ID)
I can save the contents of memory sampler on hda partitions and work using the normal sampler operations.
BUT after some time the ZuluScsi CRASH with intermitent "morse" led flashes
[zuluerr.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9743484/zuluerr.txt)
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9743485/zululog.txt)
[zuluscsi.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9743496/zuluscsi.txt)
zuluscsi.ini  renamed to txt

linochi, 2022-10-10:
seems that this conf works on SCSI2SD 
https://github.com/adamjansch/SCSI2SD-S3000XL/blob/master/SCSI2SD%20Akai%20S3000XL%20config.xml

PetteriAimonen, 2022-10-10:
This might be related to issue #71. You can try the test build from here until we get the fix merged to main code:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/3187823331

linochi, 2022-10-10:
Thanks Petteri for this test firm !!  
Just loaded and testing...  14:50
For now stable operation, only using the default ID5 SCSI on sampler and only loading from the partitions on CD5.img

linochi, 2022-10-10:
Just loading a few samples from CD5.img and see on sampler display how stops loading, then morse leds on Zuluscsi and the sampler message "HARD DISK DRIVE NOT READY" , i can continue with normal sampler operations but **can´t save anything** because ZuluScsi is on crash.

[zuluerr.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9746283/zuluerr.txt)
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9746284/zululog.txt)
[1316847ms] DBG BUS RESET
[1319265ms] DBG BUS RESET
[1340477ms] -------------- WATCHDOG TIMEOUT
[1340477ms] --------------
[1340477ms] CRASH!
[1340477ms] Platform: ZuluSCSI v1.1
[1340477ms] FW Version: 1.0.9-release Oct  5 2022 06:47:25
[1340477ms] CFSR: 0x00000000
[1340477ms] SP: 0x2001FF28
[1340477ms] PC: 0x08013C62
[1340477ms] LR: 0x080142D5
[1340477ms] R0: 0x00000011
[1340477ms] R1: 0x4001800C
[1340477ms] R2: 0x4001800C
[1340477ms] R3: 0x40018034
[1340477ms] STACK 0x2001FF28:    0x00000011 0x4001800C 0x4001800C 0x40018034
[1340477ms] STACK 0x2001FF38:    0x0000000F 0x080142D5 0x08013C62 0x61000000
[1340477ms] STACK 0x2001FF48:    0x00000451 0x0000001D 0x00000200 0x080142D5
[1340477ms] STACK 0x2001FF58:    0x00000000 0x003D33AF 0x20000340 0x003D33AF
[1340477ms] STACK 0x2001FF68:    0x20004E11 0x000001C9 0x000001C9 0x200007E8
[1340477ms] STACK 0x2001FF78:    0xF300F600 0x080138AB 0x00000200 0x00000000
[1340477ms] STACK 0x2001FF88:    0x20000340 0x20000330 0x00000001 0x0800E4FF
[1340477ms] STACK 0x2001FF98:    0x200007D8 0x00000045 0x0000002F 0x0800D123

PetteriAimonen, 2022-10-10:
Thanks for testing - apparently the issue you have is a different one.

I decoded part of the crash dump and it appears to be stuck inside the SD card driver.
There is some firmware bug here, even if the SD card is broken it should give a better error message.
It will take a few days until I have time to look at the crash dump in more detail.

If possible, you could try another SD card to see if it works better.

linochi, 2022-10-10:
Great discover Petteri !!!

One test i maded that reproduce the same situation is power up the sampler and do nothing.... just over aprox 20 minutes the morse code starts blink.

With the last firm you linked i noticed that the navigation thru the load pages is a bit (just a bit) slower i think due to the delay of Phase modified on the test firmware.

The SD used is a SanDisk Ultra SDHC UHS-I of 32Gb.

If i find another SD i´ll try and test.

PetteriAimonen, 2022-10-10:
Yeah, the test firmware has unnecessarily much delay, so it has to be modified before it can be merged to main firmware.

linochi, 2022-10-10:
I have a K2600 Rack version and a Yamaha A4000 sampler with an SCSI2SD and works without problems.
If the ZuluScsi project is based on SCSI2SD and BlueSCSI, i understand that the code is different due to the "manufacturing problems" of the silicons used, that are different.

linochi, 2022-10-10:
Yeah !! ZuluScsi emulate with code the FPGA chip to interact with the SCSI bus.

Before i decided install a ZuluSCSI on the AKAI i´ll try using an LVD-SCSI to IDE Brigde from ACARD company with the model AEC 7722 and then and IDE to CF adapter. 
But i discover that this model AEC 7722 if not modified **only** works to emulate CD units.
The model AEC 7020 works with HD´s and CD units.

linochi, 2022-10-12:
Thanks Petteri !!!
I go to test with this new test build and report results

linochi, 2022-10-12:
Normal operation on sampler only loading from CD5.img until the crash
[zuluerr.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9767788/zuluerr.txt)
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9767789/zululog.txt)
Now the WATCHDOG shows info
[285060ms] WATCHDOG TIMEOUT at PC 0x08013C90 LR 0x08014EA1 attempting bus reset
[300060ms] -------------- WATCHDOG TIMEOUT
[300060ms] --------------
[300060ms] CRASH!
[300060ms] Platform: ZuluSCSI v1.1
[300060ms] FW Version: 1.1.0-rc1 Oct 12 2022 09:29:31
[300060ms] CFSR: 0x00000000
[300060ms] SP: 0x2001FF90
[300060ms] PC: 0x08013C90
[300060ms] LR: 0x08014EA1
[300060ms] R0: 0x0000000D
[300060ms] R1: 0x4001800C
[300060ms] R2: 0x0001868A
[300060ms] R3: 0x40018034
[300060ms] STACK 0x2001FF90:    0x0000000D 0x4001800C 0x0001868A 0x40018034
[300060ms] STACK 0x2001FFA0:    0x0000000F 0x08014EA1 0x08013C90 0x21000000
[300060ms] STACK 0x2001FFB0:    0x0000000D 0x0001868B 0x4001800C 0x20000308
[300060ms] STACK 0x2001FFC0:    0x2001FFE4 0x08014EA1 0x080133A3 0x20000308
[300060ms] STACK 0x2001FFD0:    0x20000208 0x080133AB 0x080133A3 0x080091E5
[300060ms] STACK 0x2001FFE0:    0x00001388 0x4001800C 0x00000000 0x20000208
[300060ms] STACK 0x2001FFF0:    0x41404A80 0x0800B6EF 0x20007C75 0x0800857F

linochi, 2022-10-12:
The SD card used is the same, not used another one and it´s brand new, but can be a faulty card the results of the crash?
The ZuluSCSI card is conected to the internal 50 pin SCSI sampler bus and too powered with an FDD 4 pin power cable.

PetteriAimonen, 2022-10-13:
That's weird, it is stuck in the same place as before ([r1_error_check()](https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/main/lib/ZuluSCSI_platform_GD32F205/gd32_sdio_sdcard.c#L2004)), even though there is a timeout in that loop now. The firmware is up to date so it's not a problem with that either.

I don't currently know what would explain the problem, I'll have to think about it more..

linochi, 2022-10-13:
I go to try with another SD and test. 

The mainboard of my model S3000XL has an internal 50 pin SCSI bus conector.
In the manufacturer manual don´t reflect that can´t use internal SCSI drives only trhu the external DB25 conector. 
The machine was designed only with the idea of 1 bay slot for an FDD. No molex extenders on PSU or manboard for supply power to internal HD, but is the same mainboard of other models that have space to allocate HDD.
I don´t think that the last firmware 2.0 contain some code to cause interference with the internal 50 pin SCSI bus.

linochi, 2022-10-13:
Can be a loose of contacts on the SD card holder due to thermal events? 
This was a problem with SATA data cables of poor quality loosing conection.
Or RFI / EMI interferences?
I don´t know whats the frequency clock of the ZuluSCSI card, but on high MHz can cause errors if not shielded against interferences. 
The original LCD of the sampler need a voltage for the CFC backlight tube and uses a coil that elevate the voltage.

linochi, 2022-10-13:
I go to test first with the same card but with the ZuluSCSI card out of the sampler.

linochi, 2022-10-13:
Stable now and no morse leds!!!! aprox 40 min of operation
Last tests in aprox 20 min... CRASH
Only loading from ID5  (CD5.img)

linochi, 2022-10-13:
continue testing:
 saving sample memory content on HD0.hda   GREAT!!!  NO CRASH!!!
 saving sample memory content on HD1.hda   GREAT!!!  NO CRASH!!!

linochi, 2022-10-13:
Now think to modify a better position of the ZuluSCSI inside the sampler.

linochi, 2022-10-13:
Petteri, i go to test with the 1.07 1.08 firmwares , but this time with the ZuluSCSI card outside the sampler

linochi, 2022-10-14:
Good and stable on any firm with the ZuluSCSI outside the sampler.
Tested hot swap : 5 led blinks removing SD, reinsert SD and recognized ok.

BUT.... 
one of the times ZuluSCSI entered on crash... 
I think the main problem was the "lock system" of the SD card holder...  
If you push the SD card to insert and plug out on the SD holder you can hear the click insert mechanism..... but if you think is "locked" is NOT.
You **can** remove the SD card **without** push using the click mechanism.
If for some reason the SD card once inserted you accidentally touch the SD card and is not properly alligned.... over the time cause the crash.
The hot swap works with this host (S3000XL) but keep in mind that the "lock system" **NOT** assure the correct SD card pin contacts.

closed 2022-10-16 by linochi