ZuluSCSI/ZuluSCSI-firmware #90: Akai MPC60 written audio data has artifacts

CFD2, 2022-10-28:
### Description
Reading saved audio files back into memory results in them having artifacts on Akai MPC60 audio sampler.

### Steps taken to reproduce
 - wrote a small file and 4 large audio files to SD card
 - power cycled the unit and loaded saved files back into memory from the SD card
 - auditioning the audio showed digital artifacts as in random clicks overlayed on original audio

### Observations
 - The issue can be reproduced 100% of the time on SD Cards with capacity larger than 16GB.
 - The issue is random when SD Cards are 16GB.
 - The unwanted audible clicks added are of digital nature.

### Additional Information
Used [default zuluscsi.ini](https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/main/zuluscsi.ini) with debug flag set.
[zululog_write.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9885436/zululog_write.txt)
[zululog_readback.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9885435/zululog_readback.txt)

### Platform
ZuluSCSI 1.1

### Firmware
_1.1.1_

PetteriAimonen, 2022-10-28:
Would it be possible to save the corrupted files again under different name, and then upload the image file somewhere?
That would let me see exactly what kind of corruption occurs to the files.

CFD2, 2022-10-28:
@PetteriAimonen I'm not sure I'd have a way to extract files from the images. I could upload the disk image and then you could probably strip with `dd`. Let me know if that works for you.

If that does not work, I could probably figure out a way of extracting just those files with tools that exist but this would be adding complexity to the issue since I do not know how reliable those tools are. It would also take some time and does not guarantee an outcome.

To clarify, **_are you interested in seeing what is being read back from the SD_**, i.e. the result of original audio in memory saved, read back (aka the moment we get the evidence of corruption) and then saved again to see if it corrupts even more?  

**_Or are you trying to figure out whether it is a read or a write issue?_** I'm almost certain the issue is with write operation. The reason for that is what was once read back without artifacts would always read back without any corruption. And on the other hand, what was once read back with corruption would always read back with _clicks_ at the same file. I tested this on SD Cards with smaller capacities because, as I mentioned earlier, the issue is random with those.

PetteriAimonen, 2022-10-28:
I'm trying to figure out in what kind of change happens in the write issue - i.e. does a whole block get corrupted or just a single byte, and at what position in the buffer it is.

If you can do this:

1. Load some file or files and write them under different name.
2. Send me the debug log and the resulting image file from this.

I can get the sector addresses from the log so I can dump both the file that was loaded, and the file that was written and compare them.

CFD2, 2022-10-28:
@PetteriAimonen that makes sense. MPC 60 formats hard drives into 26 partitions (`A` - `Z`) where each partition is 23MB, if I recall correctly.

Here is another SD card from yesterday where I managed to have saved files without corruption on one partition and with corruption on another.

Image file:
https://www.dropbox.com/s/3ymub3ewjc27l9k/HD0.img?dl=0

The logs below demonstrate loading files without corruption:
[zululog_good.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9891029/zululog_good.txt)

and the logs for loading corrupted files from another partition on the same disk image:
[zululog_bad.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9891028/zululog_bad.txt)

I reckon, based on your reply, that should be enough information to determine correct offsets.

Let me know if there's anything else I can provide.

PetteriAimonen, 2022-10-28:
Thanks! I'll take a deeper look on Monday.

CFD2, 2022-11-01:
### New logs and disk image
The log below is a load from **partition A** and save to **partition D** on 64GB SD card (`PhyMode = 0`):
[zululog_write_64gb.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9906414/zululog_write_64gb.txt)

This log is a load from **partition A** and save to **partition E** on 64GB SD card (`PhyMode = 1`):
[zululog_write_64gb_phy1.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9906552/zululog_write_64gb_phy1.txt)

This log is a load from **partition D** (where we have just saved material) with `PhyMode = 1`. The unit seems to load the first (_smaller_) file OK but becomes completely unresponsive when encountering a (_larger_) corrupted file:
[zululog_read_64gb_phy1_freeze.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9906412/zululog_read_64gb_phy1_freeze.txt)

Disk image file:
https://www.dropbox.com/s/cta7ezg4d2as60x/HD0_64gb.img?dl=0

### Observations
Setting `PhyMode` back to default (in my case that is `GREENPAK_DMA`) makes the unit able to load the files even though they are corrupted. Seems like `GREENPAK_DMA` is a lot more forgiving to the file contents.

When saving to 64GB SD Card, I can hear occasional signal spilling through audio outs of the unit. The signal is of digital nature. In other words, it reminds me of a mobile phone about to receive a call near an audio system. (I'm sure this noise is familiar) Except the bleeding occurs in occasional spurts. When I hear that, I know for sure that there is going to be artifacts in  saved data. _I absolutely do not hear it when the write operation is successful with 16 GB card._

PetteriAimonen, 2022-11-01:
Those logs confirm that the issue is on the **SCSI interface** side. The checksums in write commands match the data that actually got written to corrupted file on SD card.

It is interesting that that `PhyMode = 1` works worse. It should result in slower data transfer, which one would assume to be more robust. Earlier we have seen that MPC60 has problems with reading if the SCSI device pauses during a sector transfer. There is attempt at avoiding pauses during writing but it might not be fully effective, I'll have to check with logic analyzer.

Here is a test firmware that disables interrupts during SCSI receive. If you could check whether it still corrupts files with `PhyMode = 0` (download "ZuluSCSI binaries" zip at bottom of page). It is not suitable for generic use because it disables watchdog and reset events, but should help determine whether the pauses are the problem:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/3367950482

Regarding the sounds during writing, that would sound like a possible power supply issue. If you are powering ZuluSCSI through the SCSI port, it might be worth to try connecting an USB cable for extra power.

CFD2, 2022-11-01:
@PetteriAimonen the experimental software load (version [1ac3f58](https://github.com/ZuluSCSI/ZuluSCSI-firmware/commit/1ac3f5899fc3f021324c88a414601cadd9f08c25)) did not resolve the issue. The data is still getting corrupted during writes.

ZuluSCSI was powered through the 12 volt connector. Powering through the USB instead made things better (the data still corrupts on writes, though). Thanks!

CFD2, 2022-11-01:
@PetteriAimonen From my personal observation, the issue might be dependent on the SD Card write speed rather than capacity. I do not have experience with SDIO so I can only be guessing.

SD Cards tried so far:
 * Sandisk Extreme Pro 170 MB/s (reads?) 64GB SD Card. The fastest of them all. 10/10 data write attempts result in corruption. The audio files have **the most (audible) artifacts in them with this card.** Used for this issue.
 * Sandisk Extreme 90 MB/s 16GB SD Card (gold foil label). Slower. 4/10 data write attempts result in corruption. The corrupted audio files have **less frequent (audible) artifacts with this card.** Used for this issue.
 * Sandisk Edge class10 A1 16GB **_microSDHC_** card. The slowest of them all. Had the most success with this one so far but still far from perfect. There is a chance that only some audio files will have (audible) artifacts when corrupted write occurs.

The thought of trying 16GB was not my own as there are other people who seem to be having this issue with their MPC60 and ZuluSCSI. I found it difficult to believe at first.

PetteriAimonen, 2022-11-02:
Thanks for testing.

Regarding the "12 volt connector", ZuluSCSI is actually only using the 5V pin of the 4-pin floppy power connector.
Probably the sound artifacts are due to power draw but that does not necessarily have any direct effect on the write corruption.

My current theory is based on that SD cards perform internal wear-levelling after a certain amount of writing.
This could take more power (causing the sound) and takes time (altering transfer timing).
And depends on card model and capacity how the wear leveling is implemented and how often it happens.

I have a new test build here:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/3376530141

That test firmware reads the whole SCSI request to RAM before even starting the SD card writing.
The MPC 60 seems to issue write requests of max 5 kB, while there is 64 kB of RAM buffer reserved so this fits fine.
This should isolate any delay the SD card would cause during the SCSI transfer.

CFD2, 2022-11-02:
@PetteriAimonen 

**The new test build ([3f30592](https://github.com/ZuluSCSI/ZuluSCSI-firmware/commit/3f30592dabdf204a57360d05e8ac85f02963645e)) fixed the issue!** Would you have an idea of a more permanent solution?

> but that does not necessarily have any direct effect on the write corruption.
 
It definitely does not. Let me re-iterate, using the external USB connector as power source and disconnecting the 4-pin floppy power connector did not help prevent data corruption.

PetteriAimonen, 2022-11-02:
Great to hear!

I do have a thought on a proper fix that can be merged in main code.
There is some performance improvement work for write operations coming up, which will allow them to run in background without disturbance from SD transfers.

It will take 1-2 weeks until that is ready, so in the meanwhile feel free to use & inform others about that test build.
It's usable, just a bit slower write performance than usual.

aperezbios, 2022-11-02:
@chickeneps, FYI

chickeneps, 2022-11-03:
While I validate everything here, I've done a number of tests since MPC60 got working at all 2 months ago, and I've never experienced audio/write artifacts. My SD card of choice is SanDisk Ultra SDXC 64GB UHS-1 120mb/s.

A client recently asked me about MPC issues and I got one image from him that had errors and one that didn't. Since it was unclear from him how those errors got into the first image, we left it "at that". I have an image with a program called STUDIO SET.pgm and lots of SND files (comes from a factory MPC60 floppy) and I loaded it fine into the MPC60, saved it to Part B, power-cycled and reloaded, sounded fine.

For checking, I use my program Translator which can read those images and play back the samples, and no artifacts occurred on anything saved. Tomorrow I'm going try much harder, using not this new test build but the released 1.1.0, just to check.

That's just my experience, it doesn't invalidate anything said here, and I appreciate the addressing of these issues.

PetteriAimonen, 2022-11-03:
The dependencies on SD card timing are quite weird indeed.
Even assuming that the MPC-60 blindly pushes bytes without checking the ACKs at all doesn't explain the behavior, it seems it checks sometimes but then sometimes assumes that the drive must be ready.

With the 64GB card that has problems (based on logs SanDisk SD Extreme Pro), the corruption only happens when the MPC60 writes 9 sectors per request, not when it writes anything less. And the amount of sectors per request varies, probably based on SD card write speed vs. MPC60 internal speed:

[image]

CFD2, 2022-11-03:
@PetteriAimonen 
> And the amount of sectors per request varies, probably based on SD card write speed vs. MPC60 internal speed

I reckon, you could probably confirm it using the logs from Sandisk Extreme SD card (16GB) (not Extreme Pro) which also had this issue:
> Image file: https://www.dropbox.com/s/3ymub3ewjc27l9k/HD0.img?dl=0
> 
> The logs below demonstrate loading files without corruption (partition A): [zululog_good.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9891029/zululog_good.txt)
> 
> and the logs for loading corrupted files from another partition on the same disk image (partition H): [zululog_bad.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9891028/zululog_bad.txt)

PetteriAimonen, 2022-11-04:
Early shipped units of RP2040 board had a fault in bootloader that only occurred sometimes. After restoring from .uf2, the bootloader is also upgraded to a more robust version (ef3552f).

CFD2, 2022-11-04:
> Akai made them but also third-party people did too
I was under the impression that Marion Systems made them originally but good point nonetheless. Although, if I recall correctly, there are no issues saving to zip disks. Maybe increased bandwidth demands stricter timing requirements but that's me speculating.

I am glad this issue is getting attention. Thanks so much for participating in the discussion and testing the new software load, Garth!

aperezbios, 2022-12-20:
@chickeneps is this still an issue with current versions of ZuluSCSI firmware, on the ZS RP2040, or can we close this out?

chickeneps, 2022-12-21:
Tried it with many different kits; loaded, saved, reloaded from saved - no artifacts. This is with the public 1.15.

aperezbios, 2022-12-21:
Thanks for confirming, @chickeneps

closed 2022-12-21 by aperezbios

PetteriAimonen, 2022-12-21:
I think for V1.1 this bug is probably still present, but it should be fixed when 731f392c6 gets tested and merged. I just need to verify it doesn't cause performance regressions.

For RP2040 it was always quite unlikely for this bug to occur.

chickeneps, 2022-12-21:
Thanks for mentioning that.

PetteriAimonen, 2022-12-21:
Fix for V1.1 merged into main now also and will be part of next release.