ZuluSCSI/ZuluSCSI-firmware #91: Victor 9000 / Xebec 1410 support

pauldevine, 2022-10-28:
Hi there,
I tried to use the ZuluSCSI with a [Victor 9000](http://www.oldcomputers.net/victor9000.html), which is expecting a Xebec 1410 connected to an MFM drive. It's talking a vendor-specific extensions to the SASI protocol. It's all standard SASI, except the Victor issues a series of error check commands that are Xebec-specific as part of initializing the drive. 

In the configuration I took the defaults and adjusted:
```
Quirks = 4   # 0: Standard, 1: Apple, 2: OMTI, 4: Xebec, 8: VMS
EnableSCSI2 = 0 # Enable faster speeds of SCSI2
```
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9892378/zululog.txt)
Attached above are the logs. I know the non-standard commands that are being sent match this from the xebec documentation:
[image]

It looks to me like the E0 command isn't being responded to correctly. If I adjust the scsi.c file to support that and the other related checks, would you welcome a PR to enable this to work?

Thanks,
Paul

aperezbios, 2022-10-29:
> If I adjust the scsi.c file to support that and the other related checks, would you welcome a PR to enable this to work?

@pauldevine All pull requests are welcome. We may not accept it without modification, if it's in the critical code path, but we'd have to make that decision after we've actually seen the PR. We'll look forward to it. Thanks!

PetteriAimonen, 2022-10-29:
I think the correct place would be in `vendor.c`:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/main/lib/SCSI2SD/src/firmware/vendor.c

SCSI2SD V5 has dummy handlers for the Xebec commands here, so they could probably be just copied to the SCSI2SD V6 code that ZuluSCSI is using:
https://github.com/rabbitholecomputing/SCSI2SD-V5/blob/master/software/SCSI2SD/src/vendor.c

There seems also to be some parity errors. I'm not sure whether it will be necessary to add some option to disable parity check, or if the Xebec sends parity but there is some timing problem that is corrupting data.

pauldevine, 2022-10-30:
Thanks for the tips folks. I'll take a look at making the changes to vendor.c and see how it goes.

pauldevine, 2022-10-30:
OK, I just tried adding the additional commands to vendor.c as shown in this commit https://github.com/ZuluSCSI/ZuluSCSI-firmware/compare/main...pauldevine:ZuluSCSI-firmware:xebec. It definitely seemed to get further. I'm still seeing the parity errors, which is odd. The machine started to boot from the drive and then hung. I tried reformatting it, to see if there was some incompatibility from the image I'm using with comes from RASCSI. The low-level format command hung with an error about the controller not responding. Here's the log trace. Any thoughts?

[zululog devine 30-oct.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9896048/zululog.devine.30-oct.txt)

PetteriAimonen, 2022-11-08:
Sorry for the delay, seems I have missed your reply for some reason.

Regarding the parity errors, it seems to happen only in command phase.
The parity error is only logged during the command phase, but for write requests it would abort the request.
Parity support is apparently optional in SCSI-1 hosts, but currently the RP2040 code does not have a config option for disabling it.

You could try commenting out these lines in `scsi_accel_rp2040.cpp` to test without parity check:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/main/lib/ZuluSCSI_platform_RP2040/scsi_accel_rp2040.cpp#L556-L562
```
    uint8_t byte0 = ~(paritycheck & 0xFF);
    uint8_t byte1 = ~(paritycheck >> 16);
    if (paritycheck != ((g_scsi_parity_lookup[byte1] << 16) | g_scsi_parity_lookup[byte0]))
    {
        azlog("Parity error in scsi_accel_rp2040_read(): ", paritycheck);
        *parityError = 1;
    }
```

The debug log is written only intermittenly, so there may be something missing at the end when the device hangs.
To force writing of the full log, you can try this:
1. Remove SD card
2. Wait for status LED to start blinking (can take up to 30 seconds if hung in middle of a request)
3. Reinsert SD card
4. Wait for status LED to blink once and then stop blinking
5. Now the full log should have been written.

pauldevine, 2022-11-12:
Thanks for that suggestion. Sorry it took me a few days to have time to play with this. I tried commenting out the parity check as you suggested, I didn't see any changes in how the computer reacted. The parity errors are obviously gone from the logs. I tried pulling and reinserting the SD card to make sure I captured the full log, which is below. 
[zululog-noparitycheck.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9996105/zululog-noparitycheck.txt)

PetteriAimonen, 2022-11-13:
Seems like the host does not send the 8 bytes that are expected after the vendor-specific `0x0C` command. Not sure why.

aperezbios, 2022-12-12:
@pauldevine were you able to make any further progress with this? Just curious.

pauldevine, 2022-12-13:
Thanks for asking @aperezbios. I've been a bit distracted with some work stuff. I re-read the code that @Fozztexx did on the rascsi branch and I saw some additional messages in handling DiskRequestSense. https://github.com/FozzTexx/RASCSI/commit/2737aba53b57bc3c555145619a01719acb49f5e4 I'll try playing with that later in the week.

Right now if I try a low-level format I get an error "Controller timeout waiting for busy signal." Then if I try to partition the drive, as I have an image already on the card that should already be low-level formatted, I get "112-Controller error. Unable to reset hard disk." 

I'm just a little ignorant of SCSI specifics so when I read through what FozzTexx did on the RASCSI branch to get this to work I'm not entirely clear on how to translate this to your code. The changes are all articulated as part of this branch https://github.com/FozzTexx/RASCSI/tree/Victor9k.

the-claw, 2023-02-21:
I’m also seeing issues with parity errors on a SASI Sharp X68000 machine that does not have a parity data line. Reads work fine but spam the log with parity errors, writes always error out with sense 0x4700.

I will try commenting out the parity check at https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/3e89a88bf4561a9c68164d9c3b16475b8ed2c3a1/lib/ZuluSCSI_platform_RP2040/scsi_accel_rp2040.cpp#L686 and see if this fixes my problem, however an option in the config to disable the parity check would be very useful for these machines.

aperezbios, 2023-02-22:
@PetteriAimonen can we add an option to disable parity checks?

PetteriAimonen, 2023-02-22:
@aperezbios @the-claw There is a test build here now that allows disabling parity check: https://github.com/ZuluSCSI/ZuluSCSI-firmware/pull/147

I have my desk full at the moment so I didn't check on real hardware yet.

the-claw, 2023-02-26:
@PetteriAimonen thank you so much, I've tested this on RP2040 and replied to the pull request. All working great for me.