ZuluSCSI/ZuluSCSI-firmware #95: RP2040: System hangs for a period when starting a benchmark in SCSI Director with Zulu connected to 9600 fast internal SCSI bus

garthbeagle, 2022-11-03:
On a Power Macintosh 9600 with rp2040 ZuluSCSI connected to the faster "Internal" SCSI bus, the system will hard hang for about a minute when starting a benchmark in SCSI Director. After that hang, the benchmark will run, but then the system is unresponsive again afterwards. SCSI Director will then eventually throw up an error

[image]

I see watchdog timeouts and bus reset messages in the log file
[zululog-hang.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/9932696/zululog-hang.txt)

Setup
1. Power Macintosh 9600, ZuluSCSI rp2040 (1.1.1 firmware, no ini file) connected to Internal SCSI bus (the port closest to the edge of the logic board), clean install of Mac OS 8.6 installed in a 8GB HFS+ image on a 32GB SanDisk Extreme SD card
2. Launch SCSI Director 4.0
3. Select the ZuluSCSI device from the list
4. Click Test button on left side of window
5. Save the results file and click ok

Results
The system hard hangs for about a minute. If you let it sit, it'll actually run the benchmark but hangs again at the end, where the app itself throws up an error (see screenshot above)

Does not occur
- If you instead install the ZuluSCSI rp2040 on the slower Internal/External bus (the SCSI port furthest from the edge of the logic board
- If you instead use an original gd32 ZuluSCSI (also 1.1.1 FW)

Notes
We had reported stability issues in the past when using the gd32 ZuluSCSI off this same SCSI bus, which have now been resolved

aperezbios, 2022-11-04:
@garthbeagle. The hangs are a byproduct of the watchdog timeouts. This is another, possibly-similar 10MByte/sec synchronous SCSI bug, it would seem.

Can you try this development build, and let us know if the same thing happens? https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/download/latest/ZuluSCSI_RP2040_2022-10-26_85c64b1.bin

garthbeagle, 2022-11-04:
Was there a specific change made in this release that may address this?

I ask as I won't be able to try this for awhile. I just dumped this info into a bug for you all as I know you're concerned about feature/bug parity with the original ZuluSCSI and I *think* you have a way to reproduce the stability issues originally seen there?

aperezbios, 2022-11-04:
@garthbeagle yes, there are two RP2040-specific, synchronous-specific fixes that were recently made, either of which may solve this problem. It's also a develpoment build, not a release, technically.

aperezbios, 2022-11-06:
@PetteriAimonen can you take a look at these logs as time allows?

PetteriAimonen, 2022-11-08:
The watchdog timeouts in the logs match the problem that was fixed by commit f11c0d4, involving synchronous mode and ModeSense command. That fix is included in the latest development build linked by Alex.

That fix made ModeSense work reliably on Linux, but it was reported as still failing on Amiga.
Regarding that there is a version with debug test code [here](https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/3360511545), which I'm still waiting feedback and debug logs about.
If the latest development build does not fix the problem for you, you could try that one also.

(**Note:** As a workaround one can disable synchronous mode using the .ini file. It should work reliably but slower in that case.)

PetteriAimonen, 2022-12-02:
@scsiscsi Regarding the performance degradation, I don't get that problem with the Linux based hardware I have available. All version 1.1.1, 1.1.2 and 1.1.3 give same 8.2 MB/s read and 2.8 MB/s write performance for me.

What kind of performance are you getting on PowerMac 9600?

aperezbios, 2022-12-26:
Fixed in firmware release 1.2.0

closed 2022-12-26 by aperezbios