ZuluSCSI/ZuluSCSI-firmware #97: 256 sector support is broken

Deftaudio, 2022-11-09:
Hi there, please, help, 256 sector byte emulation crashes ZuluSCSI on the latest 1.1.1 firmware
The host and images don't matter. Here is the dump with debug info below. I haven't tried other FW versions. 

`[10ms] DIP switch settings: debug log 1, termination 1
[10ms] SCSI termination is enabled
[12ms] SCSI target mode selected by DIP switch, acting as an SCSI disk
[21ms] SD card detected, FAT32 volume size: 30436 MB
[21ms] SD MID: 0x03, OID: 0x53 0x44
[22ms] SD Name: SD32G
[22ms] SD Date: 9/2016
[22ms] SD Serial: 0x3B589DFD
[25ms] Reading configuration from zuluscsi.ini
[25ms] Active configuration:
[30ms] -- SelectionDelay: 0
[36ms] -- EnableSCSI2 is on
[1008ms] Finding HDD images in directory /:
[1009ms] -- Opening /HD0_256.img for id:0 lun:0
[1021ms] ---- NOTE: File /HD0_256.img has FAT allocated size of 227200 sectors and file size of 227180 sectors
[1023ms] ---- Due to issue #80 in ZuluSCSI version 1.0.8 and 1.0.9 the allocated size was mistakenly reported to SCSI controller.
[1024ms] ---- If the drive was formatted using those versions, you may have problems accessing it with newer firmware.
[1025ms] ---- The old behavior can be restored with setting  [SCSI] UseFATAllocSize = 1 in zuluscsi.ini
[1031ms] ---- Image file is contiguous, SD card sectors 3630080 to 3857259
[1032ms] ---- Read prefetch enabled: 8192 bytes
[1081ms] SCSI ID:0 BlockSize:256 Type:0 Quirks:0 ImageSize:113590kB
[1586ms] Initialization complete!
[1586ms] Platform: ZuluSCSI RP2040
[1586ms] FW Version: 1.1.1-release Oct 25 2022 12:54:05
[6659ms] DBG BUS RESET
[10119ms] DBG ---- SELECTION: 0
[10120ms] DBG ---- MESSAGE_OUT
[10120ms] DBG ------ OUT: 0xC0 
[10120ms] DBG ---- COMMAND: Inquiry
[10121ms] DBG ------ OUT: 0x12 0x00 0x00 0x00 0x24 0x00 
[10121ms] DBG ---- DATA_IN
[10122ms] DBG ---- Total IN: 36 OUT: 0 CHECKSUM: 33334
[10122ms] DBG ---- STATUS: 0 GOOD
[10130ms] DBG ---- MESSAGE_IN
[10130ms] DBG ------ IN: 0x00 
[10130ms] DBG -- BUS_FREE
[10131ms] DBG ---- SELECTION: 0
[10131ms] DBG ---- MESSAGE_OUT
[10131ms] DBG ------ OUT: 0xC0 
[10132ms] DBG ---- COMMAND: Inquiry
[10132ms] DBG ------ OUT: 0x12 0x01 0x00 0x00 0xFF 0x00 
[10133ms] DBG ---- DATA_IN
[10134ms] DBG ---- Total IN: 255 OUT: 0 CHECKSUM: 50561
[10134ms] DBG ---- STATUS: 0 GOOD
[10138ms] DBG ---- MESSAGE_IN
[10138ms] DBG ------ IN: 0x00 
[10139ms] DBG -- BUS_FREE
[10139ms] DBG ---- SELECTION: 0
[10139ms] DBG ---- MESSAGE_OUT
[10140ms] DBG ------ OUT: 0xC0 
[10140ms] DBG ---- COMMAND: Inquiry
[10140ms] DBG ------ OUT: 0x12 0x01 0x80 0x00 0xFF 0x00 
[10141ms] DBG ---- DATA_IN
[10142ms] DBG ---- Total IN: 255 OUT: 0 CHECKSUM: 52926
[10142ms] DBG ---- STATUS: 0 GOOD
[10147ms] DBG ---- MESSAGE_IN
[10147ms] DBG ------ IN: 0x00 
[10148ms] DBG -- BUS_FREE
[10148ms] DBG ---- SELECTION: 0
[10148ms] DBG ---- MESSAGE_OUT
[10149ms] DBG ------ OUT: 0xC1 
[10149ms] DBG ---- COMMAND: Inquiry
[10150ms] DBG ------ OUT: 0x12 0x00 0x00 0x00 0x24 0x00 
[10150ms] DBG ---- DATA_IN
[10151ms] DBG ---- Total IN: 36 OUT: 0 CHECKSUM: 25158
[10151ms] DBG ---- STATUS: 0 GOOD
[10156ms] DBG ---- MESSAGE_IN
[10156ms] DBG ------ IN: 0x00 
[10156ms] DBG -- BUS_FREE
[10156ms] DBG ---- SELECTION: 0
[10157ms] DBG ---- MESSAGE_OUT
[10157ms] DBG ------ OUT: 0xC2 
[10158ms] DBG ---- COMMAND: Inquiry
[10158ms] DBG ------ OUT: 0x12 0x00 0x00 0x00 0x24 0x00 
[10159ms] DBG ---- DATA_IN
[10159ms] DBG ---- Total IN: 36 OUT: 0 CHECKSUM: 25158
[10160ms] DBG ---- STATUS: 0 GOOD
[10164ms] DBG ---- MESSAGE_IN
[10164ms] DBG ------ IN: 0x00 
[10164ms] DBG -- BUS_FREE
[10165ms] DBG ---- SELECTION: 0
[10165ms] DBG ---- MESSAGE_OUT
[10166ms] DBG ------ OUT: 0xC3 
[10166ms] DBG ---- COMMAND: Inquiry
[10166ms] DBG ------ OUT: 0x12 0x00 0x00 0x00 0x24 0x00 
[10167ms] DBG ---- DATA_IN
[10167ms] DBG ---- Total IN: 36 OUT: 0 CHECKSUM: 25158
[10168ms] DBG ---- STATUS: 0 GOOD
[10176ms] DBG ---- MESSAGE_IN
[10176ms] DBG ------ IN: 0x00 
[10177ms] DBG -- BUS_FREE
[10177ms] DBG ---- SELECTION: 0
[10177ms] DBG ---- MESSAGE_OUT
[10178ms] DBG ------ OUT: 0xC4 
[10178ms] DBG ---- COMMAND: Inquiry
[10179ms] DBG ------ OUT: 0x12 0x00 0x00 0x00 0x24 0x00 
[10179ms] DBG ---- DATA_IN
[10180ms] DBG ---- Total IN: 36 OUT: 0 CHECKSUM: 25158
[10180ms] DBG ---- STATUS: 0 GOOD
[10184ms] DBG ---- MESSAGE_IN
[10184ms] DBG ------ IN: 0x00 
[10184ms] DBG -- BUS_FREE
[10185ms] DBG ---- SELECTION: 0
[10185ms] DBG ---- MESSAGE_OUT
[10185ms] DBG ------ OUT: 0xC5 
[10186ms] DBG ---- COMMAND: Inquiry
[10186ms] DBG ------ OUT: 0x12 0x00 0x00 0x00 0x24 0x00 
[10187ms] DBG ---- DATA_IN
[10187ms] DBG ---- Total IN: 36 OUT: 0 CHECKSUM: 25158
[10188ms] DBG ---- STATUS: 0 GOOD
[10192ms] DBG ---- MESSAGE_IN
[10192ms] DBG ------ IN: 0x00 
[10193ms] DBG -- BUS_FREE
[10193ms] DBG ---- SELECTION: 0
[10193ms] DBG ---- MESSAGE_OUT
[10194ms] DBG ------ OUT: 0xC6 
[10194ms] DBG ---- COMMAND: Inquiry
[10194ms] DBG ------ OUT: 0x12 0x00 0x00 0x00 0x24 0x00 
[10195ms] DBG ---- DATA_IN
[10196ms] DBG ---- Total IN: 36 OUT: 0 CHECKSUM: 25158
[10196ms] DBG ---- STATUS: 0 GOOD
[10200ms] DBG ---- MESSAGE_IN
[10201ms] DBG ------ IN: 0x00 
[10201ms] DBG -- BUS_FREE
[10201ms] DBG ---- SELECTION: 0
[10202ms] DBG ---- MESSAGE_OUT
[10202ms] DBG ------ OUT: 0xC7 
[10202ms] DBG ---- COMMAND: Inquiry
[10203ms] DBG ------ OUT: 0x12 0x00 0x00 0x00 0x24 0x00 
[10203ms] DBG ---- DATA_IN
[10204ms] DBG ---- Total IN: 36 OUT: 0 CHECKSUM: 25158
[10204ms] DBG ---- STATUS: 0 GOOD
[10209ms] DBG ---- MESSAGE_IN
[10209ms] DBG ------ IN: 0x00 
[10209ms] DBG -- BUS_FREE
[10997ms] DBG -- BUS_BUSY
[10998ms] DBG -- BUS_FREE
[11790ms] DBG -- BUS_BUSY
[11790ms] DBG ---- SELECTION: 0
[11790ms] DBG ---- MESSAGE_OUT
[11790ms] DBG ------ OUT: 0xC0 
[11791ms] DBG ---- COMMAND: ReadCapacity
[11791ms] DBG ------ OUT: 0x25 0x00 0x00 0x00 0x00 0x00 
[11792ms] DBG ------ OUT: 0x00 0x00 0x00 0x00 
[11792ms] DBG ---- DATA_IN
[11793ms] DBG ---- Total IN: 8 OUT: 0 CHECKSUM: 30741
[11793ms] DBG ---- STATUS: 0 GOOD
[11801ms] DBG ---- MESSAGE_IN
[11801ms] DBG ------ IN: 0x00 
[11801ms] DBG -- BUS_FREE
[11801ms] DBG ---- SELECTION: 0
[11802ms] DBG ---- MESSAGE_OUT
[11802ms] DBG ------ OUT: 0xC0 
[11802ms] DBG ------ OUT: 0x01 
[11803ms] DBG ------ OUT: 0x03 
[11803ms] DBG ------ OUT: 0x01 
[11803ms] DBG ------ OUT: 0x19 
[11804ms] DBG ------ OUT: 0x08 
[11804ms] SCSI ID 0 negotiated synchronous mode 10 MB/s (period 4x25 ns, offset 8 bytes)
[11805ms] DBG ---- MESSAGE_IN
[11805ms] DBG ------ IN: 0x01 0x03 0x01 0x19 0x08 
[11806ms] DBG ---- COMMAND: Read10
[11806ms] DBG ------ OUT: 0x28 0x00 0x00 0x00 0x00 0x00 
[11807ms] DBG ------ OUT: 0x00 0x00 0x02 0x00 
[11807ms] DBG ------ Read 2x256 starting at 0
[11808ms] DBG ---- DATA_IN, syncOffset 8 syncPeriod 25
assertion "(uint64_t)sectorcount * SD_SECTOR_SIZE == count" failed: file "src/ZuluSCSI_disk.cpp", line 189, function: int ImageBackingStore::read(void*, size_t)

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 000000A1
R1   : 100352A8
R2   : 2001839C
R3   : 10028659
R4   : 1003022A
R5   : 000000BD
R6   : 00000000
R7   : 00000000
R8   : 00000000
R9   : 00000000
R10  : 00000000
R11  : 00000000
R12  : 000000A0
SP   : 200183A8
LR   : 10029839
PC   : 10028658
xPSR : 21000000
PSP  : 20018388
MSP  : 2003BFC0
CPUID: 410CC601
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

[11817ms] --------------
[11817ms] CRASH!
[11817ms] Platform: ZuluSCSI RP2040
[11817ms] FW Version: 1.1.1-release Oct 25 2022 12:54:05
[11818ms] error_status: 0x80FF013D
[11818ms] error_address: 0x10028658
[11819ms] error_value: 0x2001CDD8
[11819ms] STACK 0x200183A8:    0x1003529B 0x10030270 0x10030270 0x200198C4
[11820ms] STACK 0x200183B8:    0x00000000 0x20001C29 0x00000000 0x2001A228
[11820ms] STACK 0x200183C8:    0x00002004 0x2001A214 0x00000000 0x20003369
[11821ms] STACK 0x200183D8:    0x00000100 0x00000020 0x200198C4 0x00010174
[11822ms] STACK 0x200183E8:    0x0000039D 0x00000000 0x00010174 0x20021110
[11823ms] STACK 0x200183F8:    0x00000000 0x200014ED 0x00000001 0x00000900
[11823ms] STACK 0x20018408:    0x00000000 0x00000000 0x00000000 0x20004A2B
[11824ms] STACK 0x20018418:    0x00000000 0x2000DB45 0x00000000 0x1002269F
`

PetteriAimonen, 2022-11-09:
Indeed, looks like there are some assumptions that break with 256 byte sectors.

PetteriAimonen, 2022-11-09:
There is now a test build here (download zip at bottom of page):
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/3426537572

If you could try it, that would be great.
It seems my current test PC doesn't support 256 byte sectors on the host side.

Deftaudio, 2022-11-09:
Thank you for the reply. The problem is solved! I'm seeing some host specific problems (this is early Fairlight CMI with 256b transfers) but previously I was able to fix it by adjusting delay parameter. But it certainly boots of ZuluSCSI RPI now. Thank you.

PetteriAimonen, 2022-11-10:
Great to hear that it helped!

closed 2022-11-10 by PetteriAimonen

aperezbios, 2022-11-10:
@Deftaudio I've now merged the 256 byte fix, and this fix will be part of the next firmware release.

PR #98 (2022-11-10): Fix problems with 256 byte sector size (#97)
The raw mapping only works when SCSI sector size is a multiple of SD card sector size.
Otherwise we must go through the SdFat code to convert smaller reads / writes.

This pull request fixes #97.