ZuluSCSI/ZuluSCSI-firmware #123: Return value of SCSI-1 Inquiry Data

Kurist2010, 2022-12-17:
Please set the return value of Inquiry when EnableSCSI2=0 in zuluscsi.ini setting to 1 in the 3rd byte of Inquiry Data. (Now 2 is the return value.)

The HDD used in the SCSI-1 HA (PC-9801-55) sold long ago in the PCI SCSI-2 HA (Adaptec AHA-2940UW/J) of the Japanese retro PC NEC PC-9821 It is used to determine whether it is connected. So I want you to reproduce it.

PetteriAimonen, 2022-12-17:
So this line here:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/main/lib/SCSI2SD/src/firmware/inquiry.c#L29

If we decide to do this, should add a check based on `scsiDev.boardCfg.flags & S2S_CFG_ENABLE_SCSI2` in `s2s_getStandardInquiry`.

Kurist2010, 2022-12-17:
I don't think it's a problem.

closed 2023-05-02 by PetteriAimonen