ZuluSCSI/ZuluSCSI-firmware #127: ZuluSCSI doesn't support more than one LUN?

PacificState, 2022-12-23:
TL;DR: despite what the manual says, from the code ZuluSCSI apparently doesn't support more than one LUN right now:

src/ZuluSCSI_config.h:#define NUM_SCSILUN 1          // Maximum number of LUNs supported     (Currently has to be 1)

What would it take to enhance the firmware to support this? Otherwise, the original report I'd filed is below:

---

I have a Zulu SCSI card installed in an Atari MegaSTE. Despite the restrictions imposed by the internal SCSI adapter (only a single device with ID 0 is supported, no parity support, and a 1GB limit), a single 1GB image partitioned into two works fine - with the HDDRIVER software, I can boot and read/write from it successfully.

To get around the SCSI device limit, the Atari will (apparently) recognize multiple LUNs. I therefore duplicated the original image file a further three times, and tweaked the filenames to hang four unit numbers off ID 0 - however, this doesn't work as the boot log indicates the SCSI ID is already in use:

[0ms] Platform: ZuluSCSI v1.1
[0ms] FW Version: 1.2.0-release Dec 21 2022 18:21:02
[0ms] DIPSW3 is ON: Enabling SCSI termination
[2ms] Optional GreenPAK detected, loading firmware
[53ms] GreenPAK firmware successfully loaded
[170ms] SD card detected, FAT32 volume size: 30436 MB
[170ms] SD MID: 0x03, OID: 0x53 0x44
[171ms] SD Name: SD32G
[171ms] SD Date: 9/2022
[171ms] SD Serial: 0xFD9D636D
[174ms] Config file zuluscsi.ini not found, using defaults
[174ms] Active configuration:
[176ms] -- SelectionDelay: 255
[178ms] -- EnableSCSI2 is on
[353ms] Finding HDD images in directory /:
[354ms] -- Opening /HD01_512.hda for id:0 lun:0
[552ms] ---- Image file is contiguous, SD card sectors 4225792 to 6322943
[552ms] ---- Read prefetch enabled: 8192 bytes
[553ms] -- Ignoring /HD03_512.hda, SCSI ID 0 is already in use!
[554ms] -- Ignoring /HD00_512.hda, SCSI ID 0 is already in use!
[555ms] -- Ignoring /HD02_512.hda, SCSI ID 0 is already in use!
[565ms] SCSI ID:0 BlockSize:512 Type:0 Quirks:0 ImageSize:1048576kB
[1070ms] SCSI PHY operating mode: GREENPAK_DMA
[1070ms] Initialization complete!

Have I misunderstood that a single SCSI can have multiple LUNs, or is this a bug?

PetteriAimonen, 2022-12-24:
There is `MapLunsToIDs` option in ini file for use with such devices.
You would make images with e.g. IDs 0, 1, 2, and when the host requests LUN2, it gets mapped to ID2.

The LUN limitation comes from the SCSI2SD codebase that we are using, and would be relatively difficult to add full support for multiple LUNs. The mapping solution should work for devices that only support single ID.

PacificState, 2022-12-24:
Thanks! How do you specify the base ID, though? Is the assumption that HD0.hda

PetteriAimonen, 2022-12-24:
I think it will respond to all of the SCSI IDs, so if you have images HD0, HD1, HD2, the host can access ID0 LUN0, or ID1 LUN0, or ID2 LUN0 and they all lead to same device. But I haven't tested this, so this is just my understanding of the code.

PacificState, 2022-12-24:
That worked perfectly, thanks. I guess the fix for the issue is to add a note to the documentation? It's definitely a bit confusing that the LUN syntax is available.

aperezbios, 2023-06-02:
@PacificState the README now states "Currently, ZuluSCSI does not support multiple LUNs, only LUN 0"

closed 2023-06-02 by aperezbios

wohali, 2023-11-24:
@aperezbios while working with obscure hardware -- the TI Explorer only supports SCSI IDs 0-3 and two LUNs per SCSI ID, i'd like to request this issue be reopened. Otherwise it is not possible to max out storage on these weird platforms, which is essential to doing things like building large, complex software (since max supported drive size is like 300MB per drive from the OS).

Would you consider opening a new issue or re-opening this one? Thanks.