ZuluSCSI/ZuluSCSI-firmware #131: Abysmal speeds around 112kB/s (RP2040, Amiga 500, A590 SCSI-controller, TF536 (030 50Mhz)

aiobofh, 2023-01-24:
I'm using ZuluSCSI RP2040 and have tried fiddling with various settings. I currently believe I have a "correct" set-up. HD1_512_System.img, HDToolBox with 0xFFFFFF MaxTransfer and i only get around 112kB/s which seems awfully low if I compare with the performance other people seem to get out of it.

I have tried with HD1_4096.img too and created partitions and formatted. About the same result. 

(Using a 128GB microSD-card in an adapter (which i suppose should not cause any issues))

My zuluscsi.ini
`[SCSI]
Debug = 0
SelectionDelay = 255
PhyMode = 0
EnableSCSI2 = 1
PrefetchBytes = 8192

[SCSI1]
Vendor = "FOOBAR"
Version = "1.0"
Product = "System"
Type = 0
`

And my zululog.txt
`[10ms] Platform: ZuluSCSI RP2040
[10ms] FW Version: 1.2.0-release Dec 21 2022 18:21:45
[11ms] DIP switch settings: debug log 0, termination 1
[11ms] SCSI termination is enabled
[12ms] Flash chip size: 2048 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[21ms] SD card detected, FAT64 volume size: 121877 MB
[21ms] SD MID: 0x03, OID: 0x53 0x44
[21ms] SD Name: SD128
[22ms] SD Date: 7/2022
[22ms] SD Serial: 0xFA3BCEB4
[24ms] Reading configuration from zuluscsi.ini
[25ms] Active configuration:
[27ms] -- SelectionDelay: 255
[30ms] -- EnableSCSI2 is on
[358ms] Finding HDD images in directory /:
[358ms] -- Opening /HD3_4096_Data.img for id:3 lun:0
[359ms] ---- Image file is contiguous, SD card sectors 139529728 to 143724031
[360ms] ---- Read prefetch enabled: 8192 bytes
[361ms] -- Opening /HD1_512_System.img for id:1 lun:0
[362ms] ---- Image file is contiguous, SD card sectors 134286848 to 135335423
[363ms] ---- Read prefetch enabled: 8192 bytes
[363ms] -- Opening /HD2_4096_Software.img for id:2 lun:0
[365ms] ---- Image file is contiguous, SD card sectors 135335424 to 139529727
[365ms] ---- Read prefetch enabled: 8192 bytes
[378ms] -- Platform supports ROM drive up to 1692 kB
[379ms] ---- ROM drive image not detected
[379ms] SCSI ID:1 BlockSize:512 Type:0 Quirks:0 ImageSize:524288kB
[380ms] SCSI ID:2 BlockSize:4096 Type:0 Quirks:0 ImageSize:2097152kB
[381ms] SCSI ID:3 BlockSize:4096 Type:0 Quirks:0 ImageSize:2097152kB
[881ms] Initialization complete!
[8111ms] WARNING: Host used command 0x1A which is affected by drive geometry. Current settings are 63 sectors x 255 heads = 16065 but image size of 1048576 sectors is not divisible. This can cause error messages in diagnostics tools.
[8196ms] WARNING: Host used command 0x1A which is affected by drive geometry. Current settings are 63 sectors x 255 heads = 16065 but image size of 524288 sectors is not divisible. This can cause error messages in diagnostics tools.
[8358ms] WARNING: Host used command 0x1A which is affected by drive geometry. Current settings are 63 sectors x 255 heads = 16065 but image size of 524288 sectors is not divisible. This can cause error messages in diagnostics tools.
`

What am I missing? I would expect a 68030@50MHz 64MB FastRAM would be good enough for way faster speeds.

aperezbios, 2023-01-24:
@aiobofh If you haven't already, you might want to try another SD card, just to rule things out. It would also be very helpful to have a debug log of a boot. You'd need to turn everything off, enable the debug DIP switch on the ZuluSCSI (or set debug=1 in your zuluscsi.ini file, but be sure to turn it off after you're done) and then turn it on and boot up. Then shut down, collect the zululog.txt file (which will be much, much larger) and then compress it and upload it here.

aiobofh, 2023-01-24:
Tnx @aperezbios - I attach a zip:ed debug log. I was running a full disk-speed-test with SysInfo while this was logging so you can see everything.
[zululog2.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/10493754/zululog2.zip)

aperezbios, 2023-01-26:
@PetteriAimonen any thoughts here?

PetteriAimonen, 2023-01-27:
From the logs it appears the raw SCSI bus transfer speed is about 120 kB/s, which is of course a lot slower than expected.
This rules out the SD card as a culprit, instead it must be something with the SCSI configuration.

For most hosts in asynchronous mode, the SCSI transfer speed is 4-5 MB/s.
In e.g. [this review](https://www.pureamiga.co.uk/2022/05/20/zuluscsi-review/) the Amiga SysInfo reports a speed of 4.4 MB/s - even though that is the GD32 version, it shouldn't matter for this.

I don't know much about Amiga, but by Googling there doesn't appear to be any settings that would affect this.

@aiobofh Have you used the same hardware with any other SCSI drives? If you have any available, that could help troubleshoot where the problem is.

aiobofh, 2023-01-27:
@PetteriAimonen i did try with a mechanical Seagate disk too. Same issue. BUT!!! I think i figured it out. DMA transfers are only possible between the SCSI-controller and the on-board FastRAM inside the A590. Since I have the TF536 accelerator with its own FastRAM (with higher priority) this impacts the transfer speed, since DMA can not be used. I'll do a check on the actual target hardware for the ZuluSCSI which has no accelerator (Obviously it wont be super fast either but should exceed the 112kB/s). I'll report back. Could be a good knowledge in general, i guess, if that's so.

PetteriAimonen, 2023-01-27:
@aiobofh Good to hear that there is at least a potential explanation :)

SezaroSteve, 2023-01-28:
I have an A500 Plus with a TF536. I use it with a GVP Impact Series II A500-HD8+ and ZuluSCSI RP2040 with Firmware 1.2.0
The performance is a bit lower than it was without the TF, in other words, with the stock 68000.
When the RAM hardware inside the GVP is disabled using the onboard jumpers (leaving only 2MB ChipRAM and 64MB FastRAM on the TF), the performance is terrible. As it turns out, the TF does not play nice with the DMA capable GVP. It does not do DMA at all. Or uses Chip RAM for DMA which is super slow.
With the RAM in the GVP enabled and using the correct MASK in HDtoolBox, the GVP is much faster but still a little bit slower than with a stock 68000.
The TF can do very fast HDD transfers with it's own IDE controller. I get 5.4 MB/s of it then with a good CF-Card. But it hogs the CPU.
With the GVP and its DMA enabled, despite doing a measly 1.1 MB/s in comparison, the system actually feels snappier, more responsive. This is because the CPU is not burdened as much with disk I/O. I therefore decided to keep using the GVP as in real-life, the system feels nicer to work with.

There is an extensive discussion about this issue here:  https://eab.abime.net/showthread.php?t=113167
RP2040 Performance info here:  https://github.com/ZuluSCSI/ZuluSCSI-firmware/discussions/133

aiobofh, 2023-01-29:
@SezaroSteve brilliant contribution. That's what I guessed then. Haven't tried mine out yet to confirm my issue is the same tho. I'll try to find some time to fiddle with it during the week. 

Right.... The Mask is an interesting thing. Have never had to bother with it, since I always had the interface on the accelerator, before. I could try to fiddle with that too, just to learn. I guess the Mask is AND:ed to the address space you want to handle the transfers from the SCSI device. And in the case of A590/HD8+ and alike this is where the magic happens with DMA. However.... The data still need to leave the 24bit DMA FastRAM to reach the CPU at some point... :) 

But again. Thanks for the logic boost on this issue.