ZuluSCSI/ZuluSCSI-firmware #140: FR: Ability to operate in standalone image mode.

PrimalNaCl, 2023-02-08:
I thought I had seen this mentioned as a potential feature for rascsi but I can't seem to find it atm nor in zulu/blue/macsd.

Is/could there be support for zulu scsi to come up in controller mode, interrogate connected ids and back-to-front dump them to the SD card?  This would make it a bit easier to archive several drives and/or those w/out scsci controllers or systems that readily support scsi controllers.  (e.g. no pci/isa slots)

aperezbios, 2023-02-08:
ZuluSCSI RP2040 has initiator support, and can do this today.

PrimalNaCl, 2023-02-08:
> ZuluSCSI RP2040 has initiator support, and can do this today.

I _thought_ I had seen this before.  But when looking at the manual [here ](http://zuluscsi.com/ZuluSCSI-Manual), it's not mentioned so unfortunate conflation with likely prior sighting of [this ](https://github.com/PiSCSI/piscsi/issues/454) wrt rascsi.

I _DO_, now/again, see it in the repo's readme [here](https://github.com/ZuluSCSI/ZuluSCSI-firmware#scsi-initiator-mode).

Apologies for the trouble/confusion and thank you for your understanding/kindness!

closed 2023-02-08 by PrimalNaCl