ZuluSCSI/ZuluSCSI-firmware #146: Needs Help To configure Zuluscsi RP 2040

Nicols07, 2023-02-20:
Hi,
Sorry for my english, i'm french !

At the first time, we have a SCSI floppy disk and SCSI HDD
At the second time, we have one scsi2sd configure in floppy mode and another one scsi2sd configure in HDD mode with terminaison. 
Configuration were save in the card with good configuration. In the floppy scsi2sd we insert a sd card with a floppy partition only when we want to load and save parameters or reinstall the system. In the HDD scsi2sd we have a SD card with the system file.

Now, i want to test two ZuluSCSI RP 2040 because we can't find scsi2sd card..., but i can't understand how i can configure my system...

For floppy, I need to put a empty FD0.img on my SD card for floppy disk with zuluscsi.ini for information if i need. But when i remove SD card and i reboot the system. How zuluscsi knows what is the good parameters like floppy configuration without sd card ?
For HDD, I need to put a empty HD1.img with good size like 358Mo for example and zuluscsi.ini if i need. But when system format and install system on SD card, what the system do when i reboot or switch off the system ?

Maybe i don't understand something...

Tomorrow i try to connect zuluscsi with floppy scsi and hdd scsi in initiator mode to check HDxx_imaged.hda .

In complement : in zuluscsi manual two references are missing ... http://zuluscsi.com/ZuluSCSI-Manual#using-transferring , http://zuluscsi.com/ZuluSCSI-Manual#limitations-formatting

Thx for information and help ...

MeTalOneOEight, 2023-02-21:
Maybe you should start understanding that a floppy emulator like hxc2001 is not the same as zuluscsi or scsi2sd, even they use SD cards.

On the zulu you can put a floppy image to be presented as SCSI drive, not as a floppy drive.
With the RP2040 you can store an image in the ROM  so that would be available without SD card.

If you have problems with the translation maybe try deepl.com, it's an AI based translation website.

PetteriAimonen, 2023-02-21:
If I understand correctly, you want a SCSI emulated floppy drive, but the floppy should be present only sometimes?

There is currently no specific support for "floppy drive present but floppy removed" in ZuluSCSI.

One way could be to put a file `FD0.rom` once on the SD card to program it to RP2040 SPI flash.
This floppy image will then be presented whenever you boot without SD card.
The SD card can then have `FD0.bin` which will override the ROM image when inserted.
More info [here](https://github.com/ZuluSCSI/ZuluSCSI-firmware#rom-drive-in-microcontroller-flash)

PetteriAimonen, 2023-02-23:
There is no technical limit, the configuration could be stored in the flash memory chip instead of on the card.
It's more of a question of coming up with a feature set that is easy to use and fits widest user base.

Do I understand correctly that the main missing feature is emulating a floppy drive with no floppy inserted?
The existing ROM drive support should work fine for making sure a floppy drive appears on boot even without SD card.

Nicols07, 2023-02-23:
Hi thx for you reply and support.

Absolutely, there must not be any floppy disk inserted in the floppy drive during a power outage, because the system would not boot from the hard disk.

[image]

Solution will be to create a virtual floppy drive empty like scsi2sd and when i want to boot, load or save i insert a sd card with a floppy size partition to do what i want.

If you think it's possible, tell me and i can do test.

Regards.
Nicolas.

PetteriAimonen, 2023-02-24:
Thanks, that clears it up. It will be some time until I can get to this, and I'm not yet sure what will make most sense for implementation.

PetteriAimonen, 2023-02-24:
Actually, I *think* this may be possible to do with a minimal code change.

If you have time to test, do this:

1. Download test build from https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/4260151256 (zip at bottom of page) and put the `ZuluSCSI_RP2040-...bin` on the SD card to upgrade.
2. Create `FD0.bin` with a size that is not zero but is less than 512 bytes. E.g. you can type `empty` with notepad to the file.
3. Test that the floppy drive appears empty when you boot.

If that works, you can then do the same with the image name being `FD0.rom` and it will be remembered even without SD card.

Nicols07, 2023-02-24:
Sorry i hadn't time to test this. 
I'll try monday.

Have a nice weekend.

Nicols07, 2023-03-10:
Sorry for my late answer. I had to wait to receipt a USB to RS232 cable because when i'm using the computer console of our automation engineer, I deleted all the partitions from his computer….
So i tested the configuration with FD0, with a empty img the system don't boot. With a img of an empty floppy disk, that's works but system write inside the img and it's not the good solution for me. Because i need to modify this file simply and with this configuration i must use a software to extract the file inside the img, then modifiy the file, then upload file in the img.
It's for that, SCSI2SD was more appropriated for my utilisation.
Final decision was to conserve the SCSI floppy drive and replace SCSI HDD.
Thx for your help and for this product.
Regards.

Nicols07, 2023-03-10:
i forgot to tell, i had some problem with new Sandisk SD card when i formated her, after differents tests. 
I need to delete partition with a specialised software for could use again. 
And sometimes SD Card was detected but FD0 or HD1 img didn't load at boot. 
That worked with same img but with another one SD card....

aperezbios, 2023-03-31:
@Nicols07 I'm not sure I follow your most recent comment. Are you still having issues on this front?

aperezbios, 2023-04-12:
Closing, as no response for over two weeks from original reporter.