ZuluSCSI/ZuluSCSI-firmware #149: ZuluScsi Emu III XS/XP Samplers and Cd.iso's

cidermassive, 2023-03-02:
I am using a Zuluscsi Mini RP2040 with an Emu III XP Sampler, and a Kingston 8GB Class HC1 (SDC10/8GB). I experience the same issue with Sandisk Class4 cards - 2gb, 4gb, 8gb. I have been trying to get it to load a cd.iso files - but they seem to break the 'functioning' of Zuluscsi - I am using this iso image: https://archive.org/details/eiiix-e-mu-systems-volume-3-orchestral I download it to the sd card and rename it to HD1.hda. I copy this file and rename it HD2.img. I then boot the sampler and it sees both drives, both drives get mounted. I load a bank from the image into the sampler. All fine. Then I copy the file again on the sd card and rename it CD3.iso. So there are three identical files on the sd card (all the same iso file from above) just with different names - HD1.hda, HD2.img, CD3.iso. Once Cd3.iso is on the card, none of the drives work. I would not expect a 'faulty' cd rom iso file to corrupt the loading of the other two, previously working files. If I delete cd3.iso then it all works fine again. What I have also encountered is that in the case above, using class4 cards, that if you have subfolders on the card the card gets corrupted - and you loose all the subfolders. 

I am including 2 log files in case that is of use
First when it works fine, HD1.hda&HD2.img, and load a bank so you can see it working fine
Second adding the CD3.img and then it 'breaks' the other two images

[zuluscsi.ini.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/10871292/zuluscsi.ini.txt)
[zululog-HD1hda-HD2img.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/10871296/zululog-HD1hda-HD2img.txt)
[zululog-HD1hda-HD2img-CD3.iso.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/10871297/zululog-HD1hda-HD2img-CD3.iso.txt)
[zuluerr-HD1hda-HD2img-CD3iso.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/10871299/zuluerr-HD1hda-HD2img-CD3iso.txt)

PetteriAimonen, 2023-03-02:
Based on logs it seems there are two issues:

1. Data transfer to the sampler is breaking when used with the CD image
2. Error handling doesn't work correctly but instead crashes with `assertion "g_scsi_dma_state == SCSIDMA_IDLE" failed`

I think the first one may be related to SCSI block size. It is set to 2048 by default for CD images, because that is what most systems expect. But this sampler may be different.

Could you try the filenames `CD03_512.iso` and `CD03_1024.iso`? The 512 is what harddrives default to, while reading from the logs it appears the host has read 1024 bytes when in 2048 mode so that may be what it expects.

MeTalOneOEight, 2023-03-03:
According to
https://github.com/dagargo/emu3fs#mounting-cds-and-other-drives
the Emu III filesystem uses a 512 block size.

Wouldn't it be a more general solution to actually read out the block size of the ISO file than to assume/hardcode/parse it from the filename?

PetteriAimonen, 2023-03-03:
As far as I know the ISO file does not have a standard way of indicating the block size. Or rather, the CD-ROM physical format always uses 2352 byte sectors with 2048 bytes data and 304 bytes error correction, but what is represented to the operating system depends on CD drive implementation.

cidermassive, 2023-03-03:
Thank you for the replies

CDX_512.iso solves the issue in the case above  (see log)

I then changed the name to _1024 and then it didnt work (see log) 

I then added a different CD image, renamed to _512 and although seen in the list of available disks, it wouldnt load anything (see log - btw CD4 is the new CD)  (see log)
I then changed the name of this to _1024 (so a _512 named disk and a _1024 named disk on SD card (and then the _512 CD image was no longer seen and the _1024 image, although seen, and I could scroll thru banks on the CD, it wouldnt load anything) (see log)

This is defintely a step in the right direction in that, even if the disks dont work, the other disks are available and work.......previously, even if there was no zuluscsi crash, if i used CD.iso, none of the other hda disks would work....

[zululog_512.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/10880715/zululog_512.txt)
[zululog_1024.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/10880724/zululog_1024.txt)
[zululog_2cds_512.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/10880725/zululog_2cds_512.txt)
[zululog_512cd_1024cd.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/10880729/zululog_512cd_1024cd.txt)

MeTalOneOEight, 2023-03-03:
isoinfo reads the block size from iso_primary_descriptor.logical_block_size
https://github.com/Distrotech/cdrkit/blob/distrortech-cdrkit/genisoimage/iso9660.h

But don't know large the effort to implement it would be for the benefit.
Maybe adding something to manual how to determine the block size of an image would be enough.

PetteriAimonen, 2023-03-03:
Hmm, that is interesting. Indeed it might make sense to read that field - I wonder how real CD drives work, whether they check the disk to determine block size.

cidermassive, 2023-03-03:
CD_512.iso is a preferred solution, in that the good drives on the SD card still work and are accessible.......so with _512, when an iso image doest 'work' you can see the banks on the cd image, but nothing loads.....least then you know theres an 'issue' with the image and it cant be used...this to me is a better solution compared to the one where if theres an issue with an image, then nothing works at all.....ie CDx.iso......

MeTalOneOEight, 2023-03-04:
Found some text describing how a CD drive works with the data:
https://www.staff.uni-mainz.de/tacke/scsi/SCSI2-14.html

Determing the block size of an image depends on which OS you are using.
But usually any burning app or image mounter should be able to tell you.
When it comes to sampler images, maybe using the free version of chickensys translator can help you figuring out if the image is for your sampler.