ZuluSCSI/ZuluSCSI-firmware #168: Stability Issues RP2040

darkalpha84, 2023-04-01:
Greetings - i bought myself a ZuluSCSi RP2040.
Im using a 256 gb Sandisk card.
The device was bought to give my old P3 550 mhz new storage possibilities.

The PCI-SCSI adapter is a Tekram DC-390 with SCSI2 and sync 10MB/S transfer speed support.
I'm experiencing the same issues from this post:

https://github.com/ZuluSCSI/ZuluSCSI-firmware/issues/18

Im using a 50gb hdd image which is accepted by the controller and fdisk (DOS disk setup utility) can utilize all space.
Booting Windows ME only works every 2nd time. Like described in the other thread the led stays red. I have to power off the whole system and remove the PSU powercable. Booting then works.
Windows 2000 can't even fnish the setup process without a bluescreen concerning the filesystem.

zuluerr.txt comes up with this:

[152ms] SCSI ID:0 BlockSize:512 Type:0 Quirks:0 ImageSize:51199155kB
[253ms] Initialization complete!
[34988ms] SCSI ID 0 negotiated synchronous mode 10 MB/s (period 4x25 ns, offset 15 bytes)
[558070ms] SCSI ID 0 negotiated synchronous mode 10 MB/s (period 4x25 ns, offset 15 bytes)
[610271ms] --------------
[610271ms] WATCHDOG TIMEOUT, attempting bus reset
[610272ms] GPIO states: out 0x3BFCEEFF oe 0x178F3BFF in 0x3BFDECFF
[610273ms] STACK 0x2001A990:    0x00000001 0x00000001 0x2001F80C 0x2001D908
[610273ms] STACK 0x2001A9A0:    0x000003E8 0x2000CA43 0x2000D118 0x21000000
[610274ms] STACK 0x2001A9B0:    0x2001C908 0x20003DC9 0x00000200 0x00000008
[610275ms] STACK 0x2001A9C0:    0x2001BAE0 0x00010174 0x000000CD 0x00000000
[610276ms] STACK 0x2001A9D0:    0x00010174 0x00000000 0x00000000 0x2000166B
[610276ms] STACK 0x2001A9E0:    0x00000000 0x00000900 0x00000000 0x00000000
[610277ms] STACK 0x2001A9F0:    0x00000000 0x20005477 0x00000000 0x2000F0D9
[610278ms] STACK 0x2001AA00:    0x00000000 0x1002280F 0x00000000 0x1002285F
[625279ms] --------------
[625279ms] WATCHDOG TIMEOUT!
[625279ms] Platform: ZuluSCSI RP2040
[625280ms] FW Version: 23.03.30-release Mar 30 2023 16:53:18
[625280ms] GPIO states: out 0x3BFCEEFF oe 0x178F3BFF in 0x3BFDECFF
[625281ms] STACK 0x2001A990:    0x00000000 0x2001B958 0x2001DA4B 0x2001DB07
[625282ms] STACK 0x2001A9A0:    0x000003E8 0x2000CA43 0x2000CA42 0x21000000
[625282ms] STACK 0x2001A9B0:    0x2001C908 0x20003DD9 0x00000200 0x00000008
[625283ms] STACK 0x2001A9C0:    0x2001BAE0 0x00010174 0x000000CD 0x00000000
[625284ms] STACK 0x2001A9D0:    0x00010174 0x00000000 0x00000000 0x2000166B
[625285ms] STACK 0x2001A9E0:    0x00000000 0x00000900 0x00000000 0x00000000
[625285ms] STACK 0x2001A9F0:    0x00000000 0x20005477 0x00000000 0x2000F0D9
[625286ms] STACK 0x2001AA00:    0x00000000 0x1002280F 0x00000000 0x1002285F

Ive already tried disabling sync and used async mode - but no change.

Any suggestions? :)

Thanks :)

aperezbios, 2023-04-01:
@darkalpha84 thanks for the report. We need to know which firmware version you're running, and the log file output you've provided is truncated, and omits that detail. Can you let us know which firmware version is currently on your ZuluSCSI?

aperezbios, 2023-04-01:
@PetteriAimonen can you take a look at this crash log, at your convenience? Thanks.

aperezbios, 2023-04-01:
@darkalpha84 I attempted to replicate the behavior you reported, with the same firmware version, and am so far unable to replicate it. I'm using a PCI SCSI controller from the same 53C8xx family as your Tekram DC-390, which is the most similar PCI controller I have handy at the moment. if you could please provide us with a full, un-truncated Zululog.txt file, it will be helpful to us.

It's possible your SD card is not quite keeping up with the RP2040. Can you tell us which speed class it is marked as, as well as its size and make/model? Thanks!

PetteriAimonen, 2023-04-01:
The crash location is in [here](https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/main/src/ZuluSCSI_disk.cpp#L1838), waiting for a data transfer to host to finish.

Would it be possible to get a debug log? This can be enabled either by dip switch, or by putting the following in `zuluscsi.ini`, whichever is more convenient. The debug messages then go in `zululog.txt`.

    [SCSI]
    Debug = 1

If you are in interested in trying, `PrefetchBytes = 0` may make some difference in behavior.

darkalpha84, 2023-04-02:
Thank you a lot for the fast responses.
Informations requested:

The sd card used was a SanDisk Ultra 256 GB Micro SD XC I with a 10 in a circle. The upper part of the sd card is white and the lower part is grey. The specs say Class 10 120 MB/S.

The 2nd card i used to test was a SanDisk Ultra 32 GB MicroSD HC I Class 10 A1
Upper part of the card is red lower part is grey. Specs say 98 MB/S     Class 10     UHS Speed U1

I did some further testing and ill try to describe it as clear as possible.

The best and latest experience was with the 32 GB card.
i did debug log off
PrefetchBytes = 0
EnableSCSI2 = 1
EnableParity = 1
MaxSyncSpeed = 0
using an 8 gb hdd image.

I was able to install win2k.
Using win2k on the first boot after PC has been off for the night worked flawless. I did stresstesting two of the partitions with scandisk at the same time and all went fine while doing other stuff. I rebooted and before i hit the desktop i got bluescreen ntfs file system error. That reminds me of the problems with windows ME. 1st boot fine, 2nd crash. Although i didnt use the log dipswitch i got some zuluerr.txt output:

[10ms] Platform: ZuluSCSI RP2040
[10ms] FW Version: 23.03.30-release Mar 30 2023 16:53:18
[11ms] DIP switch settings: debug log 0, termination 1
[11ms] SCSI termination is enabled
[12ms] Flash chip size: 4096 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[38ms] SDIO SD card error on line 434, error code 2 [Device had power before SD card was inserted]
[41ms] SDIO SD card error on line 434, error code 2
[44ms] SDIO SD card error on line 434, error code 2
[47ms] SDIO SD card error on line 434, error code 2
[47ms] SD card init failed, sdErrorCode: 2 sdErrorData: 0
[10665ms] SD card init succeeded after retry  [SD card was put in]
[10665ms] SD card detected, FAT64 volume size: 29908 MB
[10666ms] SD MID: 0x6F, OID: 0x03 0x03
[10666ms] SD Name: SDABC
[10666ms] SD Date: 12/2022
[10667ms] SD Serial: 0xAA000124
[10668ms] Reading configuration from zuluscsi.ini
[10669ms] Active configuration:
[10669ms] -- SelectionDelay: 255
[10670ms] -- EnableSCSI2 is on
[10672ms] -- EnableParity is on
[10686ms] -- Opening /boot.img for id:0, specified in zuluscsi.ini
[10686ms] ---- Read prefetch disabled
[10779ms] Finding HDD images in directory /hdds/:
[10779ms] Could not open directory: /hdds/
[10780ms] Finding HDD images in additional directory Dir2 = "/cds/":
[10781ms] -- Could not open directory: /cds/
[10786ms] -- Platform supports ROM drive up to 3740 kB
[10787ms] ---- ROM drive image not detected
[10787ms] SCSI ID:0 BlockSize:512 Type:0 Quirks:0 ImageSize:8217247kB
[10888ms] Initialization complete!
assertion "g_scsi_dma_state == SCSIDMA_IDLE" failed: file "lib/ZuluSCSI_platform_RP2040/scsi_accel_rp2040.cpp", line 955, function: void scsi_accel_rp2040_setSyncMode(int, int)

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 000000B1
R1   : 10035656
R2   : 2001A954
R3   : 100286C5
R4   : 10032D4E
R5   : 000003BB
R6   : 20023828
R7   : 2001F78A
R8   : 00000000
R9   : 00000000
[... 91 lines cut ...]
[28297ms] DBG ------ OUT: 0xC0 
[28298ms] DBG ---- COMMAND: ReadCapacity
[28298ms] DBG ------ OUT: 0x25 0x00 0x00 0x00 0x00 0x00 
[28299ms] DBG ------ OUT: 0x00 0x00 0x00 0x00 
[28299ms] DBG ---- DATA_IN, syncOffset 15 syncPeriod 25
[28300ms] DBG ---- Total IN: 8 OUT: 0 CHECKSUM: 35851
[28300ms] DBG ---- STATUS: 0 GOOD
[28305ms] DBG ---- MESSAGE_IN
[28305ms] DBG ------ IN: 0x00 
[28305ms] DBG -- BUS_FREE
[28306ms] DBG ---- SELECTION: 0
[28306ms] DBG ---- MESSAGE_OUT
[28306ms] DBG ------ OUT: 0xC0 
[28307ms] DBG ---- COMMAND: WriteVerify
[28307ms] DBG ------ OUT: 0x2E 0x00 0x00 0x00 0x00 0x04 
[28308ms] DBG ------ OUT: 0x00 0x00 0x00 0x00 
[28308ms] DBG ------ Write 0x512 starting at 4
[28308ms] DBG ---- DATA_OUT, syncOffset 15 syncPeriod 25
[28309ms] DBG ---- Total IN: 0 OUT: 0 CHECKSUM: 0
[28310ms] DBG ---- STATUS: 0 GOOD

I'll add a compressed version of the logfiles .

Thanks :)
[zululogs issue 168.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11131748/zululogs.issue.168.zip)

PetteriAimonen, 2023-04-03:
Thanks for the logs!

There seems to be several things happening:

1. Prefetch of next sector data is ignoring BUS RESET from host. Just need to add a check for that, the workaround of `PrefetchBytes = 0` is good for now.
2. The host is issuing BUS RESETs every now and then. This could be indicative of parity errors in the transfer, but it could be some other kind of communication problem.
3. The message `assertion "g_scsi_dma_state == SCSIDMA_IDLE" failed` is similar to what was fixed in 4c1750726ea9a60, so probably a similar bug elsewhere.

I think there may be some underlying noise or cable problem here, which causes the host to issue a lot of retries and then it triggers bugs in the error recovery code. The error recovery bugs will be fixed, but it would be good to fix the communication problems also.

Some questions:
* How long is the SCSI cable?
* Are there other devices on the bus?
* Is the SCSI termination at each end of the bus? Usually host card has termination, and because your ZuluSCSI has termination enabled, it should be last device on the cable.
* Is ZuluSCSI powered from the SCSI cable only, or separately with the 4-pin connector?

darkalpha84, 2023-04-03:
Again thanks for your reply :)!
(1) The SCSI cable is about 50 cm long. It's the shortest ive got. It got 3 plugs only.
(2) The ZULUScsi is the only device on this 50 pin bus. On the back there is an iomega jaz 2gb connected with the 50 pin 8 bit external interface.
(3) The hostadapter is configured with termination on, as is the ZULU.
(4) Its only buspowered - i already tried it with the 4 pin connector too, but i felt it made no difference - you want me to have it connected to the PSU?

PetteriAimonen, 2023-04-03:
I noticed some of those messages in the debug logs as well, it seems the host wants to always execute `ReadCapacity` command in asynchronous mode.

It's a bit excessive to log every mode change like that, but most hosts only change the mode on boot and on error conditions. Shouldn't cause any problems except long logs.

darkalpha84, 2023-04-03:
You said - "I noticed some of those messages in the debug logs as well, it seems the host wants to always execute ReadCapacity command in asynchronous mode."

These happen with MaxSyncSpeed = 10 <- isnt that synchronous mode?
Testing with MaxSyncSpeed = 0 wont produce 

[208981ms] SCSI ID 0 negotiated synchronous mode 10 MB/s (period 4x25 ns, offset 15 bytes)
[208983ms] SCSI ID 0 negotiated synchronous mode 0 MB/s (period 4x0 ns, offset 0 bytes)

stuff...

PetteriAimonen, 2023-04-03:
Yeah, MaxSyncSpeed is just the maximum speed. The host can negotiate a slower speed. In this case the log messages look like: Set asynchronous mode (logged as 0 MB/s), do ReadCapacity, set synchronous mode 10 MB/s. I'm not sure why the host switches modes, maybe there is some ancient drive that couldn't do ReadCapacity with sync mode on :)

I cobbled together a 60 cm long cable and ran `random_tester.py` on Linux at 10MB/s sync mode for a while, no errors in logs. So it's not at least directly caused by cable length. But there could be some invisible corrosion or damage on the cable, so if you have another one it might be worth trying.

Here is a test build that should avoid crashing on these error conditions: https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/4596341235

darkalpha84, 2023-04-03:
Hi there - tested your firmware.
Im not getting a zuluerr.txt anymore - still booting Windows ME ended in the PC shutting off. Hitting the reset button had the effect, that the zulu scsi wasnt found anymore by the SCSI controller....
I had that before and had to cut off power completley.
Maybe the 50 gig hdd image is to much for the controller?

darkalpha84, 2023-04-04:
Hi there - tried win98 on usual IDE hdd - works perfect.
As soon as im trying scsi 98 wont load after finishing installing.
Stuck on the screen windows 98 is loading.

aperezbios, 2023-04-04:
@darkalpha84 please enable the debug DIP switch, and re-attempt to boot via SCSI. Then, once it hangs, send along the resulting zululog. We need to see the debug logs to be able to tell what's going on.

PetteriAimonen, 2023-04-05:
The theory that Win98 has some kind of problem with SCSI on your system may be true. There is nothing obvious in the log this time, it still has bus resets in 0.004% of transfers but seems to recover without problems.

I think I may try Win98 installation on my test system at some point, just to see how it behaves there. But from what I remember, it wasn't that rare for Win98 to just mysteriously hang on boot back in the 1990s either :D

darkalpha84, 2023-04-05:
Cool - im really interested if it works for you.

PetteriAimonen, 2023-04-06:
Looks like Win98SE setup doesn't even want to boot on my test system - perhaps it is too new (some Intel Atom from 2007). Not much more that I can do to debug this issue at the moment, but good to hear that it at least works reasonably ok with Win2k now.

closed 2023-04-06 by PetteriAimonen