ZuluSCSI/ZuluSCSI-firmware #175: Attribution Missing For BlueSCSI V2 Hardware Compatibility

androda, 2023-04-07:
Hello,

It looks like code changes for hardware compatibility with BlueSCSI V2 have been copied into this repo without attribution.

Here are some links to relevant changes, pointing to some (not all) code that appears to have been copied.
https://github.com/ZuluSCSI/ZuluSCSI-firmware/pull/167/files#diff-a815740a9895194e56d6f516a20f7b4397a1b36a77b36a841d0c9e9a7406acd3R106
https://github.com/ZuluSCSI/ZuluSCSI-firmware/pull/167/files#diff-006b5816f2d4fc3faa45939d225b33268ea9129143a01811700cdcd15663ac51R131
https://github.com/ZuluSCSI/ZuluSCSI-firmware/commit/1730c8e8cc0d635377c5745c9d0e1ace6c2e7d39#diff-9cae356e0af1d5e43056b5f4522b74aecb4b7de5b9387c4dc40aac8135ce3328R27

The source of these specific changes in the BlueSCSI V2 repository appears to be (respectively):
https://github.com/BlueSCSI/BlueSCSI-v2/commit/8971584485c42d1c711ca537c8be171de348582b#diff-a6c5cccd102e1fa0199b42d1f4c8beaefe78d3d2fdd5a878180c5b9782adf983R80
https://github.com/BlueSCSI/BlueSCSI-v2/commit/8971584485c42d1c711ca537c8be171de348582b#diff-7cdda8d0914c4f184e9a9e6c9ea45379dde760fdb0584d0b3dde0e254ee42eafR131
https://github.com/BlueSCSI/BlueSCSI-v2/commit/8971584485c42d1c711ca537c8be171de348582b#diff-f2c252c6c720fc30e782528c05344192822d1649f8bc166bae0baf920f7987fdR29

Would you be so kind as to restore attribution for BlueSCSI V2 hardware compatibility?

aperezbios, 2023-04-07:
Would you be so kind as to point out where we removed attribution, specifically? That would be a pre-requisite for "restoring" (in reality, adding) it.  See attached screenshot of BlueSCSI_platform_gpio.h that's currently in the BlueSCSI-v2 Git repository, which is derived from ZuluSCSI code, and where there's no attribution in the source header. To the best of my knowledge, we didn't remove attribution, it simply wasn't there in the first place. We'd be happy to _add_ attribution to non-trivial chunks of substantive, functional code you can demonstrate you actually wrote. Re-mapping pin assignments or re-naming functions does not appear to rise to the level of copyrightability. We're willing to be fair here, but we all know where the BlueSCSI-V2 (RP2040-specific) code base came from.

[image]

androda, 2023-04-07:
Thanks for your prompt response.  The attribution to which I refer is in the Git Log / Git History.  Below are three code locations from before, but in the history view:

https://github.com/ZuluSCSI/ZuluSCSI-firmware/blame/main/lib/ZuluSCSI_platform_RP2040/scsiPhy.cpp#L106

https://github.com/ZuluSCSI/ZuluSCSI-firmware/blame/main/lib/ZuluSCSI_platform_RP2040/ZuluSCSI_platform_gpio_BS2.h#L130

https://github.com/ZuluSCSI/ZuluSCSI-firmware/blame/1730c8e8cc0d635377c5745c9d0e1ace6c2e7d39/lib/ZuluSCSI_platform_BS2/ZuluSCSI_platform_gpio.h#L27

As you can see from the Git History view (called 'blame' here, as it shows who made the last change), it is showing that Petteri Aimonen and morio are the writers of the changes referenced in ZuluSCSI, when it appears that these changes were made by me in BlueSCSI V2.

When merging changes over from ZuluSCSI into BlueSCSI V2, care is taken to try and maintain this Git History Attribution. I would request the same courtesy when code changes are merged from BlueSCSI V2 into ZuluSCSI.

Picture [from the first link in ZuluSCSI repo](https://github.com/ZuluSCSI/ZuluSCSI-firmware/blame/main/lib/ZuluSCSI_platform_RP2040/scsiPhy.cpp#L106), showing authorship by Petteri and morio.
[image]

Picture from [the corresponding BlueSCSI V2 code change](https://github.com/BlueSCSI/BlueSCSI-v2/blame/8971584485c42d1c711ca537c8be171de348582b/lib/BlueSCSI_platform_RP2040/scsiPhy.cpp#L79), showing authorship by me and Eric.
[image]

aperezbios, 2023-04-07:
Thanks for being specific. I'll sync up with the team at some point on Monday, and we'll decide how to proceed. Petteri is in Finland, where it's already after 1AM.

androda, 2023-04-10:
Good day,

I believe proper attribution for all of the mentioned changes would be something to the effect of:
`BlueSCSI V2 Hardware Compatibility Changes (C) 2023, Tech by Androda, LLC`

Certainly not this:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/commit/1048627f8445c2677f736253b970791327aadd67
[image]

I disagree with your assertion that these changes do "not appear to rise to the level of copyrightability".  These changes have enabled a whole new hardware ecosystem to use a fork of the ZuluSCSI software - this is a significant usage.

PetteriAimonen, 2023-04-10:
@androda I have submitted #177 which brings the single related commit as part of the chain. It would be great if you can review and see if you consider that acceptable.

Morio originally rewrote the GPIO definitions in cleaner format and that is the code we will use in future also. But the numbers of course came from your file. It appears that github GUI doesn't follow the chain same way as e.g. `git log --follow ZuluSCSI_platform_gpio_BS2.h`, so the commit is not directly visible in the commit history of a renamed file.

closed 2023-04-10 by aperezbios