ZuluSCSI/ZuluSCSI-firmware #176: ZuluSCSI RP4020 Issue with Akai MPC 60 Mk2 Sampler

pbrs, 2023-04-10:
Hi everyone,

I am desperate to get my ZuluSCSI Board working in my Akai MPC 60 II Sampler/Drummachine.  It is connected to a SCSI-Breakoutoard by a vendor called "Jazzcat" which uses an AMD AM5380 SCSI interface controller. I have a SanDisk Ultra 32gb Class 10 SD-Card with a 56mb "HD5.ima" image file in the root directory. When powering the unit without an SD-card in it will search for an scsi hard-disk on start-up, then time out and work as usual. If I put the SD-card in it will freeze while "...searching for a SCSI hard disk". 

What I've tried:
- powering the board via usb instead of the internal power supply (does not power via scsi-port as no termination power is supplied by the controller board)
- tried a different SD-card
- tried multiple different .iso /.hda files.
- triple checked all connections
- tried all kinds of settings in the zuluscsi.ini

Unfortunately I don't have an alternative scsi drive to try.

This is the output of the zululog.txt with Debug-mode switched on:

[10ms] Platform: ZuluSCSI RP2040
[10ms] FW Version: 23.03.30-release Mar 30 2023 16:53:18
[11ms] DIP switch settings: debug log 1, termination 1
[11ms] SCSI termination is enabled
[12ms] Flash chip size: 4096 kB
[14ms] SCSI target/disk mode selected by DIP switch, acting as a SCSI disk
[20ms] SD card detected, FAT32 volume size: 29897 MB
[20ms] SD MID: 0x6F, OID: 0x03 0x03
[21ms] SD Name: SDABC
[21ms] SD Date: 12/2022
[21ms] SD Serial: 0xAA000080
[22ms] Config file zuluscsi.ini not found, using defaults
[22ms] Active configuration:
[23ms] -- SelectionDelay: 255
[23ms] -- EnableSCSI2 is on
[24ms] -- EnableParity is on
[33ms] Finding HDD images in directory /:
[33ms] -- Opening /HD5_512.IMA for id:5 lun:0
[52ms] DBG ---- Image file is contiguous, SD card sectors 41088 to 153087
[52ms] ---- Read prefetch enabled: 8192 bytes
[53ms] -- Platform supports ROM drive up to 3740 kB
[54ms] ---- ROM drive image not detected
[54ms] SCSI ID:5 BlockSize:512 Type:0 Quirks:0 ImageSize:56000kB
[155ms] Initialization complete!
[78849ms] DBG BUS RESET
[79773ms] DBG ---- SELECTION: 5
[79775ms] DBG ---- COMMAND: RequestSense
[79776ms] DBG ------ OUT: 0x03 0x00 0x00 0x00 0x00 0x00 
[79776ms] DBG ---- DATA_IN
[79777ms] DBG ---- Total IN: 4 OUT: 0 CHECKSUM: 30
[79778ms] DBG ---- STATUS: 0 GOOD

The long time between SD initialization and the first bus reset is caused because I powered on the zuluscsi via usb first and later the MPC. However I get the same output when powering via the internal power-adaptor.

Any help is greatly appreciated. Thanks in advance!

aperezbios, 2023-04-10:
@pbrs thanks for the debug log. From the perspective of the ZuluSCSI, everything seems to be working as expected. Since you don't have another SCSI device to test against, it's very possible you have a cabling issue. Can you describe how your ZuluSCSI is cabled up? Is it mounted internally? Externally? A photo of the cabling could also potentially be of use in further diagnosis.

pbrs, 2023-04-10:
Yea, thats what I suspected as well but can't really wrap my head of what could be wrong. There are multiple conversions from 50 pin to 26 pim connectors so possibly there is one installed incorrectly. Is there a way to check the correct wiring between the stages? I have a multimeter and a basic oscilloscope avaliable. 

Here are pictures of the wiring:
[image]
[image]

Thanks again for your help!

PetteriAimonen, 2023-04-10:
Because the "RequestSense" command came through, I think the cabling is at least somewhat ok. But with the long cabling and especially 26-pin cabling, there could be some noise issues. The 26 pin flat cables don't have the alternating GND pins like 50-pin cables, and they don't have a cable shield either like the DB25 cables.

Where does the other end of the 26-pin cable continue to? Is there bus termination at both ends of the bus (one at ZuluSCSI, other at whatever device is the last in the cable)?

If you have serial terminal program on PC, you could use it to view the log in realtime through USB.
Sometimes if a request is stuck, it may not get written to log file until 30 seconds later when watchdog timer expires.

From other MPC60 logs, the beginning of the log seems normal.
The next command should be "Read6", which is apparently where it gets stuck.

aperezbios, 2023-04-10:
From the second photo you provided, it looks like the 26-pin ribbon cable is also going to the rear of the unit, to the DB25 connector. If this is unterminated, which it likely is, the unterminated lines between the SCSI controller itself and the DB25 connector on the rear of the unit can and will act like an antenna, and can pick electromagnetic interference from your environment, even EMI generated within the sampler itself, by it's power supply or other internal electronics, particularly if it's not terminated properly.

Do you have any external SCSI cabling attached to the DB25 connector?

pbrs, 2023-04-10:
@PetteriAimonen 
1. the other side of the cable goes to an DB25 connector in the back of the unit, that is not populated. I did measure voltage on the 25th pin and it was at 4.7V. The cable came with the adaptorboard which the seller claimed works in his unit. 
2. I will try this. Is there a specific program you recommend? Do I just connect the Zulu via USB to do this? 

@aperezbios 

No, the DB25 connector is not connected to anything. Should I try to find a suitable cable directly from the adaptor to the ZuluSCSI?

pbrs, 2023-04-11:
I have downloaded a basic serial port monitor but I'm not sure which baudrate etc. I need to set it to to log the activity of the ZuluSCSI. Sorry if I missed it in the documentation.

PetteriAimonen, 2023-04-11:
It doesn't actually matter, because it is only software emulated serial port it doesn't care about what baudrate PC asks. You should see output as soon as the zulu is connected and you open the port.

Avoiding the empty branch in the cable could help, though usually SCSI is not that sensitive.

pbrs, 2023-04-13:
So, I've managed to log the serial output during startup and indeed it shows a little more before freezing:

[308897ms] DBG BUS RESET
[309821ms] DBG ---- SELECTION: 5
[309823ms] DBG ---- COMMAND: RequestSense
[309823ms] DBG ------ OUT: 0x03 0x00 0x00 0x00 0x00 0x00 
[309824ms] DBG ---- DATA_IN
[309825ms] DBG ---- Total IN: 4 OUT: 0 CHECKSUM: 30
[309825ms] DBG ---- STATUS: 0 GOOD
[309831ms] DBG ---- MESSAGE_IN
[309831ms] DBG ------ IN: 0x00 
[309832ms] DBG -- BUS_FREE
[309832ms] DBG ---- SELECTION: 5
[309834ms] DBG ---- COMMAND: Read6
[309834ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00 
[309835ms] DBG ------ Read 1x512 starting at 0
[309836ms] DBG ---- DATA_IN
[314839ms] scsi_accel_rp2040_stopWrite() timeout, FIFO levels 0 0 PC 23
[314839ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 63139
[314840ms] DBG -- BUS_FREE

Is there any hint onto why the MPC keeps freezing after this?

Thanks again!

PetteriAimonen, 2023-04-13:
Hmm, it appears that the host is not releasing ACK after the final byte.

Is the image known good? Or might there be something in the first sector that would make the MPC60 crash?

pbrs, 2023-04-13:
I just tried it again with a "freshly" formatted image. I noticed, that every now and then it will go through to the main screen, as soon as I try to access the drive again it will freeze however. Looking at the log output on the serial monitor it looks like something funky is going. This was a log where I first put the SD card with the "fresh" image file in, then booted "normally", I then tried to access the drive when it stalled at the same point as above. I then took out the SD erased the image and put the same file back on as initially and it got stuck again during the booting sequence but after ... searching for SCSI hard disk. it continued to "...reading files" and got stuck there.  

I am very thankful for you guys sticking with me on solving this. 

[308897ms] DBG BUS RESET
[309821ms] DBG ---- SELECTION: 5
[309823ms] DBG ---- COMMAND: RequestSense
[309823ms] DBG ------ OUT: 0x03 0x00 0x00 0x00 0x00 0x00 
[309824ms] DBG ---- DATA_IN
[309825ms] DBG ---- Total IN: 4 OUT: 0 CHECKSUM: 30
[309825ms] DBG ---- STATUS: 0 GOOD
[309831ms] DBG ---- MESSAGE_IN
[309831ms] DBG ------ IN: 0x00 
[309832ms] DBG -- BUS_FREE
[309832ms] DBG ---- SELECTION: 5
[309834ms] DBG ---- COMMAND: Read6
[309834ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00 
[309835ms] DBG ------ Read 1x512 starting at 0
[309836ms] DBG ---- DATA_IN
[314839ms] scsi_accel_rp2040_stopWrite() timeout, FIFO levels 0 0 PC 23
[314839ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 63139
[314840ms] DBG -- BUS_FREE
[2001714ms] DBG BUS RESET
[2002638ms] DBG ---- SELECTION: 5
[2002639ms] DBG ---- COMMAND: RequestSense
[2002639ms] DBG ------ OUT: 0x03 0x00 0x00 0x00 0x00 0x00 
[2002640ms] DBG ---- DATA_IN
[2002641ms] DBG ---- Total IN: 4 OUT: 0 CHECKSUM: 30
[2002642ms] DBG ---- STATUS: 0 GOOD
[2002646ms] DBG ---- MESSAGE_IN
[2002647ms] DBG ------ IN: 0x00 
[2002648ms] DBG -- BUS_FREE
[2002648ms] DBG ---- SELECTION: 5
[2002649ms] DBG ---- COMMAND: Read6
[2002649ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00 
[2002650ms] DBG ------ Read 1x512 starting at 0
[2002651ms] DBG ---- DATA_IN
[2007653ms] scsi_accel_rp2040_stopWrite() timeout, FIFO levels 0 0 PC 23
[2007653ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 63139
[2007654ms] DBG -- BUS_FREE
[23671989ms] DBG Timeout waiting for response in rp2040_sdio_command_R1(13), PIO PC: 13 RXF: 0 TXF: 0
[23671990ms] SDIO SD card error on line 217, error code 2
[23671993ms] DBG Timeout waiting for response in rp2040_sdio_command_R1(13), PIO PC: 14 RXF: 0 TXF: 0
[23671994ms] SDIO SD card error on line 217, error code 2
[23671994ms] SD card removed, trying to reinit
[23756477ms] SD card reinit succeeded
[23756477ms] SD card detected, FAT32 volume size: 29897 MB
[23756478ms] SD MID: 0x6F, OID: 0x03 0x03
[23756478ms] SD Name: SDABC
[23756478ms] SD Date: 12/2022
[23756479ms] SD Serial: 0xAA000080
[23756499ms] Reading configuration from zuluscsi.ini
[23756499ms] Active configuration:
[23756500ms] -- SelectionDelay: 0
[23756502ms] -- EnableSelLatch is on
[23756503ms] -- EnableParity is on
[23756605ms] Finding HDD images in directory /:
[23756606ms] -- Opening /HD00_512 MPC 56MB.IMA for id:0 lun:0
[23756625ms] DBG ---- Image file is contiguous, SD card sectors 41088 to 153087
[23756625ms] ---- Read prefetch enabled: 8192 bytes
[23756632ms] -- Platform supports ROM drive up to 3740 kB
[23756632ms] ---- ROM drive image not detected
[23756633ms] SCSI ID:0 BlockSize:512 Type:0 Quirks:0 ImageSize:56000kB
[23809188ms] DBG BUS RESET
[23810461ms] DBG BUS RESET
[23811748ms] DBG BUS RESET
[23813047ms] DBG BUS RESET
[23814347ms] DBG BUS RESET
[23815646ms] DBG BUS RESET
[23816946ms] DBG BUS RESET
[... 67 lines cut ...]
[24501010ms] DBG ------ OUT: 0x03 0x00 0x00 0x00 0x00 0x00 
[24501011ms] DBG ---- DATA_IN
[24501013ms] DBG ---- Total IN: 4 OUT: 0 CHECKSUM: 30
[24501013ms] DBG ---- STATUS: 0 GOOD
[24501017ms] DBG ---- MESSAGE_IN
[24501017ms] DBG ------ IN: 0x00 
[24501019ms] DBG -- BUS_FREE
[24501019ms] DBG ---- SELECTION: 0
[24501020ms] DBG ---- COMMAND: Read6
[24501021ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00 
[24501021ms] DBG ------ Read 1x512 starting at 0
[24501022ms] DBG ---- DATA_IN
[24501024ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 63139
[24501024ms] DBG ---- STATUS: 0 GOOD
[24501028ms] DBG ---- MESSAGE_IN
[24501028ms] DBG ------ IN: 0x00 
[24501028ms] DBG -- BUS_FREE
[24501040ms] DBG ---- SELECTION: 0
[24501041ms] DBG ---- COMMAND: Read6
[24501041ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00 
[24501042ms] DBG ------ Read 1x512 starting at 0
[24501043ms] DBG ---- DATA_IN
[24506047ms] scsi_accel_rp2040_stopWrite() timeout, FIFO levels 0 0 PC 23
[24506047ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 63139
[24506048ms] DBG -- BUS_FREE

PetteriAimonen, 2023-04-13:
@pbrs Thanks for the logs and info.

I have a new test build here (download zip at bottom of the page):
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/4691230351

It adds some more logging to the timeout message, in hopes that it would reveal something.

pbrs, 2023-04-13:
@PetteriAimonen amazing! Thank you!

I updated to the provided build and tried it again, this is the output I am getting now (I cropped the beginning as there is nothing new really):

[35271ms] Active configuration:
[35272ms] -- SelectionDelay = 0
[35273ms] -- EnableUnitAttention = No
[35273ms] -- EnableSCSI2 = No
[35274ms] -- EnableSelLatch = Yes
[35275ms] -- MapLunsToIDs = No
[35275ms] -- EnableParity = Yes
[35382ms] Finding HDD images in directory /:
[35383ms] -- Opening /HD00_512 MPC 56MB.IMA for id:0 lun:0
[35402ms] DBG ---- Image file is contiguous, SD card sectors 41088 to 153087
[35403ms] ---- Read prefetch enabled: 8192 bytes
[35409ms] -- Platform supports ROM drive up to 3740 kB
[35410ms] ---- ROM drive image not detected
[35410ms] SCSI ID:0 BlockSize:512 Type:0 Quirks:0 ImageSize:56000kB
[35511ms] Initialization complete!
[50478ms] DBG BUS RESET
[51402ms] DBG ---- SELECTION: 0
[51403ms] DBG ---- COMMAND: RequestSense
[51403ms] DBG ------ OUT: 0x03 0x00 0x00 0x00 0x00 0x00 
[51404ms] DBG ---- DATA_IN
[51405ms] DBG ---- Total IN: 4 OUT: 0 CHECKSUM: 30
[51406ms] DBG ---- STATUS: 0 GOOD
[51410ms] DBG ---- MESSAGE_IN
[51411ms] DBG ------ IN: 0x00 
[51412ms] DBG -- BUS_FREE
[51412ms] DBG ---- SELECTION: 0
[51413ms] DBG ---- COMMAND: Read6
[51413ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00 
[51414ms] DBG ------ Read 1x512 starting at 0
[51415ms] DBG ---- DATA_IN
[56417ms] scsi_accel_rp2040_stopWrite() timeout
[56417ms] SCSI DMA state: WRITE_DONE
[56417ms] Current buffer: 0x00000000/0x00000000, next 0x00000000 bytes
[56418ms] SyncOffset: 0 SyncPeriod 0
[56418ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 28, instr 0x000080A0
[56419ms] PIO Data SM: tx_fifo 0, rx_fifo 0, pc 23, instr 0x000090E0
[56420ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[56421ms] DMA CH A: ctrl: 0x00000011 count: 0x00000000
[56421ms] DMA CH B: ctrl: 0x01020809 count: 0x00000001
[56422ms] DMA CH C: ctrl: 0x00008805 count: 0x00000000
[56422ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[56423ms] GPIO states: 0x3BFDEA55
[56423ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 63139
[56424ms] DBG -- BUS_FREE

PetteriAimonen, 2023-04-13:
Thanks. Indeed it is waiting host to release ACK after the final byte.

Here's another test build, this time trying to slow down the transfer speed with delays to see if it makes any difference:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/4691652538

pbrs, 2023-04-13:
Unfortunately it is still not working but the debug output changed slightly again (sorry but I am not able to decipher any of this :-)):

[23ms] Reading configuration from zuluscsi.ini
[24ms] Active configuration:
[25ms] -- SelectionDelay = 0
[26ms] -- EnableUnitAttention = No
[27ms] -- EnableSCSI2 = No
[27ms] -- EnableSelLatch = Yes
[28ms] -- MapLunsToIDs = No
[29ms] -- EnableParity = Yes
[136ms] Finding HDD images in directory /:
[137ms] -- Opening /HD00_512 MPC 56MB.IMA for id:0 lun:0
[156ms] DBG ---- Image file is contiguous, SD card sectors 41088 to 153087
[157ms] ---- Read prefetch enabled: 8192 bytes
[163ms] -- Platform supports ROM drive up to 3740 kB
[163ms] ---- ROM drive image not detected
[164ms] SCSI ID:0 BlockSize:512 Type:0 Quirks:0 ImageSize:56000kB
[264ms] Initialization complete!
[42455ms] DBG BUS RESET
[43380ms] DBG ---- SELECTION: 0
[43381ms] DBG ---- COMMAND: RequestSense
[43381ms] DBG ------ OUT: 0x03 0x00 0x00 0x00 0x00 0x00 
[43382ms] DBG ---- DATA_IN
[43383ms] DBG ---- Total IN: 0 OUT: 0 CHECKSUM: 0
[43383ms] DBG ---- STATUS: 0 GOOD
[43387ms] DBG ---- MESSAGE_IN
[43388ms] DBG ------ IN: 0x00 
[43388ms] DBG -- BUS_FREE
[43390ms] DBG ---- SELECTION: 0
[43390ms] DBG ---- COMMAND: Read6
[43391ms] DBG ------ OUT: 0x08 0x00 0x00 0x00 0x01 0x00 
[43391ms] DBG ------ Read 1x512 starting at 0
[43392ms] DBG ---- DATA_IN
[58301ms] --------------
[58302ms] WATCHDOG TIMEOUT, attempting bus reset
[58302ms] Platform: ZuluSCSI RP2040
[58302ms] FW Version: 23.04.12-devel Apr 13 2023 16:38:00
[58303ms] GPIO states: out 0x3BFEEE55 oe 0x178F3A00 in 0x3BFBEBFF
[58304ms] scsiDev.cdb: 0x08 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[58304ms] scsiDev.phase: 4
[58305ms] SCSI DMA state: IDLE
[58305ms] Current buffer: 0x00000000/0x00000000, next 0x00000000 bytes
[58306ms] SyncOffset: 0 SyncPeriod 0
[58306ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 3, instr 0x000060C8
[58307ms] PIO Data SM: tx_fifo 0, rx_fifo 0, pc 11, instr 0x000090A0
[58308ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[58308ms] DMA CH A: ctrl: 0x00020021 count: 0x00000000
[58309ms] DMA CH B: ctrl: 0x001F9007 count: 0x00000000
[58309ms] DMA CH C: ctrl: 0x00029009 count: 0x00000000
[58310ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[58310ms] GPIO states: 0x3BFFEBFF
[58311ms] STACK 0x2001A830:    0x00010168 0x00000000 0x00000400 0xD0000000
[58311ms] STACK 0x2001A840:    0x10027699 0x2000C90F 0x2000C92C 0x61000000
[58312ms] STACK 0x2001A850:    0x20023997 0x20023998 0x0000A080 0x2000CCD9
[58313ms] STACK 0x2001A860:    0x20023798 0x00000200 0x0000A080 0x2000CCF1
[58314ms] STACK 0x2001A870:    0x00000200 0x2001C860 0x0000A080 0x20001B1D
[58314ms] STACK 0x2001A880:    0x2001F6FB 0x20023798 0x0000A080 0x2000E581
[58315ms] STACK 0x2001A890:    0x02961CC7 0x20001AE1 0x00000900 0x00000200
[58316ms] STACK 0x2001A8A0:    0x20011100 0x2001BAFC 0x2001AEC0 0x00000200
[58317ms] DBG ---- Total IN: 512 OUT: 0 CHECKSUM: 63139
[58317ms] DBG -- BUS_FREE

PetteriAimonen, 2023-04-13:
Yeah, same cause of hang, just the slower code path has a bit different logging.

Do you have a multimeter? Could you measure the voltage on the DB25 connector after the hang:

* REQ pin: between 1 (+) and 24 (-), should be about 3.0 V
* ACK pin: between 5 (+) and 24 (-), I expect this will be below 0.5 V

[image]

This is to debug cabling or termination problems.

pbrs, 2023-04-13:
Ok, for some reason I measure about - 2V on both the REQ and ACK pin after the hang.

PetteriAimonen, 2023-04-13:
Hmm.. that could be the problem. 2V probably means the host is trying to have ACK high, but for some reason it is not rising all the way.

The pins should show 3V also when booting without ZuluSCSI connected, so you can try if it always stays at 2V even without device connected.

If it does, it might point to some problem with the motherboard.

pbrs, 2023-04-14:
Ok, so I just took some time to measure the pins of the DB25 connector again. During the hang I measured 2,7V between both 1-24 and 5-24. When I disconnect the Zulu the system starts up normally and the voltage shows 2,3V between both ACK and REQ. I confirmed this with my oscilloscope and there does not seem to be any switching of voltages as far as I could detect.

PetteriAimonen, 2023-04-14:
Hmm, for some reason ZuluSCSI is seeing the ACK as low when stuck - though it also must be toggling correctly because the transfer works before that. But in any case the 2.7V voltage seems ok, so it's not an issue with that.

I don't currently have more ideas on how to debug this, it seems the MPC60 just stops after the final byte has been read.

pbrs, 2023-10-03:
Hi everyone, 

after all I found the reason for this issue. It was indeed caused by a power supply issue where a ground connection of a molex connector coming from the power supply was bottlenecking the 5V rail current so that during power hungry tasks (which loading and saving seem to be) the supply on the firmware ICs would drop and the sampler would freeze. 

I managed to find the problem through testing the power supply outputs under load.

Thanks a ton again for all the help trying to pinpoint the problem.

aperezbios, 2023-10-03:
@pbrs thanks for circling back and providing this information in the issue. I really appreciate it, and I'm glad you've nailed down the source of the problem. These sorts of issues can be really tricky to nail down.