ZuluSCSI/ZuluSCSI-firmware #181: ZuluScsi ibm 8550 not working

Diegorobo, 2023-04-23:
hello all i hope you can help me.. i bought zulu scsi for my ibm 8550 ps/2 it uses a future domain mca card where i connected the scsi emulator. I put the .img files I followed the whole procedure but nothing is displayed.. in the logs it says that everything is ok. but unfortunately it doesn't work.

PetteriAimonen, 2023-04-23:
Turn on the "debug" DIP switch on ZuluSCSI and boot. If the system has "SCSI scan" or some similar function in e.g. BIOS, run it.

After that, you should have a file "zululog.txt" on the SD card. Post it here and we will take a look.

Diegorobo, 2023-04-23:
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11304465/zululog.txt)
Thanks, I enclose the file.. unfortunately Zulu scsi is not recognized at all.. I tried in every way before writing here and creating disturbance.

PetteriAimonen, 2023-04-23:
What SCSI address is your SCSI host card using? The addresses 0 and 7 are both quite common for the host card, so it might make some problem if you pick the same one for the device.

Do you have other SCSI devices that work with the system?

PetteriAimonen, 2023-04-23:
This is also worth trying in `zuluscsi.ini`:

<pre>
[SCSI]
EnableSelLatch = 1
EnableSCSI2 = 0
SelectionDelay = 0
</pre>

Based on the log it seems any selection pulses from the host are not getting through, the above settings help support devices with non-standard SCSI selection sequence.

Diegorobo, 2023-04-23:
use address 7. i have no other scsi system. when I start the computer the scsi future domain bios screen comes up. searches for scsi devices and finds nothing giving me error to check if the device is connected. while if I move the switch of the zulu scsi on boot at startup it sees scsi 0..1..2..3 etc.. but then nothing happens.

Diegorobo, 2023-04-23:
I change the settings with the zuluscsi.ini?

Hd286, 2023-04-23:
The log says SD card detected, FAT64 volume size: 4963 MB. Shouldn't it be FAT or FAT16 with a volume size of 504mb?

Diegorobo, 2023-04-23:
then I entered the new parameters as suggested above... the problem is that nothing has changed in the txt log. the card is formatted fat32 and the img file is 4gb. it is 2 gb.. in fact I don't understand why the different values.. I was also trying to figure out how to completely reset the zulu scsi card but I don't understand how to do it, it only says the boot switch.. but nothing changes..

Diegorobo, 2023-04-23:
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11304567/zululog.txt)
new log with changed parameters.. as mentioned above... but nothing has changed why???

aperezbios, 2023-04-23:
@Diegorobo Thanks. Unfortunatley, both of the log files you've provided to us don't have debug mode enabled, which we need in this case. If you switch on the debug DIP switch, the ZuluSCSI must be power cycled before it will detect the DIP switch change, as it only reads this state on initial power-up. 

Here's the relevant part of the log file, showing that debug mode is not enabled (0 is disabled, the default state, and 1 is enabled)

```[11ms] DIP switch settings: debug log 0, termination 1```

Diegorobo, 2023-04-24:
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11310960/zululog.txt)
Hello everyone, I am attaching the file with the logs enabled. Thanks again for the help you give me

PetteriAimonen, 2023-04-24:
Good spot about the debug being off in earlier logs. Now it is correctly on, but still nothing in the log.

It's weird that there is still nothing in the log. The fact that the Future Domain bios detects (fake) devices when ZuluSCSI is in bootloader mode indicates that at least the cable has some connection.

I have a test build for you here which increases logging specific to this: https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/4789364966

To try it you need to do this:

1. Download the ZIP from bottom of the page linked above.
2. Extract the ZIP and copy **ZuluSCSI_RP2040_2023-04-24_d9847f3.bin** to your SD card.
3. Next time you power up ZuluSCSI with the SD card, the LED should blink rapidly while it is updating.
4. Then try again, with debug enabled, to see if we get anything new in the log.

PetteriAimonen, 2023-04-25:
Thanks. It seems the BSY line from controller is getting through, but the SEL line is not. Could be some kind of cable problem, if you have another SCSI cable that is worth trying.

I have yet another debug build for narrowing down on what is happening:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/4795256497

Diegorobo, 2023-04-25:
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11320429/zululog.txt)
new log

Diegorobo, 2023-04-25:
the cable I have 2 both give the same problem. I would exclude the cable.. the zuluSCSI power supply comes directly from the scsi.. I also tried to power it separately but the result does not change.

Thanks for everything

PetteriAimonen, 2023-04-25:
Thanks, I have a guess on what may be happening now.

I think this is a combination of SCSI-1 single initiator mode, with very short selection pulses.
The shortest that `EnableSelLatch = 1` currently handles is about 200 ns, while a relaxed interpretation of the standard permits down to 90 ns.

I have another test build here, this time reacting to even very short SEL pulses:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/4795844653

PetteriAimonen, 2023-04-25:
Hmm, I guess it releases the SCSI data pins also too quickly to detect the selected SCSI ID.

This will require a bit more changes for a real fix, but to check if my guess is right here is a test build which always selects drive 0:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/4796130153

I expect that should make every SCSI ID 0-7 be detected as present, and all will access the same image file.

Diegorobo, 2023-04-25:
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11321321/zululog.txt)
I uploaded the new fw. but I get no results. it always tells me no scsi device or cable unplugged

PetteriAimonen, 2023-04-25:
Hmm, very weird. I'll have to think about this a bit..

Diegorobo, 2023-04-25:
from what I understand the communication from the zuluSCSI to the future domain card is partially there or am I wrong?

PetteriAimonen, 2023-04-25:
Yeah, it seems that the Future Domain card is using some non-standard way of SCSI device selection.

We have seen similar before on several early SCSI devices, especially ones made before the 1990 release of SCSI-2 standard. But this seems to be somehow different than others.

Does the Future Domain card have any more specific model number or year visible?

Diegorobo, 2023-04-25:
the model is the following MCS-700 & IBM SCSI-2 last updated bios is 1994 version 3.61

Diegorobo, 2023-04-25:
[image]
because when I remove the termination on zuluscsi it does this... and with the termination inserted it tells me no scsi connected?

PetteriAimonen, 2023-04-25:
That sounds like no termination on host card. Both ends of cable should
have termination.

> Message ID: ***@***.***>
>

aperezbios, 2023-04-25:
@Diegorobo can you attach a photo of the top of the MCA SCSI controller card you're using? This may be helpful.

Diegorobo, 2023-04-25:
[image]

Diegorobo, 2023-04-25:
[image]

Diegorobo, 2023-04-25:
I updated the bios eeprom to 3.61. Why version 1.01 doesn't work on ibm 8550.

PetteriAimonen, 2023-04-26:
W2 TERMENA seems to be on, so termination should be on which is correct.

What happens when it scans for devices with no cable connected?

Diegorobo, 2023-04-26:
both with cable and without it tells me disconnected.. while with hdd scsi it is recognized

Diegorobo, 2023-04-28:
solved.. i bought an ibm scsi cable and as if by magic everything worked.. i didn't understand why my scsi cables weren't good.
Thank you all for the help and support you have given me.

Diegorobo, 2023-04-28:
[image]