ZuluSCSI/ZuluSCSI-firmware #194: Print PhyMode in the log file

tenox7, 2023-05-03:
Could ZuluSCSI firmware log what is the current negotiated phy mode for the SD card?

I'm referring to this config line:

```
#PhyMode = 0   # 0: Best available  1: PIO  2: DMA_TIMER  3: GREENPAK_PIO   4: GREENPAK_DMA
```

In case of 0 "Best available" it would be awesome to see what has been negotiated. 

Thanks

PetteriAimonen, 2023-05-03:
The PhyMode only affects GD32 platform. On it it will be printed as line such as:

    SCSI PHY operating mode: GREENPAK_DMA

aperezbios, 2023-05-04:
@tenox7 as Petteri mentioned, PhyMode is a platform-specific setting, and as of today, it only applies to ZuluSCSI V1.1. 

In f9a328a, I've gone ahead and improved the messaging in the comments of the example zuluscsi.ini file.

closed 2023-05-04 by aperezbios

tenox7, 2023-05-06:
awe thanks so much!