ZuluSCSI/ZuluSCSI-firmware #200: Issue booting cdrom on HP9000 C200 HP-UX 11.11

tenox7, 2023-05-06:
The CDROM is detected by search for bootable path but then errors when trying to boot from it.

It boots just fine with the same iso image burned to a physical cd.

See attached log and screenshot. Thanks!

[image]
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11411724/zululog.txt)

PetteriAimonen, 2023-05-06:
It seems to be failing at the read at sector 256368 because this repeats many times:

```
[645807ms] DBG -- BUS_FREE
[646810ms] DBG ---- SELECTION: 1
[646810ms] DBG ---- MESSAGE_OUT
[646811ms] DBG ------ OUT: 0xC0 
[646811ms] DBG ---- COMMAND: Read10
[646811ms] DBG ------ OUT: 0x28 0x00 0x00 0x03 0xE9 0x70 
[646812ms] DBG ------ OUT: 0x00 0x00 0x27 0x00 
[646813ms] DBG ------ Read 39x2048 starting at 256368
[646813ms] DBG ---- DATA_IN, syncOffset 8 syncPeriod 50
[646830ms] DBG ---- Total IN: 79872 OUT: 0 CHECKSUM: 36408
[646831ms] DBG ---- STATUS: 0 GOOD
[646837ms] DBG ---- MESSAGE_OUT
[646838ms] DBG ------ OUT: 0x06 
[646838ms] DBG -- BUS_FREE
```

Can you try with this setting in `zuluscsi.ini`:

    [SCSI]
    MaxSyncSpeed = 0

This bug has some similarities to #189 in that both are using 5MB/s synchronous mode and failing in early boot. But the other issue is with GD32 V1.1 and this is with RP2040, so the code paths are quite different.

aperezbios, 2023-05-06:
I have an HP9000 C200, and can dig it out if need be for further testing, @PetteriAimonen

tenox7, 2023-05-06:
Thanks for a prompt response! Yeah so this allowed to boot it but then when the OS starts reading the CD contents I get CRC errors.

[image]

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11413589/zululog.txt)

```
[SCSI]
Debug = 1
MaxSyncSpeed = 0
```

tenox7, 2023-05-06:
Nevermind... I re-wrote the iso image to be sure and now it booted.

tenox7, 2023-05-06:
well only a little further

[image]

tenox7, 2023-05-07:
Im happy to assist in any way to debug it. I helped debug some issues in SCSI2SD 6.0.

PetteriAimonen, 2023-05-08:
Is it possible to boot the system from some different media, and try to read out the image from ZuluSCSI when system is fully booted, instead of in bootloader?

Also just a double-check, make sure that termination is correct. I see that ZuluSCSI termination is enabled - make sure that it is the last device on the cable, and if there are other devices on the bus, that they have termination disabled.

tenox7, 2023-05-09:
Yeah it's quite possible there is some issue with termination. Zulu is the last device on the internal side. Whats curious is that with just zulu + terminator (or no terminator) on the external port the system wont even see ZuluSCSI. If I connect a cdrom drive with termination enabled to the other side, then ZuluSCSI will show.

I will spend some more time on this and also will try to boot it from the older HDD and read whats on Zulu from the OS.

tenox7, 2023-05-16:
I'm going to close this for now and reopen when I have more data.

closed 2023-05-16 by tenox7