ZuluSCSI/ZuluSCSI-firmware #206: Documentation seems wrong with regards to SD cards with a capacity smaller than 4GB

hugovangalen, 2023-05-11:
The documentation states, under "SD requirements" the following:

> With blue ZuluSCSI V1.x boards, Older SD cards manufactured prior to around 2008, generally ~4GB and smaller, may not be detected by the SDIO interface of the microcontroller.

I am unsure that what I observe is the same symptom as those SD cards on the 1.x boards would have. The red RP2040 based boards (rev. 2022d) also seems to have issues with such SD cards. 

No matter how I format them, it keeps blinking the led three times.

(I later figured out I could see debug log via /dev/ttyACM0, which gave me the output below, it seems it *does* detect something, just not the filesystem, and falls back to RAW mode.

I also notice the line "FAT0 volume size: 0 MB". Could it be that I am doing something wrong that I am not aware of?)

```[466542ms] SD card detected, FAT0 volume size: 0 MB
[466542ms] SD MID: 0x74, OID: 0x4A 0x60
[466543ms] SD Name: USD  
[466543ms] SD Date: 8/2022
[466543ms] SD Serial: 0x82014C1E
[466544ms] Config file zuluscsi.ini not found, using defaults
[466544ms] Active configuration:
[466545ms] -- SelectionDelay = 255
[466545ms] -- EnableUnitAttention = No
[466545ms] -- EnableSCSI2 = Yes
[466546ms] -- EnableSelLatch = No
[466546ms] -- MapLunsToIDs = No
[466546ms] -- EnableParity = Yes
[466548ms] Finding HDD images in directory /:
[466548ms] Could not open directory: /
[466548ms] -- Platform supports ROM drive up to 3740 kB
[466549ms] ---- ROM drive image not detected
[466550ms] No images found, enabling RAW fallback partition
[466550ms] Limiting RAW image mapping to SD card sector count: 3921920
[466551ms] DBG ---- Image file is contiguous, SD card sectors 0 to 3921919
[466552ms] ---- Read prefetch enabled: 8192 bytes
[468052ms] Failed to open log file: 0x00
```

aperezbios, 2023-05-11:
@hugovangalen thanks for the report. It would be helpful to know the brand and model number of the SD card you're using when the above happens. Without that information, all we can do is record the limitation, and move on. If there's anything laser-etched on the bottom of the SD card, a good in-focus, well-lit photo, legibile photo of this might also be helpful. Your initial report provides no detail about the SD card in use.

hugovangalen, 2023-05-12:
> @hugovangalen thanks for the report. It would be helpful to know the brand and model number of the SD card you're using when the above happens. Without that information, all we can do is record the [...]

Ah right, I will add that information!

hugovangalen, 2023-05-12:
> Seems that the card gets detected as 2 GB, though some errors at the beginning. But data transfer is probably failing, i.e. readSector() returns garbage/failure, which is why filesystem is not detected and the raw mapping won't really work either.

@PetteriAimonen Deep down, I was kind of hoping that it was some user error after all, or even a known problem that I could easily fix. (No such luck apparently.)

> So yeah, seems RP2040 version has some issues with small cards also. This should be fixable, but I'm not sure how much of a priority it is.

It would be really great, and I wonder what the reason for the failure is. The documentation mentions that for those older cards and boards, it's a limitation of the sillicon. It is unclear to me whether what I experience is the exact same issue, or whether this is something else. I did discover that, if I mess around with the start sector of the partition when setting it up with fdisk, it either reports only errors, or it detects it as "FAT0 with 0MB". The fact that it detect something does imply that the sillicon seems to operate fine, so possibly I'll experiment a bit more with different layouts. Just because I'm quite stubborn. :)

(BTW, I'm using many of these cards on my sampler succesfully with SCSI2SD (the average bank is just a couple of MB and it can't read more than 99 banks from a partition, on average, 500MB per partition is already overkill). I really would like to use ZuluSCSI to use different capacities more easily, because re-flashing the SCSI2SD every time when I want to access a different size SD is *obivously* a major hassle.)

PetteriAimonen, 2023-05-12:
I haven't myself been involved in the GD32 sd card problem debugging that much. But in general, very old SD cards require different initialization routine and some commands work differently. I'm not sure if GD32 SDIO hardware perhaps has some limitations in commands it can support.

In RP2040 the whole SDIO interface is our custom code, running on the programmable IO peripheral. So at least in theory there won't be silicon limitations, it's just a question of effort.

Regarding the timeouts, it is typical for SD cards to just ignore commands that they don't support. Then the CPU will time out waiting for response. The fix would be to make the code detect the situation and issue some older version of the command instead. (Unfortunately new cards don't support all of the old commands either, so it really needs version detection).

hugovangalen, 2023-05-12:
To clarify: this isn't even an old SD card, it was manufactured in 2022, but yeah, very likely it is an "old design" that they keep producing, they are speed class 2. So perhaps it is in any way related to that.

The SD cards that *do* work fine are all speed class 10. I will try to confirm that suspicion later today, I am sure I have a bunch of different SD cards lying around of varying speed classes and capacity.

hugovangalen, 2023-05-12:
That makes sense for those cards, yeah.

So I was curious and started looking a bit more at the code. The log I posted initially implies that it can correctly read the CID but fails after.

I was looking at the SdioCard::begin method inside ZuluSCSI_platform_RP2020/sd_card_sdio.cpp and I notice that the clock is set to 1Mhz when it initialises (via rp2040_sdio_init) and after reading that the clock gets set back to 25Mhz.

I can imagine the "older spec" cards simply not seeing the commands at that rate, so I wonder what happens if I keep that on 1Mhz.

Two questions regarding building the firmware myself:
- I don't need to cryptographically sign it to be able to install it, right?
- If I do screw things up, the bootloader mode uses its own SD logic, so I can always safely revert back to the official firmware if the clock division change *does* screw things up?

PetteriAimonen, 2023-05-12:
There is no signature in the firmware. Building should work directly with platformio, after it is [installed](https://platformio.org/install) you can run `pio run` to build firmware for all platforms, or `pio run -e ZuluSCSI_RP2040` to build just the RP2040 version.

The SD card bootloader never overwrites itself, and it executes first so from power-on restart it will be able to access SD card even if main firmware messes up communication. It's also possible to flash by USB using the RP2040 ROM bootloader, which is enabled by the dip switch.

I found a 2GB card myself, it gives the same `SdioCard::readSector(0x00000001)` errors for me.
I did a quick test commenting out the clock rate increase in `sd_card_sdio.cpp` but that didn't make a difference for me.
I can take a closer look next week.

This may be useful, even though it details the SPI mode: http://elm-chan.org/docs/mmc/mmc_e.html#spiinit
*"[For SDHC] The data read/write operations described below are commanded in block addressing (LBA) insted of byte addressing. The size of data block at block addressing mode is fixed to 512 bytes."* sounds like the smaller cards may need the byte addressing instead - currently the driver uses block addressing only.

hugovangalen, 2023-05-12:
Alright, thanks a lot for also looking into this.

In that document you linked, in the paragraph under "Initialization" I read this (but I am too unfamiliar with SD cards or controllers to know how to interpret it): "_The initial read/write block length might be set 1024 on 2 GB cards, so that the block size should be re-initialized to 512 with CMD16 to work with FAT file system._" Set to 1024 where, I wonder. In the chips on the SD card itself?

Anyway, looking at the firmware code, it seems that CMD16 is never issued on RP2040 but it *is* in the GD32 code. Though it seems to me that a different block size would just cause the chip/controller to return more bytes than the code expects and would not result in the observed time-out error situation but I am at this point blissfully ignorant of any underlying intricacies.

(BTW I could not find any class 10 low capacity SD cards yet, but I did find 64MB class 2 SD card which reports the exact same errors.)

PetteriAimonen, 2023-05-12:
CMD16 is issued here, though for some reason I have it written just `16` instead of the `CMD16` define:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/main/lib/ZuluSCSI_platform_RP2040/sd_card_sdio.cpp#L434-L436

aperezbios, 2023-05-12:
@hugovangalen Panasonic made 4GB Class 10 SDHC SD cards before they exited the SD card manufacturing market a year or two ago. Delkin Devices also manufactures a 4GB pSLC SD card, model [SF04APGJN-U3000-3](https://www.newark.com/webapp/wcs/stores/servlet/ProductDisplay?catalogId=15003&CMP=KNC-BUSA-GEN-Shopping-ReStructure-MID&productSeoURL=delkin-devices&partNumber=36AJ0766), which you can buy from through the electronics component distributor Newark, for about $15 each. They also manufacture the [SF04APYE9-U1000-3](https://www.newark.com/delkin-devices/sf04apye9-u1000-3/sdhc-card-uhs-1-class-10-4gb-mlc/dp/64AH8307), a 4GB SD card which is a little less expensive.

hugovangalen, 2023-05-15:
@PetteriAimonen Thanks for your work, I appreciate it. I will give that a whirl today and report back.

hugovangalen, 2023-05-15:
@aperezbios Thanks, though I am in Europe so we don't have Newark here but I do see that I can source 4GB class 10 locally. Though I really want to use all the 2GB that I have, and thanks to the work done by Petteri it seems I may be able to keep using them all, so that's awesome.

PetteriAimonen, 2023-05-15:
Yeah, the 2GB problem in RP2040 was definitely a bug that had just missed our testing. The issue referred to in the documentation is something more rare and less easy to fix.

hugovangalen, 2023-05-15:
It seems to work perfectly fine on my sampler now: I could succesfully load a sample bank, format an empty drive image, and save the sample bank onto that newly formatted drive. I was eyeballing the log via `cat /dev/ttyACM0` and it looks perfectly fine. 

Thanks again!

(And I understand that the older boards with older cards not working being a different issue. At least that paragraph stating that newer cards (though with less than 4GB capacity) solely won't on the older boards - thereby implying they could work fine on newer boards, is now technically correct. Well, with this latest firmware anyway.)

PR #208 (2023-05-15): RP2040: Fix support for 2GB SD cards
RP2040 SDIO driver had a bug/missing feature that prevented standard capacity (up to 2GB) SD cards from working.

Also fixed a crash bug when switching between FAT and exFAT formatted cards on the fly.

Fixes #206.
closed 2023-05-15 by PetteriAimonen