ZuluSCSI/ZuluSCSI-firmware #210: CD-ROM: MSF address behavior does not match specifications

saybur, 2023-05-17:
There are some minor glitches related to MSF addresses, including:

- Addresses in the firmware start at 00/00/00 when they should be 00/02/00.
- Track relative addresses should be positive descending numbers when in a pause/pre-gap.
 
 I'd be happy to put in a PR for these if desired.

PetteriAimonen, 2023-05-17:
PRs welcome :)

The first one I'm very unsure of based on specification. It does say the data track has unencoded 2 second pregap. I guess if you have a host that does use the Read MSF command it will be obvious what way it should be.

Regarding track relative addresses, "T10/1545-D MMC-4 Revision 5a" says:

> The Track Relative CD Address field gives the current location relative to the logical beginning of the
current track. If the CDB TIME bit is zero, this field is a track relative LBA. If the current block is in
the pre-gap area of a track, this is a negative value, expressed as a two’s-complement number. If the
TIME bit is set to one, this field is the relative TIME address from the Q Sub-channel formatted
according to Table 29.

So am I interpreting this correctly that in LBA mode, they are negative numbers, but in MSF mode something else? Positive descending numbers sounds strange, but the MMC spec is pretty new to me :)

saybur, 2023-05-17:
I'm definitely not super familiar with this stuff, so I am more than happy to be proven incorrect on anything!

The 00/02/00 minimum is shown a little better in MMC-1 Table 1, particularly with note 3. That table also reflects the weird downward-counting relative MSF addresses. MMC-1 Table 173 has the corresponding math that includes the 150 sector offset for LBA conversion. Interestingly, the current ZuluSCSI implementation seems to work fine, so I don't know if any of this actually _matters_ in practice: as of now it's mostly an exercise in adhering to the spec for its own sake.

If you haven't used it yet, I've also found the (freely available) ECMA-130 to be useful for Yellow Book information, especially to help clarify the sparse descriptions for Q-channel data in MMC. Section 20 contains the info for track requirements, which has that 2 second pause for the first track. 22.3.3.3 has the relative MSF description.

I'll get working on a PR here and see what I can come up with.

PetteriAimonen, 2023-05-18:
Thanks for the references, I hadn't looked at those specific documents!

PR #215 (2023-05-23): CDROM: Table of contents and addressing improvements
Fixes #210 and adjusts other aspects of the CD-ROM subsystem. This covers a bit of ground, so if it would be better to break up the PR I'd be happy to do so. Main changes:

- TOC handling has been retrofitted to more closely match observed behavior on devices, including some vendor-specific SCSI-2 options that appear to have been _de-facto_ standards before MMC.
- Addressing has been adjusted to improve the behavior of audio players. Playback tracking is now much more reliable. Fast-forwarding/rewinding and track seeking is still a bit iffy  [... cut]
closed 2023-05-23 by aperezbios