ZuluSCSI/ZuluSCSI-firmware #211: Implement stub support for additional tape commands to avoid I/O errors

aperezbios, 2023-05-18:
@PetteriAimonen, I did some basic SCSI tape testing with 'mt' under Linux, and identified eight commands that immediately induce an I/O error. The erase command is probably the most critical, followed by the compression command. These should probably all be no-ops for now. Do you have an opinion?
```
[root@PW490 ~]# mt -f /dev/st0 fsf
/dev/st0: Input/output error
[root@PW490 ~]# mt -f /dev/st0 fsfm
/dev/st0: Input/output error
[root@PW490 ~]# mt -f /dev/st0 bsf
/dev/st0: Input/output error
[root@PW490 ~]# mt -f /dev/st0 bsfm
/dev/st0: Input/output error
[root@PW490 ~]# mt -f /dev/st0 bsr
/dev/st0: Input/output error
[root@PW490 ~]# mt -f /dev/st0 tell
/dev/st0: Input/output error
[root@PW490 ~]# mt -f /dev/st0 erase
/dev/st0: Input/output error
[root@PW490 ~]# mt -f /dev/st0 compression
/dev/st0: Input/output error
```

sapperlott, 2024-03-16:
One candidate for an existing variable length record tape image format to use could be AWSTAPE which is used by the Hercules mainframe emulator. It already comes with a tool called "tapecopy" which can copy between AWSTAPE image files and real tapes. Its source can be used as an example to understand how the format is used:
https://github.com/SDL-Hercules-390/hyperion/blob/master/tapecopy.c

While Hercules provides an open source implementation of the format, it originates from IBM and has been used in their virtual tape libraries for a long time. So there's probably a lot of tape images already out there in that format.

aperezbios, 2024-03-16:
Great. Thanks for the suggestion, @sapperlott