ZuluSCSI/ZuluSCSI-firmware #212: ModeSense with AllocationLength 31 provides one byte too much

PetteriAimonen, 2023-05-18:
Log (Amiga 3000 with ZuluSCSI RP2040):

```
[58607ms] DBG -- BUS_BUSY
[58607ms] DBG ---- SELECTION: 0
[58608ms] DBG ---- MESSAGE_OUT
[58609ms] DBG ------ OUT: 0xC0 
[58609ms] DBG ---- COMMAND: ModeSense
[58610ms] DBG ------ OUT: 0x1A 0x00 0x04 0x00 0xFE 0x00 
[58610ms] DBG ---- DATA_IN
[58611ms] DBG ---- Total IN: 32 OUT: 0 CHECKSUM: 14460
[58612ms] DBG ---- STATUS: 0 GOOD
[73176ms] --------------
[73176ms] WATCHDOG TIMEOUT, attempting bus reset
[73177ms] Platform: ZuluSCSI RP2040
[73177ms] FW Version: 23.05.04-release May 4 2023 16:37:33
[73178ms] GPIO states: out 0x3BFCE4FF oe 0x178F3BFF in 0x3BFDE4FF
[73178ms] scsiDev.cdb: 0x1A 0x00 0x04 0x00 0xFE 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[73179ms] scsiDev.phase: 6
[73180ms] SCSI DMA state: IDLE
[73180ms] Current buffer: 0x00000000/0x00000000, next 0x00000000 bytes
[73181ms] SyncOffset: 0 SyncPeriod 0
[73181ms] PIO Parity SM: tx_fifo 0, rx_fifo 0, pc 28, instr 0x000080A0
[73182ms] PIO Data SM: tx_fifo 0, rx_fifo 0, pc 23, instr 0x000090E0
[73182ms] PIO Sync SM: tx_fifo 0, rx_fifo 0, pc 0, instr 0x00000000
[73183ms] DMA CH A: ctrl: 0x00000011 count: 0x00000000
[73184ms] DMA CH B: ctrl: 0x00020809 count: 0x00000000
[73184ms] DMA CH C: ctrl: 0x00008805 count: 0x00000000
[73185ms] DMA CH D: ctrl: 0x00009809 count: 0x00000000
[73185ms] GPIO states: 0x3BFDE4FF
[73186ms] STACK 0x2001B1D0:    0x00010168 0x3BF9E4FF 0x400140F4 0x00000400
[73186ms] STACK 0x2001B1E0:    0x100276CD 0x2000D173 0x2000D18C 0x21000000
[73187ms] STACK 0x2001B1F0:    0x00000000 0x0003409C 0x2002409C 0x00010198
[73188ms] STACK 0x2001B200:    0x00000000 0x2000A22B 0x00000000 0x000100E0
[73189ms] STACK 0x2001B210:    0x00000000 0x2002409C 0x00000000 0x2001FF74
[73190ms] STACK 0x2001B220:    0x00010168 0x2000A99D 0x00000000 0x00000006
[73190ms] STACK 0x2001B230:    0x2001B238 0x00000000 0x00000000 0x00000000
[73191ms] STACK 0x2001B240:    0x00000000 0x00000000 0x00000000 0x20001A91
[73192ms] DBG ---- MESSAGE_IN
[73205ms] DBG -- BUS_FREE
```

The host requests ModeSense page 4 with maximum allocation length of 0xFE = 31 bytes. For some reason ZuluSCSI returns 32 bytes. The extra byte is apparently interpreted as status byte, resulting in timeout when the actual status byte never gets read.

The ModeSense command seems to correctly limit the number of bytes read, so it's not clear where the extra byte is coming from:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/blob/main/lib/SCSI2SD/src/firmware/mode.c#L620

PetteriAimonen, 2023-05-18:
Err, no.. 0xFE = 254 not 31..

MeTalOneOEight, 2023-05-19:
Are you returning a C Array here?
Maybe you did not account that it starts at index 0 and if you return 0..idx you get idx+1 values?

PetteriAimonen, 2023-05-19:
Actually ZuluSCSI is working correctly in this case, it was a problem with me reading the debug log last night.
I somehow thought it was requesting 31 bytes, when it was requesting 254 bytes (in which case providing the full 32 is fine).

The actual problem with this specific machine appears to be something else.

closed 2023-05-19 by PetteriAimonen

aperezbios, 2023-05-19:
Customer responded this morning with "Looks like it was an issue with my A3000 board. It’s all working now. Just going to stress test it to make sure" :)