ZuluSCSI/ZuluSCSI-firmware #216: SD not recognized when used in external enclosure - RP2040 compact

PPC505, 2023-05-20:
Trying to replace a 50-pin SCSI drive in an external SCSI (DB25) enclosure. Device will boot normally when attached internally, but will throw 5-blink (SD not detected) code when attached in enclosure. SD is a 32GB SanDisk Ultra micro with a known working ~7.5GB partition with MacOS 8.6 installed.

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11521737/zululog.txt)

PetteriAimonen, 2023-05-20:
If SD card is not detected, the zululog is probably from a previous run.

Most likely cause is if the cable is not able to supply enough power to operate the device. You could try powering with USB cable or the 4-pin connector to see if it makes a difference.

PPC505, 2023-05-21:
Good call. That was it. I'll have to get the floppy power connector attached for this setup then. Thanks!

closed 2023-05-21 by aperezbios