ZuluSCSI/ZuluSCSI-firmware #223: 2023.05.24: CDROM defined in zuluscsi.ini is gone

ThatRed, 2023-05-25:
Hi, 

after installing version 2023.05.24, the CDROM defined in zuluscsi.ini is gone:

[SCSI6]
Product = "CDROM"
Type = 2

# If IMG0..IMG9 are specified, they are cycled after each CD eject command.
IMG0 = UAKM_DISK1.ISO
IMG1 = UAKM_DISK2.ISO
IMG2 = UAKM_DISK3.ISO
IMG3 = UAKM_DISK4.ISO

After downgrading to the previous version it's back.

Is this a bug or were commands in zuluscsi.ini changed?

Thanks in advance.

aperezbios, 2023-05-25:
Thank you for the report. We will investigate today.

PetteriAimonen, 2023-05-25:
I can confirm the bug, I'll see what caused it..

aperezbios, 2023-05-25:
Yep, I was just able to reproduce it myself. You have to have at least two ISOs defined for it to manifest.

PetteriAimonen, 2023-05-25:
I think it should be fixed now:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/latest

aperezbios, 2023-05-25:
@PetteriAimonen with the latest nightly, it still isn't working for me.

aperezbios, 2023-05-25:
@ThatRed This regression has been fixed in [v2023.05.25](https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/v2023.05.25), which I've just made and tested.

closed 2023-05-25 by aperezbios