ZuluSCSI/ZuluSCSI-firmware #225: Can't read HFS track on 3-track BeOS 5.0.3 BIN/CUE

aperezbios, 2023-05-26:
Using today's released firmware, and the following bin/cue files,  Mac OS does not fully recognize the 96MB Mac-specific HFS track in the BeOS 5.0.3 Professional bin/cue I have linked below. I burned the exact same bin/cue file to CD-ROM and used it in the optical drive, and it mounted right away. There's some sort of bug to be tracked down here. @PetteriAimonen, you should be able to test/reproduce this on your 68K Mac. Finder never mounts the filesystem (with System=Mac set in zuluscsi.ini) even though System Profiler sees that there is an HFS filesystem on the volume

The cue file itself is very basic: 

```
FILE "beos-5.0.3-professional-gobe.bin" BINARY
  TRACK 01 MODE1/2352
    INDEX 01 00:00:00
  TRACK 02 MODE1/2352
    INDEX 01 10:48:58
  TRACK 03 MODE1/2352
    INDEX 01 46:07:03

```

https://ia802306.us.archive.org/35/items/beos-5.0.3-professional-gobe/beos-5.0.3-professional-gobe.bin
https://ia802306.us.archive.org/35/items/beos-5.0.3-professional-gobe/beos-5.0.3-professional-gobe.cue

PetteriAimonen, 2023-05-26:
Hmm, I can't seem to detect any problem. The 2352 byte sector access is not very well tested, but appears to work for me.
Tests I did:

1. Added test case to CUEParser checking that this specific CUE file is parsed correctly.
2. Mounted with ZuluSCSI on Linux and also with cdemu-client on Linux, the results compare equal.
3. ZuluSCSI RP2040 with Mac LC III booted from `HD60_512 System 7.5.3.hda`, it appears to mount the partition:

[image]

Or is there supposed to be more volumes visible than the "BeOS_Tools"?

aperezbios, 2023-05-26:
That's strange, I will try it again. Maybe I was doing something wrong.

aperezbios, 2023-06-02:
It works fine here as well, not sure what was happening. Closing as invalid.

closed 2023-06-02 by aperezbios