ZuluSCSI/ZuluSCSI-firmware #230: ReinsertCDOnInquiry behavior inconsistent with documentation

saybur, 2023-06-02:
The documented behavior for `ReinsertCDOnInquiry` in the INI file is:

```
#ReinsertCDOnInquiry = 1 # Reinsert any ejected CD-ROM image on Inquiry command
```

This seems to describe a 'CD tray retract' type operation. However, in ZuluSCSI_disk.cpp this setting is used to call cdromReinsertFirstImage(). This prevents multi-image configurations from working on INQUIRY-happy operating systems until the setting is disabled.

For reference, I ran into this on modern Linux: at least on Debian bullseye it likes to issue INQUIRY after every new disk is inserted.

PetteriAimonen, 2023-06-19:
This functionality was added in #141 based on private discussion with @aperezbios. The main issue to solve was:

> I've had some feedback from a few customers that the current eject behavior of ZuluSCSI firmware is problematic. Can you think of a way to "re-insert" the current/sole CD-ROM ISO upon a bus reset? As it stands right now, the only way to force this to happen is to de-energize (or maybe just hardware re-set) the board.

> CD-ROM is already reinserted after bus reset.
But some hosts won't issue a bus reset on reboot (debug log should tell).

Prior to that, reinsertion happened on BUS RESET only. It appears that some systems didn't issue BUS RESET, though I wasn't able to reproduce this on Mac LC III at the time.

I think closing the tray instead of returning to first image would be consistent with the needs for INQUIRY command, where the problem was that Mac has no software selection for "close tray". But maybe on BUS RESET it still makes sense to restart from first image?

@aperezbios How does this sound:

1) If CD tray is open on `INQUIRY`, close it but don't switch image.
2) On BUS RESET, return to first image.

aperezbios, 2023-06-19:
That sounds perfectly reasonable to me.