ZuluSCSI/ZuluSCSI-firmware #231: WATCHDOG TIMEOUT when using latest release 23.05.25

WhatIsTheWeather, 2023-06-02:
When using the latest ZuluSCSI release with my digital master unit I have been experiencing the error WATCHDOG TIMEOUT, followed by a bus reset, with the machine unable to read the iso stored on the SD card. I can confirm that this error is not present in the previous ZuluSCSI release. I have attached a log from the previous release that works with the machine and log from the newest release with the bug. The ZULUSCSI is set up as a CDROM and I have not changed the iso when testing both versions.

[zululogworking.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11636924/zululogworking.txt)

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11636929/zululog.txt)

PetteriAimonen, 2023-06-02:
<pre>
[9913ms] DBG -- BUS_BUSY
[9913ms] DBG ---- SELECTION: 0
[9915ms] DBG ---- COMMAND: Read6
[9915ms] DBG ------ OUT: 0x08 0x00 0x00 0x20 0x00 0x00 
[9915ms] DBG ------ Read 0x2048 starting at 32
[9916ms] DBG ---- DATA_IN
[9917ms] DBG ------ Found 0 sectors in prefetch cache
[9917ms] DBG ---- Total IN: 0 OUT: 0 CHECKSUM: 0
[9918ms] DBG ---- STATUS: 0 GOOD
</pre>

From [SCSI-2 spec](https://www.staff.uni-mainz.de/tacke/scsi/SCSI2-09.html#9.2.5):

> The transfer length field specifies the number of contiguous logical blocks of data to be transferred. A transfer length of zero indicates that 256 logical blocks shall be transferred. Any other value indicates the number of logical blocks that shall be transferred. 

but the new code added in fcac5ae7c lacks the special case of 0 = 256.

I'll get this fixed, thanks for the report!

PetteriAimonen, 2023-06-02:
There is a test build here that you can try (download zip at bottom of page):
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/5157595032

aperezbios, 2023-06-02:
@WhatIsTheWeather thanks for the report. Please try the build @PetteriAimonen linked to above if you're able to.

WhatIsTheWeather, 2023-06-02:
Thanks for getting a new build created so quickly. I can confirm that the new build has fixed the issue. I have installed two ZuluSCSI units into the machine to replace the A and B side CD drives and both are working fine with the latest build. I will keep updating both with the new releases as they appear and will report any further bugs I find. 

Thanks :).

closed 2023-06-06 by WhatIsTheWeather