ZuluSCSI/ZuluSCSI-firmware #240: Latest FW release caused write issues to SD card, rolling back to May release resolved?

luvwagn, 2023-06-11:
Perhaps my card is failing, but it is working fine in my PC for WinUAE (Amiga emulator usage).

[10ms] Platform: ZuluSCSI RP2040
[10ms] FW Version: 23.06.03-release Jun  4 2023 00:20:41
[14107ms] SDIO card reports write CRC error, status 0xD63FFFFF
[14108ms] SdioCard::writeSectors(0x02C45482,...,18) failed: 7
[14110ms] SD card write failed: 0x00

I rolled back the firmware to latest May release on the SD card, re-inserted back into my Amiga, and write errors went away?

PetteriAimonen, 2023-06-12:
Hmm, there shouldn't be any difference in SD card handling between those releases.
CRC error would indicate a problem with connections to the SD card (e.g. dirty contacts or card not fully inserted), rather than a problem with the card itself.

Did the error messages occur repeatedly or just once?

PetteriAimonen, 2023-06-19:
@luvwagn Do you have any new information? If this only occurred once, might it be that there was some problem with SD card connection?

luvwagn, 2023-06-19:
Apologies, my scsi card is being updated to latest fw and offline. Once I get it back, I'll try to recreate the problem.

eharmon, 2023-06-20:
I can confirm the same problem. Constant write errors with the same code in the log with the June releases.

ZuluSCSI RP2040, Quadra 650

Rolling back also appears to resolve for me.

PetteriAimonen, 2023-06-20:
@eharmon Thanks for confirmation. Can you provide a zululog.txt of the problem with debug messages turned on (either by DIP switch or with zuluscsi.ini)?

aperezbios, 2023-06-20:
@eharmon it would also be helpful to know which exact SD card you I u are using. Make and model, which is often laser-engraved on the bottom of the SD or MicroSD card.

eharmon, 2023-06-27:
After downgrading and upgrading the firmware, I can't get it to reproduce anymore. But I changed another variable which gives me an idea it might not just be a fluke: I unplugged a PiSCSI I had connected to the external SCSI port.

I've noticed the Zulu repeatedly running off the PiSCSI's termination power. That is, if the machine is off, the Zulu keeps running. And if the machine is on, and the Zulu has heavy activity, the Pi starts to throw under-voltage warnings significantly often.

I'm wondering if the newer firmware increases the power load on the Zulu which might be leading to bus errors to the SD card when it's sapping termination power from another device on the SCSI chain.

What's odd is one is on the internal bus and the other on the external, so it's not like the bus is being over-terminated.

I'll continue to see if I can replicate the problem again.

djfarrell, 2023-07-08:
I saw a number of errors in the log file. When I re-imaged (using initiator mode) I got errors in different sectors from the first time.  I binary compared both 9GB image files and they were identical.  The log may be generating false positives.

PetteriAimonen, 2023-08-02:
@djfarrell When a write error is detected, it should be handled and the write re-attempted, which would correct the transfer error. But normally there shouldn't be transfer errors at all.

closed 2023-12-03 by aperezbios