ZuluSCSI/ZuluSCSI-firmware #247: Issue with CD eject command

guytaylor, 2023-06-17:
I can't get the CD eject command cycling CD-Rom iso images working using a ZuluSCSI 1.1 (latest firmware) on a Kurzweil K2500

This is my .ini file:

```
[SCSI1]
Product = "CD-ROM Drive"
Type=2 # CDROM drive
IMG0 = img1.iso
IMG1 = img2.iso
IMG2 = img3.iso
EjectButton = 1
```

and I see this in the log:
```
[288ms] -- Opening 'img1.iso' for id:1, specified in zuluscsi.ini
...
[314ms] SCSI ID:1 BlockSize:2048 Type:2 Quirks:0 ImageSize:97336kB
...
[22717ms] Eject button 1 pressed, passing to CD drive SCSI1
[29099ms] Eject button 1 pressed, passing to CD drive SCSI1
[41098ms] Eject button 1 pressed, passing to CD drive SCSI1
[52984ms] Eject button 1 pressed, passing to CD drive SCSI1
[74900ms] Eject button 1 pressed, passing to CD drive SCSI1
```

It is stuck on img1.iso and never changes. 
I have tested all the iso separately and they work.

PetteriAimonen, 2023-06-17:
Can you post a full zululog.txt with debug messages enabled?

PetteriAimonen, 2023-06-17:
I think the issue may be that we currently switch the image only after the host has polled for status. If the host is not checking the CD-ROM status, that probably won't work. But debug log will tell more.

guytaylor, 2023-06-17:
I did the following:

1. Powered Zulu
2. Powered K2500
3. Looked at SCSI 1 on the K2500, and I could see disc1.iso
4. Pressed Button 1
5. Browsed away from SCSI 1 and returned to scsi1. Still saw disc1. 
6. Pressed Button 1
7. Browsed away from SCSI 1 and returned to scsi1. Still saw disc1. 
8. Powered down k2500, powered it back up (left Zulu powered on)
9. Looked at SCSI 1 on the K2500, I could see disc1.iso
10. Pressed Button 1
11. Pressed 'SCSI sleep' on K2500, Browsed away from SCSI 1,  and returned to scsi1. Still saw disc1. 
12. Pressed Button 1
13. Browsed away from SCSI 1 and returned to scsi1. Still saw disc1. 

Here is the full log:
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11780224/zululog.txt)

If all is working ok from the Zulu side and it is a case that the K2500 isn't "polling for status" Is there any way that there could be an option added for the Zulu to 'force switch image' on the button press and not wait for the status poll (it would seem the k2500 isn't issuing one)?

Hot swap works, as in the K2500 will read the card if I removed it, then reinsert it, so I would assume forcing a switch of image would be safe if it is just for read-only ISO disc images? 

I would so like this feature to work! Thank you :)

PetteriAimonen, 2023-06-19:
I think this may be same problem as #230.

Can you try adding this in `zuluscsi.ini`?

    [SCSI]
    ReinsertCDOnInquiry = 0

guytaylor, 2023-06-19:
Sadly that didn't work:

New .ini:

```
[SCSI]
Debug = 1
ReinsertCDOnInquiry = 0

[SCSI1]
Product = "CD-ROM Drive"
Type=2 # CDROM drive
IMG0 = disc1.iso
IMG1 = disc2.iso
IMG2 = disc3.iso
EjectButton = 1
```

Log:
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11786500/zululog.txt)

PetteriAimonen, 2023-06-19:
Hmm, apparently there was also a bug on how that setting was implemented.

I have a test build here that should work better (download zip at bottom of page).
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/5310561294

I think it should work for you no matter if ReinsertCDOnInquiry is on or off in the config file.

guytaylor, 2023-06-19:
Thanks for the zip. Yes that worked :)

```
[10429ms] Eject button 1 pressed, passing to CD drive SCSI1
[10429ms] DBG ------ CDROM open tray on ID 1
[10430ms] Switching to next CD-ROM image for 1: disc2.iso
[10438ms] DBG ---- Image file is contiguous, SD card sectors 29624640 to 29836663
[10438ms] ---- Read prefetch enabled: 8192 bytes
[12495ms] DBG -- BUS_BUSY
```

djfarrell, 2023-07-08:
Nice product, saves quite a bit of time and money.
For RP2040 initiator mode 9GB IBM drive.  Generated the 9G image but the log indicates several errors.  If I re-image the errors appear at different sectors.  A binary compare of both images indicates they match, including areas where sector errors were logged in different places.  This is for a HP 16700 logic analyzer running HPUX, image boots and so far runs fine.
It should be noted that if a CD drive is loaded and on the same bus, the initiator will also try to image the CD after the drive.
The LED blink is a hard to use as a indicator.  The 9GB image took about two hours, the blinking was hard to figure out when things were done.  Since there is a I2C port, maybe a cheap OLED display for status?

tenox7, 2023-08-08:
I have same problem. I can test the new build if you want.

Does the eject button generate unit attention?

PetteriAimonen, 2023-08-08:
@tenox7 Currently it does not generate UNIT_ATTENTION condition, but reports the ejected status as response to polling commands.

Can you enable debug (`[SCSI] Debug=1` in ini or with the DIP switch) and post another log?

I assume the problem is that the host doesn't detect disc change? Because based on the log the image seems to be changed correctly.

tenox7, 2023-08-12:
See attached. Whats happening is I press the button, the image changes, but the host is not aware of it and I get errors in file explorer that directory is corrupted. I think that unit attention would help with this. Could we make it as an option?
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/12327141/zululog.txt)

PetteriAimonen, 2023-08-17:
@tenox7 Currently in your log, at e.g. `1800726ms` timestamp, the ejection is reported as sense 0x3A00 (medium not present), which works for most hosts.

I have a test build here that additionally reports UNIT ATTENTION with 0x2800 (medium may have changed):
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/5889113588

It needs a zuluscsi.ini with:

    [SCSI]
    EnableUnitAttention = 1

tenox7, 2023-08-19:
Unit Attention worked for me! I can change the CDs now! Thank you so much.

PetteriAimonen, 2023-08-21:
@tenox7 Thanks for testing! I'll make a pull request and this will be part of next release.

closed 2023-08-21 by aperezbios

tenox7, 2023-08-22:
Thank you so much!!!

commit d62283bf (2023-08-21): Merge pull request #283 from ZuluSCSI/dev_unit_attention_cd_remove

Post UNIT_ATTENTION condition after CD medium change (fixes #247)