ZuluSCSI/ZuluSCSI-firmware #248: EjectButton 2 is not recognised anymore

ThatRed, 2023-06-17:
Hello,

after installing v2023.06.13 the "EjectButton = 2" is not recognized anymore. 

I went back to v2023.06.03 and it's still broken. I don't know when this problem occurred. 

If I change the physical connection of button 1 to 2 and 2 to 1, it's still not detected. I measured the cables and buttons and they all work.    

I tried to download this [https://github.com/ZuluSCSI/ZuluSCSI-firmware/issues/135#issuecomment-1564169055](url) version, but it's gone...

PetteriAimonen, 2023-06-17:
Which hardware are you on? Can you post a debug log?

ThatRed, 2023-06-18:
@PetteriAimonen 

Platform: ZuluSCSI RP2040

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11780842/zululog.txt)

I hope this helps, thank you.

ThatRed, 2023-06-19:
@saybur Damn! You were right, I have somehow choosen the _Audio Version... 

Thank you, and sorry for the confusion.

closed 2023-06-20 by ThatRed