ZuluSCSI/ZuluSCSI-firmware #258: Issue using BIN/CUE audio cd files with Otari Digital Master

WhatIsTheWeather, 2023-07-01:
Hi, I recently tested using bin/cue audio cd rips with my Otari digital master unit. Looking at the log it appears to open the bin/cue files fine and detects the 12 tracks on the cue sheet, however the machine throws out an error and does not load the audio. It looks like it is erroring out after attempting to read the TOC. 

Here is the corresponding log.
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/11926671/zululog.txt)

aperezbios, 2023-07-01:
@WhatIsTheWeather thanks for the report. Everything in the log looks great until the last (unknown) command:

```[11288ms] DBG -- BUS_FREE
[11289ms] DBG ---- SELECTION: 0
[11290ms] DBG ---- COMMAND: Unknown
[11291ms] DBG ------ OUT: 0xD8 0x00 0x00 0x00 0x00 0x02 
[11291ms] DBG ---- STATUS: 2 CHECK_CONDITION, sense 0x00002400```

We may need to eventually get you a test firmware build to extract more information, as this is a device-specific situation where we don't have access to hardware to replicate, but it may be a few weeks, as there are some developers on holiday/vacation for much of July.

WhatIsTheWeather, 2023-07-21:
Sounds good to me. I will keep an eye on this issue for any updates on the post. Happy to test any test firmware that is created to help diagnose the issue further. Thanks.

WhatIsTheWeather, 2024-01-15:
Just here to confirm that I am still experiencing this issue when trying to use audio cd bin/cue files with the latest release.

morio, 2024-01-18:
After searching a bit I found that the 0xD8 SCSI command is vendor specific to Plextor CD-ROM drives. Did the Otari digital master unit originally ship with a Plextor CD-ROM drive?
Out of curiosity what is the model number of the Otari system?

In any case it will be hard to recreate the command as I couldn't find the structure of the command.
I'm adding links here that better describe the SCSI op code D8 for future reference:
https://github.com/superg/redumper/blob/main/README.md
http://forum.redump.org/topic/2057/offset-detection-using-the-d8-read-command-discussion/
http://fxr.watson.org/fxr/source/cam/scsi/scsi_all.c

morio, 2024-01-18:
Could you try this firmware, hopefully it will print out the command in full,
It doesn't address the issue, just collects a little more data
Please attach the debug log
[zuluscsi-custom-cmd-d8.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/13982071/zuluscsi-custom-cmd-d8.zip)

morio, 2024-01-18:
Could you try this firmware, hopefully it will print out the command in full,
It doesn't address the issue, just collects a little more data
Please attach the debug log

[zuluscsi-custom-cmd-d8-v1.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/13982136/zuluscsi-custom-cmd-d8-v1.zip)

WhatIsTheWeather, 2024-01-18:
Thanks for getting back to me. Yeah, the Otari originally shipped with two Plextor Ultraplex SCSI CD-ROM drives. The model number is DP-3. The unit plays audio cds at 16x normal speed for tape duplication. It is possible to play two audio cds simultaniously to record both sides at the same time, when doing this the unit outputs the audio for the b side in reverse.

morio, 2024-01-20:
@WhatIsTheWeather This probably won't work, not exactly sure what the block size should be for the Plextor 0xD8 read. The way the docs read, it seems to output raw CD data, so that's what is attempted here. I'll push up the code changes I've made to, if you feel like trying your hand at the source code.
[zuluscsi-custom-cmd-d8-v2.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/13996157/zuluscsi-custom-cmd-d8-v2.zip)

WhatIsTheWeather, 2024-01-20:
Thanks for that, I have just tried the test firmware and it is somewhat working. The bin/cue is getting played for a few seconds but then the unit displays a disc read error on the LCD. After checking the the recorded tape the audio seems to be playing but with skips and glitches before stopping. I have also tried enabling SCSI-2 support thinking that might help but with no luck there. 

Here is the log file

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/13997174/zululog.txt)

morio, 2024-01-22:
@WhatIsTheWeather 
With the v3 firmware under 
`[SCSI]`
could you add `PlextorBytesPerSector=`
and try 
```
[SCSI]
PlextorBytesPerSector=2048
```

and

```
[SCSI]
PlextorBytesPerSector=2352
```

[zuluscsi-custom-cmd-d8-v3.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14013660/zuluscsi-custom-cmd-d8-v3.zip)

WhatIsTheWeather, 2024-01-22:
Thanks for the update. I have tried the new firmware with both PlextorBytesPerSector values and the results are the same. The machine starts playing the audio for around 7-8 seconds and then displays disc read error on the LCD. When checking the recorded tape the audio seems to be still slightly glitchy and skipping forward 1-2 seconds at regular intervals. 

I have attached the latest log file using the v3 firmware below. I'm wondering if there is some type of delay happening between each d8 command which is causing the 1-2 second skipping of the audio? The audio is slightly glitchy but seems to be playing back correctly aside from the 1-2 second skips and a bit of additional noise. 

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14014068/zululog.txt)

morio, 2024-01-22:
@WhatIsTheWeather 
Sorry, made a mistake with the last build.
This one, v4,  should actually use the different bytes per sector size. v3 read the zuluscsi.ini PlextorBytesPerSector but didn't actually use the value.
[zuluscsi-custom-cmd-d8-v4.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14015093/zuluscsi-custom-cmd-d8-v4.zip)

WhatIsTheWeather, 2024-01-22:
No worries, I appreciate that you are looking into this. I have just tested again with the v4 build and using PlextorBytesPerSector=2048 seems to sort out the skipping issue, however the unit still has disc read errors at around the same point of roughly 8 seconds of copy time which translates into around 120 seconds of duplicated audio. Maybe there is some type of error checking going on but I cannot see anything related in the logs. 

Attached is the most recent log

[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14015360/zululog.txt)

morio, 2024-01-22:
I think 2352 bytes is the actual recording size for audio CDs, but what is happening is the SD card is based on 2048 bytes per block. The 2352 bytes is causing unaligned reads from the SDCard which slow the system causing the skips. I'm going to try doing two aligned reads and extracting the the 2352 bytes from there.

morio, 2024-01-22:
@WhatIsTheWeather 
This hopefully does aligned reads off the SD card, but writes just the CD sector to the SCSI bus.
This should be
```
PlextorBytesPerSector=2352 
```
[zuluscsi-custom-cmd-d8-v5.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14016483/zuluscsi-custom-cmd-d8-v5.zip)

WhatIsTheWeather, 2024-01-23:
> @WhatIsTheWeather This hopefully does aligned reads off the SD card, but writes just the CD sector to the SCSI bus. This should be
> 
> ```
> PlextorBytesPerSector=2352 
> ```
> 
> [zuluscsi-custom-cmd-d8-v5.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14016483/zuluscsi-custom-cmd-d8-v5.zip)

With this firmware I am unable to load the audio at all, with a disc read error appearing when loading up the bin/cue. I have set PlextorBytesPerSector=2352 in the INI but am unfortunately having no luck with this change.

morio, 2024-01-30:
Haven't forgot this, should have a binary to test tomorrow.

morio, 2024-01-30:
@WhatIsTheWeather 
Hopefully this version won't skip and that data will agree with the cue file. I was able to do some testing with Linux, so I'm a little more confident that the SCSI commands will work. There is an end case that still needs to addressed in reading from the SD card. 

By any chance could you post your `.cue` file along with the` zululog.txt` or even better your bin file too?

[zuluscsi-custom-cmd-d8-v6.zip](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14102821/zuluscsi-custom-cmd-d8-v6.zip)

morio, 2024-02-01:
@WhatIsTheWeather 
You can find the latest build here: https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/7734296068
Under 
`[SCSI]`
 or 
`[SCSIx]` where x is the SCSI ID
You'll need the line 
`VendorExtensions =  1`

I tested a ZuluSCSI with a windows app that dumps data via the SCSI 0xD8 command (px_d8.exe) in hex and checked it against a .bin file via a hex editor and it matches.

It would be helpful if you could post your .cue file so I can double check that the ZuluSCSI should dump the correct data.

WhatIsTheWeather, 2024-02-01:
> @WhatIsTheWeather You can find the latest build here: https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/7734296068 Under `[SCSI]` or `[SCSIx]` where x is the SCSI ID You'll need the line [...]

Thanks for the update. I have just tested this recent build and it seems to be reading the data from the bin/cue correctly without any audio glitches. However the unit exhibits a disk read error after around 60 seconds which would be 16 minutes into the audio as the unit outputs at 16x normal speed. Within the bin/cue there are two audio tracks, track 1 for side an and track 2 for side b. I have recorded two subsequent tapes and it seems to be dropping out at the same point each times. 

Here are the bin/cue files I have been using, which were created using Sonoris audio mastering software. Had to upload via wetransfer as GitHub did not let me upload it internally. 

[https://we.tl/t-BLMEPq3MrP](url)

Here is the recent ZuluSCSI log. 
[zululog.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14131212/zululog.txt)

Thanks for the help with this, I appreciate you taking the time to look into this.

morio, 2024-02-08:
@WhatIsTheWeather could you run the recording a few times with debug on? I'd like to see if it stops at the same sector or is random.
Also try running it with debug off to see if it runs longer that when with it on (maybe a speed issue).

Thanks

WhatIsTheWeather, 2024-02-08:
> @WhatIsTheWeather could you run the recording a few times with debug on, capturing the log each time? I'd like to see if it stops at the same sector or is random. Also try running it with debug off to see if it runs longer that when with it on (maybe a speed issue).
> 
> Thanks

I have just tested it again, once with debug off and it seems to run without errors with debug mode switched off, however with debug mode on it seems to error out at exactly the same point when trying multiple times. So I think you a right in saying it is probably a speed issue. 

This is the point in the debug log which it seems to error out at every time.

Total IN: 602112 OUT: 0 CHECKSUM: 7649

Here are the three logs: 

1. Debug mode off
[zululog-nodebug.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14215538/zululog-nodebug.txt)

2. Debug mode on first try
[zululog-debug1.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14215542/zululog-debug1.txt)

3. Debug mode on second try 
[zululog-debug2.txt](https://github.com/ZuluSCSI/ZuluSCSI-firmware/files/14215543/zululog-debug2.txt)

Thanks.

morio, 2024-02-09:
@WhatIsTheWeather 
Thanks for the update, glad it is working!
When I have time to revisit it, I may want to try to integrate back into preexisting code before pull it to main. 
But for now good luck with the Otari!

WhatIsTheWeather, 2024-02-09:
> @WhatIsTheWeather
> Thanks for the update, glad it is working!
> When I have time to revisit it, I may want to try to integrate back into preexisting code before pull it to main.
> But for now good luck with the Otari!

Thanks for all the help with this, it is very much appreciated! Do you think having debug mode enabled was causing the slowdowns before erroring out? When I get a free moment over the weekend I will have a look at the output of the master unit with an oscilloscope to check how the waveforms are looking. I understand audio cds do not have error correction so any errors would be introduced as glitches and noises in the audio before the machine errors.

WhatIsTheWeather, 2024-03-21:
I have just tried the latest ZuluSCSI firmware 2024.03.07 and the audio on the b side has started erroring out during playback, I have also retested the custom firmware and that works with no issue. Do you know what could be causing this issue with the latest firmware? I have not enabled debug logs as that seemed to be contributing to the previous issue with the test firmware so it may be hard to diagnose the issue. I am using the same zuluscsi.ini with both versions and have debug logs disabled when using the latest firmware. The A side seems to playback fine but the B side seems to error out at the same point every time with the latest firmware. There are no perceivable audio glitches but it seems to cut off at the same point each time.

aperezbios, 2024-03-21:
@WhatIsTheWeather can you explicitly state which version of the firmware you were using where you are NOT experiencing this issue?

WhatIsTheWeather, 2024-03-22:
> @WhatIsTheWeather can you explicitly state which version of the firmware you were using where you are NOT experiencing this issue?

https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/7734296068 - this is the version of the firmware that seems to work without issue when debug mode is turned off. The latest main release seems to be having issues with the B side audio erroring out. I have tested both builds with the same bin/cue files and zuluscsi.ini.

morio, 2024-03-23:
@WhatIsTheWeather I switched code to use the original CD playback code instead of the custom data loading I did, hoping that it was fast enough. I'll see if I can optimize the original code or add back the custom loading next week. For now use that special build.

morio, 2024-03-25:
Hi @WhatIsTheWeather, I found a bug and fixed it here:
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/8426156287

This has a minor optimization if the above does not work
https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/8426610283

I added a initialization message if the plextor extension is enabled
```
[287ms] ---- Configuring as CD-ROM drive
[288ms] ---- Plextor 0xD8 vendor extension enabled 
```
If neither work, I'll add the aligned reads back. I'm hesitant to add it back because it will fail if the bin file isn't contiguous.

WhatIsTheWeather, 2024-03-25:
> Hi @WhatIsTheWeather, I found a bug and fixed it here: https://github.com/ZuluSCSI/ZuluSCSI-firmware/actions/runs/8426156287
>
> This has a minor optimization if the above does not work [...]

Thanks for pushing the swift fix. The top link to the firmware with the bug fix and optimisation seems to have fixed the issue - I have ran off a few different bin/cue files for the A and B sides and have not had a single issue with this latest update. It is much appreciated!

aperezbios, 2024-04-02:
@WhatIsTheWeather This fix is now incorporated in to the [v2024.04.01](https://github.com/ZuluSCSI/ZuluSCSI-firmware/releases/tag/) release, which you should now upgrade to, when you have a chance. I'm going to close this as completed, but feel free to comment here if there's anything that doesn't work as expected, after upgrading to firmware v2024.04.01