ZuluSCSI/ZuluSCSI-firmware #259: Feature request

djfarrell, 2023-07-08:
The LED is difficult to use when initiator mode copies large (9GB) drives.  Since there is I2C, can a cheap OLED display be an option?

aperezbios, 2023-07-09:
@djfarrell yes I agree it would be nice to have something like that, and we will eventually have something like that. In the interim, the easy way to know exactly what's going on during a hard drive dump is to connect your ZuluSCSI RP2040 to a computer (or even a mobile phone, using a USB OTG cable), and use the virtual serial port/output to monitor the logged status of the activity.  Exactly how you go about doing that depends on the platform/OS of the machine you've got the board attached to.

csulaguy, 2023-07-31:
Funny, I came on here to request this same feature just now. I'm interested in following up with this potential feature as well.

tenox7, 2023-08-22:
+1 for OLED to display status

lausvi, 2024-03-10:
I'll try the serial console next time, but I'd like to also chime in on the LED indication being somewhat hard to read... Not sure how easy the timing of the blinking is to change, but I was thinking that could it be something like this: 4 fast blinks (like now, or something else, to indicate we are in initiator mode) and then 1-10 slow blinks to indicate a decade of percentage done? And then 4 fast again. Or if the LED can be faded, a dim-bright indication of progress?